arendst/Tasmota #2629: KNX Address Configuring

jeylites, 2018-05-05:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

I have a group of OLD sonoff T1's that works with KNX while the NEW sonoff T1's don't. Is my addressing setup correct?

Basically, I will like to group the 3 sonoff T1 (Relay 1, as one) I could turn ON/OFF a light from any of this group switches.

[image]
[image]

ascillato, 2018-05-05:
Hi,

Your KNX configuration is OK.

1- When you say new sonoff and old sonoff, which firmware version did you mean?

2- When you say erratically what do you mean?

3- Please show the status 0 and also please use the command weblog 4 on the console and show the behaviour output on the console.

jeylites, 2018-05-05:
sorry for not including info below earlier. What is status 0?

1- When you say new sonoff and old sonoff, which firmware version did you mean?

Tasmota 5.13.1a
ESP 2.4.1

2- When you say erratically what do you mean?

What I meant is there are times it works as it should occasionally. 

3- Please show the status 0 and also please use the command weblog 4 on the console and show the behaviour output on the console.

```
00:00:00 Project sonoff JEY1 (Topic jey1, Fallback 26, GroupTopic jey1) Version 5.13.1a-2_4_1
00:00:00 WIF: Connecting to AP1 SMILE in mode 11N as jey1-4683...
00:00:10 WIF: Connect failed with AP incorrect password
00:00:10 WIF: Connecting to AP1 SMILE in mode 11N as jey1-4683...
00:00:20 WIF: Connect failed with AP incorrect password
00:00:21 WIF: Connect failed with AP incorrect password
00:00:21 WIF: Connecting to AP1 SMILE in mode 11N as jey1-4683...
00:00:25 WIF: Connected
00:00:25 HTP: Web server active on jey1-4683 with IP address 192.168.0.26
00:00:26 MQT: Attempting connection...
00:00:29 MQT: Connected
00:00:29 MQT: tele/jey1/LWT = Online (retained)
00:00:29 MQT: cmnd/jey1/POWER = 
00:00:29 MQT: tele/jey1/INFO1 = {"Module":"Sonoff T1 3CH","Version":"5.13.1a","FallbackTopic":"26","GroupTopic":"jey1"}
00:00:29 MQT: tele/jey1/INFO2 = {"WebServerMode":"Admin","Hostname":"jey1-4683","IPAddress":"192.168.0.26"}
00:00:29 MQT: tele/jey1/INFO3 = {"RestartReason":"External System"}
00:00:30 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
00:00:30 MQT: stat/jey1/POWER1 = OFF
00:00:30 MQT: stat/jey1/RESULT = {"POWER2":"OFF"}
00:00:30 MQT: stat/jey1/POWER2 = OFF
00:00:30 MQT: stat/jey1/RESULT = {"POWER3":"OFF"}
00:00:30 MQT: stat/jey1/POWER3 = OFF
00:00:37 MQT: tele/jey1/STATE = {"Time":"1970-01-01T00:00:37","Uptime":"0T00:00:35","POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SMILE","RSSI":76,"APMac":"00:72:63:51:68:58"}}
11:47:23 LOG: SerialLog 2, WebLog 4, SysLog 0, LogHost , LogPort 514, TelePeriod 300
11:47:23 HTP: Configuration
11:47:26 CFG: Saved to flash at F4, Count 232, Bytes 2560
11:47:32 HTP: Main Menu
11:47:32 WIF: Checking connection...
11:47:32 WIF: Connected
11:47:34 HTP: Console
11:47:53 WIF: Checking connection...
11:47:53 WIF: Connected
11:48:13 WIF: Checking connection...
11:48:13 WIF: Connected
11:48:27 HTP: Main Menu
11:48:29 KNX: Relay 1 = 1 sent to 5.0.26
11:48:29 MQT: stat/jey1/RESULT = {"POWER1":"ON"}
11:48:29 MQT: stat/jey1/POWER1 = ON
11:48:29 CFG: Saved to flash at FB, Count 233, Bytes 2560
11:48:31 KNX: Relay 1 = 0 sent to 5.0.26
11:48:31 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
11:48:31 MQT: stat/jey1/POWER1 = OFF
11:48:31 CFG: Saved to flash at FA, Count 234, Bytes 2560
11:48:32 HTP: Console
11:48:33 WIF: Checking connection...
11:48:33 WIF: Connected
```
```
00:00:00 Project mom4 MOM 4 (Topic mom4, Fallback 23, GroupTopic sonoffs) Version 5.13.1a-2_4_1
00:00:00 WIF: Connecting to AP1 SUGAR1 in mode 11N as mom4-4871...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on mom4-4871 with IP address 192.168.0.23
00:00:05 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/mom4/LWT = Online (retained)
00:00:06 MQT: cmnd/mom4/POWER = 
00:00:06 MQT: tele/mom4/INFO1 = {"Module":"Sonoff T1 3CH","Version":"5.13.1a","FallbackTopic":"23","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/mom4/INFO2 = {"WebServerMode":"Admin","Hostname":"mom4-4871","IPAddress":"192.168.0.23"}
00:00:06 MQT: tele/mom4/INFO3 = {"RestartReason":"External System"}
00:00:07 MQT: stat/mom4/RESULT = {"POWER1":"ON"}
00:00:07 MQT: stat/mom4/POWER1 = ON
00:00:07 MQT: stat/mom4/RESULT = {"POWER2":"OFF"}
00:00:07 MQT: stat/mom4/POWER2 = OFF
00:00:07 MQT: stat/mom4/RESULT = {"POWER3":"OFF"}
00:00:07 MQT: stat/mom4/POWER3 = OFF
00:00:14 MQT: tele/mom4/STATE = {"Time":"1970-01-01T00:00:14","Uptime":"0T00:00:14","POWER1":"ON","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR1","RSSI":88,"APMac":"88:D7:F6:02:18:80"}}
00:00:14 MQT: stat/mom4/RESULT = {"POWER1":"ON"}
00:00:14 MQT: stat/mom4/POWER1 = ON
00:00:15 MQT: stat/mom4/RESULT = {"POWER2":"OFF"}
00:00:15 MQT: stat/mom4/POWER2 = OFF
00:00:15 MQT: stat/mom4/RESULT = {"POWER3":"OFF"}
00:00:15 MQT: stat/mom4/POWER3 = OFF
00:02:00 MQT: tele/mom4/UPTIME = {"Time":"1970-01-01T00:02:00","Uptime":"0T00:02:00"}
00:03:44 LOG: SerialLog 2, WebLog 4, SysLog 0, LogHost , LogPort 514, TelePeriod 300
00:03:44 HTP: Configuration
00:03:45 CFG: Saved to flash at FA, Count 162, Bytes 2560
00:03:48 HTP: Main Menu
00:03:49 WIF: Checking connection...
00:03:49 WIF: Connected
00:03:50 HTP: Console
00:04:15 WIF: Checking connection...
00:04:15 WIF: Connected
00:04:16 HTP: Console
00:04:21 HTP: Console
00:04:35 WIF: Checking connection...
00:04:35 WIF: Connected
00:04:48 HTP: Main Menu
00:04:50 MQT: stat/mom4/RESULT = {"POWER1":"OFF"}
00:04:50 MQT: stat/mom4/POWER1 = OFF
00:04:51 CFG: Saved to flash at F9, Count 163, Bytes 2560
00:04:55 WIF: Checking connection...
00:04:55 WIF: Connected
00:04:55 MQT: stat/mom4/RESULT = {"POWER1":"ON"}
00:04:55 MQT: stat/mom4/POWER1 = ON
00:04:56 CFG: Saved to flash at F8, Count 164, Bytes 2560
00:04:58 HTP: Console
00:05:17 WIF: Checking connection...
00:05:17 WIF: Connected
00:05:27 MQT: tele/mom4/STATE = {"Time":"1970-01-01T00:05:27","Uptime":"0T00:05:14","POWER1":"ON","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR1","RSSI":90,"APMac":"88:D7:F6:02:18:80"}}
```

ascillato, 2018-05-05:
Hi,

Status 0 is a command to type on the console. This shows the device information

There are some issues using the esp lib 2.4.1. Not only with KNX.

Please, compile it with 2.3.0 and choose lwIP 1.4

Or download the precompiled from [Tasmota](https://github.com/arendst/Sonoff-Tasmota/releases) or from [Tasmota_KNX](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases). The last one are the same Tasmota but compiled using ESP8266 Board Library v2.3.0 and lwIP v1.4

Another thing to test is: using an app on your phone like Multicast Tester, see if the KNX telegrams are being transmitted by your wifi router.

jeylites, 2018-05-05:
For the Tasmota KNX bin, are you using Tasmota MQTT library?

jeylites, 2018-05-05:
STATUS 0 (192.168.0.23) MOM
```
00:00:00 CFG: Loaded from flash at F5, Count 167
00:00:00 APP: Boot Count 14
00:00:00 Project mom4 MOM 4 (Topic mom4, Fallback 23, GroupTopic sonoffs) Version 5.13.1a-2_4_1
00:00:00 CFG: Saved to flash at F4, Count 168, Bytes 2560
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 SUGAR1 in mode 11N as mom4-4871...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Connected
00:00:06 KNX: Start
00:00:06 HTP: Web server active on mom4-4871 with IP address 192.168.0.23
00:00:07 MQT: Attempting connection...
00:00:13 MQT: Connected
00:00:13 MQT: tele/mom4/LWT = Online (retained)
00:00:13 MQT: cmnd/mom4/POWER = 
00:00:13 MQT: Subscribe to cmnd/mom4/#
00:00:13 MQT: Subscribe to cmnd/sonoffs/#
00:00:13 MQT: Subscribe to cmnd/23/#
00:00:13 MQT: tele/mom4/INFO1 = {"Module":"Sonoff T1 3CH","Version":"5.13.1a","FallbackTopic":"23","GroupTopic":"sonoffs"}
00:00:13 MQT: tele/mom4/INFO2 = {"WebServerMode":"Admin","Hostname":"mom4-4871","IPAddress":"192.168.0.23"}
00:00:13 MQT: tele/mom4/INFO3 = {"RestartReason":"Power on"}
00:00:15 MQT: stat/mom4/RESULT = {"POWER1":"OFF"}
00:00:15 MQT: stat/mom4/POWER1 = OFF
00:00:15 MQT: stat/mom4/RESULT = {"POWER2":"OFF"}
00:00:15 MQT: stat/mom4/POWER2 = OFF
00:00:15 MQT: stat/mom4/RESULT = {"POWER3":"OFF"}
00:00:15 MQT: stat/mom4/POWER3 = OFF
00:00:22 MQT: tele/mom4/STATE = {"Time":"1970-01-01T00:00:22","Uptime":"0T00:00:22","POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR1","RSSI":92,"APMac":"88:D7:F6:02:18:80"}}
00:00:23 HTP: Main Menu
00:00:23 RSL: Received Topic cmnd/mom4/POWER1, Data Size 3, Data OFF
00:00:23 RSL: Group 0, Index 1, Command POWER, Data OFF
00:00:23 MQT: stat/mom4/RESULT = {"POWER1":"OFF"}
00:00:23 MQT: stat/mom4/POWER1 = OFF
00:00:23 RSL: Received Topic cmnd/mom4/POWER2, Data Size 3, Data OFF
00:00:23 RSL: Group 0, Index 2, Command POWER, Data OFF
00:00:23 MQT: stat/mom4/RESULT = {"POWER2":"OFF"}
00:00:23 MQT: stat/mom4/POWER2 = OFF
00:00:23 RSL: Received Topic cmnd/mom4/POWER3, Data Size 3, Data OFF
00:00:23 RSL: Group 0, Index 3, Command POWER, Data OFF
00:00:23 MQT: stat/mom4/RESULT = {"POWER3":"OFF"}
00:00:23 MQT: stat/mom4/POWER3 = OFF
00:00:26 WIF: Checking connection...
00:00:26 WIF: Connected
00:00:26 MQT: stat/mom4/RESULT = {"POWER1":"ON"}
00:00:26 MQT: stat/mom4/POWER1 = ON
00:00:26 CFG: Saved to flash at FB, Count 169, Bytes 2560
00:00:27 MQT: stat/mom4/RESULT = {"POWER1":"OFF"}
00:00:27 MQT: stat/mom4/POWER1 = OFF
00:00:27 CFG: Saved to flash at FA, Count 170, Bytes 2560
00:00:29 HTP: Console
00:00:39 CMD: Status 0
00:00:39 RSL: Received Topic /Status, Data Size 1, Data 0
00:00:39 RSL: Group 0, Index 1, Command STATUS, Data 0
00:00:39 MQT: stat/mom4/STATUS = {"Status":{"Module":30,"FriendlyName":["MOM 4","MOM 42","MOM 43"],"Topic":"mom4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:00:39 MQT: stat/mom4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:36","StartupUTC":"","Sleep":0,"BootCount":14,"SaveCount":170,"SaveAddress":"FA000"}}
00:00:39 MQT: stat/mom4/STATUS2 = {"StatusFWR":{"Version":"5.13.1a","BuildDateTime":"2018-05-05T00:50:22","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
00:00:39 MQT: stat/mom4/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SUGAR1","SUGAR2"],"TelePeriod":300,"SetOption":["02008009","55818000"]}}
00:00:39 MQT: stat/mom4/STATUS4 = {"StatusMEM":{"ProgramSize":469,"Free":532,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
00:00:39 MQT: stat/mom4/STATUS5 = {"StatusNET":{"Hostname":"mom4-4871","IPAddress":"192.168.0.23","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"60:01:94:98:B3:07","Webserver":2,"WifiConfig":4}}
00:00:39 MQT: stat/mom4/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.200","MqttPort":1883,"MqttClientMask":"23","MqttClient":"23","MqttUser":"admin","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
[... 102 lines cut ...]
16:06:44 WIF: Connected
16:07:04 WIF: Checking connection...
16:07:04 WIF: Connected
16:07:24 WIF: Checking connection...
16:07:24 WIF: Connected
16:07:44 WIF: Checking connection...
16:07:44 WIF: Connected
16:08:01 HTP: Main Menu
16:08:04 WIF: Checking connection...
16:08:04 WIF: Connected
16:08:08 HTP: Console
16:08:14 CMD: Status 0
16:08:14 RSL: Received Topic /Status, Data Size 1, Data 0
16:08:14 RSL: Group 0, Index 1, Command STATUS, Data 0
16:08:14 MQT: stat/jey1/STATUS = {"Status":{"Module":30,"FriendlyName":["JEY1","JEY12","JEY13"],"Topic":"jey1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
16:08:14 MQT: stat/jey1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"jey1","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T04:23:26","StartupUTC":"2018-05-05T03:44:48","Sleep":0,"BootCount":18,"SaveCount":241,"SaveAddress":"FB000"}}
[... cut ...]
16:08:14 MQT: stat/jey1/STATUS7 = {"StatusTIM":{"UTC":"Sat May 05 08:08:14 2018","Local":"Sat May 05 16:08:14 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":8}}
16:08:14 MQT: stat/jey1/STATUS10 = {"StatusSNS":{"Time":"2018-05-05T16:08:14"}}
16:08:14 MQT: stat/jey1/STATUS11 = {"StatusSTS":{"Time":"2018-05-05T16:08:14","Uptime":"0T04:23:26","POWER1":"ON","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SMILE","RSSI":72,"APMac":"00:72:63:51:68:58"}}}
```

ascillato, 2018-05-06:
> For the Tasmota KNX bin, are you using Tasmota MQTT library?

Yes

> What is the difference between both bins?

* Tasmota MQTT library (it uses more flash but when there is no MQTT Server, this library don't hung the device for 5 seconds)
* Compiled using ESP8266 Board Library v2.3.0 and lwIP v1.4 (because I had wifi problems and sleep command problems with newer libraries)

ascillato, 2018-05-06:
Hi,

Do you still have the issue?

jeylites, 2018-05-06:
Compiling with core 2.3.0 ran me into wifi disconnection issues. With 2.4.0, KNX seems responsive, no weird behavior like 2.4.1. Currently having the following...

Core: 2.4.0
Tasmota: 5.13.1.a
wifi sleep mode off

Hardware: Sonoff T1, 3gang

closed 2018-05-06 by jeylites