arendst/Tasmota #2631: KNX IP (Change IP of KNX Interface)

TheInfernoMan, 2018-05-05:
Hello,

i am using a SONOFF Basic with the newest Tasmota Firmware and KNX Build (I think 13.1). I wanted to add the Basic to my KNX network and cant get it work.

I think I know what the problem is. The KNX Wiki Page of Tasmota says all UDP messages are send / received by one IP (224.0.23.12 : 3671). I think this would work perfectly with multiple SONOFF devices alone. But I want to integrate the device in my existing KNX network with an existing IP Interface.

But my IP Interface for KNX (also sending on 3671) has another IP. Is it possible to change the IP of the KNX IP Device to my IP Interface? Is there a way with a COMMAND or with the help of adding an option in the "Configure KNX" Site?

Thanks in Advance

ascillato, 2018-05-05:
Hi.

Ok. I will add that option.

That IP address/port is the KNX standard. Which address/port do you have? Why is different to the standard?

TheInfernoMan, 2018-05-05:
My IP Address is: 192.168.178.40 I think and this is because my IP Gateway is also a SmartHome Gateway (InSideControl) and like the name suggest should be reachable from the internet and so the IP is like the ones in my LAN so that I can make a port forward if I want to connect to it from the Internet.

TheInfernoMan, 2018-05-05:
I think this option is only necessary for KNX IP Gateways with other functions like SmartHome and not the standard. So it would be nice to change it in the GUI but as standard the standard IP and Port should be set.

ascillato, 2018-05-05:
Thanks for the explanation. I will add that option.

ascillato, 2018-05-12:
Ah, remember that your IP Gateway Address has nothing to do with the Multicast IP Address.

IP and Multicast IP are totally different things. An IP is the address of a particular device on a network.

A multicast IP is a virtual address to where all the devices can talk to each other there.

ascillato, 2018-05-12:
Hi, I look for the KNX insideControl Gateway and it is a KNX certified gateway. So, if your model support multicast communication it is on the default address. So, please check the config of the gateway to see if it is turned on the multicast support. Also checks your wifi router if it has multicast turned on.

For the insidecontrol it is not needed to change the default broadcast address.

spcqike, 2023-03-02:
Hi, 

i also try to setup tasmota based KNX devices. but i guess, i have an unusual setup.

my network is divided into multiple VLAN, of which one is for general IoT devices, one for internal, one for KNX only.

afaik the IP gateway translates all KNX messages and broadcasts them on the VLAN. but, my tasmota devices are in another VLAN, so the broadcast doesn't work (which is expected as one doesn't want to flood networks with udp broadcasts from unrelated devices)
virtual machines and such can communicate on the KNX bus, if they are placed in the KNX VLAN.

is it possible to set the IP of the Gateway? Like in[ Node-Red](https://flows.nodered.org/node/node-red-contrib-knx-easy) its possible to specify the gateway and send messages across different networks. 

if thats not possible, i guess one can still use a workaround (mqtt - nodered - knx), but i would prefer a more or less direct action between the devices, if somehow possible.

ty

barbudor, 2023-03-02:
Tasmota supports 1 default gateway
It may be filled from dhcp
Statically you can set it with IpAddress2

liurenqiu520, 2024-05-14:
实测有效，修改`/lib/lib_div/esp-knx-ip-0.5.2/src/esp-knx-ip.h`
```
#define MULTICAST_IP              IPAddress(192, 168, xx, xx) // [Default IPAddress(224, 0, 23, 12)]
```
修改IPAddress为knx/ip网关真实IP地址。