arendst/Tasmota #2633: '-' subtraction operator use distinguishing sensors conflicts in JSONata query

dweston, 2018-05-05:
- [x] What hardware you are using
nodeMCU
v5.13.0 firmware
nodered v0.18.4

- [x] Provide the output of command ``status 0``
``23:17:56 CMD: status 0
23:17:56 MQT: stat/nodeMcu/STATUS = {"Status":{"Module":18,"FriendlyName":["NodeMcu1","NodeMcu2","NodeMcu3","NodeMcu4"],"Topic":"nodeMcu","ButtonTopic":"0","Power":15,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
23:17:56 MQT: stat/nodeMcu/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:57:26","StartupUTC":"2018-05-05T12:20:30","Sleep":0,"BootCount":20,"SaveCount":197,"SaveAddress":"F8000"}}
23:17:56 MQT: stat/nodeMcu/STATUS2 = {"StatusFWR":{"Version":"5.13.0","BuildDateTime":"2018-05-05T22:18:19","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
23:17:56 MQT: stat/nodeMcu/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxx","xxxxxx"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
23:17:56 MQT: stat/nodeMcu/STATUS4 = {"StatusMEM":{"ProgramSize":496,"Free":504,"Heap":20,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
23:17:56 MQT: stat/nodeMcu/STATUS5 = {"StatusNET":{"Hostname":"nodeMcu","IPAddress":"10.0.1.xx","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":3}}
23:17:56 MQT: stat/nodeMcu/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.14","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_258730","MqttUser":"user","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:17:56 MQT: stat/nodeMcu/STATUS7 = {"StatusTIM":{"UTC":"Sat May 05 13:17:56 2018","Local":"Sat May 05 23:17:56 2018","StartDST":"Sun Oct 07 02:00:00 2018","EndDST":"Sun Apr 01 03:00:00 2018","Timezone":10,"Sunrise":"06:06","Sunset":"17:36"}}
23:17:56 MQT: stat/nodeMcu/STATUS10 = {"StatusSNS":{"Time":"2018-05-05T23:17:56","DS18B20-1":{"Id":"0416A1B9C9FF","Temperature":21.1},"DS18B20-2":{"Id":"0416A4495AFF","Temperature":24.1},"TempUnit":"C"}}
23:17:56 MQT: stat/nodeMcu/STATUS11 = {"StatusSTS":{"Time":"2018-05-05T23:17:56","Uptime":"0T00:57:26","Vcc":2.965,"POWER1":"ON","POWER2":"ON","POWER3":"ON","POWER4":"ON","Wifi":{"AP":1,"SSId":"xxxxxx","RSSI":70,"APMac":"xx:xx:xx:xx:xx:xx"}}}
``
This is more an observation than a bug.

When querying the device with a ``status 10`` command the module  responds with the following:
``{
  "StatusSNS": {
    "Time": "2018-05-05T22:45:30",
    "DS18B20-1": {
      "Id": "0416A1B9C9FF",
      "Temperature": 21.6
    },
    "DS18B20-2": {
      "Id": "0416A4495AFF",
      "Temperature": 20.9
    },
    "TempUnit": "C"
  }
}``
In nodered the '-' subtraction operator is not allowed in context of parsing this payload.
[image]
This can be overcome by using index notation syntax instead:
``payload.StatusSNS["DS18B20-1"].Temperature``
To minimise this confusion could iconsideration be given to using a separator that does not conflict with these JSONata operators:
[](http://docs.jsonata.org/operators.html)

arendst, 2018-05-06:
See #2207 and others.

dweston, 2018-05-07:
I guess that's a no then. No problems.
I'll just have to remember - use index notation when parsing multiple sensors.

closed 2018-05-08 by arendst