arendst/Tasmota #2635: Rule enchansment for value "Switch1":"OFF" / "ON" state check in TELE-

roblad, 2018-05-05:
Make sure these boxes are checked before submitting your issue - Thank you!

-[x] What hardware you are using Wemos D1

Hi,

Could you add a checking of state Switch ON/OFF in TELE in SENSOR string "Switch1":"OFF"

wongnam, 2018-05-13:
I try Rule on Switch1#State=ON do Power1 ON endon, maybe the rule does not effect with Switch#State=ON or Switch#State=1 but it effect with  Switch#State

roblad, 2018-05-13:
Hi,

I have 

rule on Switch#state=ON do publish cmnd/podlewanie/POWER4 1 endon 

and it does not work

arendst, 2018-05-13:
Right. Did some tests and it works but....

- The correct syntax for a switch rule is one of the following:
```
- rule on Switch1#state=on do …
- rule on Switch1#state=off do …
- rule on Switch1#state=toggle do …
```
- When switchmode is 0 (toggle) the trigger for a switch change will only provide a state of ``toggle``
- when switchmode is 1 (follow) the trigger for a switch change will provide either a state of ``on`` or ``off``

So to get what you want (I think) is execute command ``switchmode 1`` once and make a rule like
```
rule on Switch1#state=ON do publish cmnd/podlewanie/POWER4 1 endon
```

arendst, 2018-05-13:
BTW rules currently do NOT work on a tele or status message containing switch.

As switch is in the root of the JSON message the rules code does not handle this correctly.

arendst, 2018-05-13:
Did you try switchmode 2? It works for me as expected.

roblad, 2018-05-13:
let me update firmware :-)

roblad, 2018-05-13:
Hi,

No both switch mode 2 and 1 does not work, even when i disabled domoticz

19:12:29 CMD: rule
19:12:29 MQT: stat/woda/RESULT = {"Rule":"ON","Once":"OFF","Rules":"on Switch1#state=ON do publish cmnd/podlewanie/POWER4 1 endon"}
19:12:36 CMD: switchmode 1
19:12:36 MQT: stat/woda/RESULT = {"SwitchMode1":1}
19:12:45 MQT: cmnd/deszcz/POWER = OFF
19:12:46 MQT: cmnd/deszcz/POWER = ON
19:12:47 MQT: cmnd/deszcz/POWER = OFF
19:12:49 MQT: cmnd/deszcz/POWER = ON

I do not have any device any relay on that wemos d1 module

Schubi1, 2018-06-06:
@roblad 
had the same problem and tried for many hours.
My rule is now working replace Switch1#sate=ON with Switch1#state=1 and Switch1#state=OFF with Switch1#sate=0
Hope it helps!

ascillato2, 2018-07-16:
Hi

Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks

closed 2018-07-16 by ascillato2