arendst/Tasmota #2638: Sonoff Pow does not publish Sensor or Energy via MQTT

stephan1980, 2018-05-06:
Hi

I just flashed the Tasmota 5.12.0i to my new Sonoff Pow.
Flashing worked fine,  the web interface is working properly showing all the desired information.
I configured the MQTT in exact the same way I did with all of my other Sonoff devices but if I listen to my pow on # I only get the topics Pow1/LWT and Pow1/POWER. There is no topic wich shows me the current measurements of the POW.
Via the console I could see the SENSOR Message

`00:58:37 RSL: Pow1/SENSOR = {"Time":"2018-05-06T00:58:37","ENERGY":{"Total":0.237,"Yesterday":0.237,"Today":0.000,"Period":0,"Power":0,"Factor":0.00,"Voltage":231,"Current":0.000}}`

but it wouldn't be published via MQTT. In the example above the water pump I am monitoring was switched off.

Calibration is done.

Is there a way to fix this issue? 

Here my Status 0
`00:52:26 RSL: Pow1/STATUS = {"Status":{"Module":6,"FriendlyName":["SonoffPow1"],"Topic":"SonoffPOW-1","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
00:52:26 RSL: Pow1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:19:04","StartupUTC":"2018-05-05T23:33:22","Sleep":0,"BootCount":13,"SaveCount":50,"SaveAddress":"F4000"}}
00:52:26 RSL: Pow1/STATUS2 = {"StatusFWR":{"Version":"5.12.0i","BuildDateTime":"2018-05-05T22:18:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:52:26 RSL: Pow1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Solaris",""],"TelePeriod":300,"SetOption":["00008029","55818000"]}}
00:52:27 MQT: Pow1/STATUS4 = {"StatusMEM":{"ProgramSize":509,"Free":428,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
00:52:27 RSL: Pow1/STATUS5 = {"StatusNET":{"Hostname":"SonoffPOW-1-7816","IPAddress":"192.168.60.152","Gateway":"192.168.60.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.60.1","Mac":"EC:FA:BC:00:9E:88","Webserver":2,"WifiConfig":4}}
00:52:27 RSL: Pow1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.60.5","MqttPort":1883,"MqttClientMask":"SonoffPOW-1","MqttClient":"SonoffPOW-1","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:52:27 RSL: Pow1/STATUS7 = {"StatusTIM":{"UTC":"Sat May 05 23:52:27 2018","Local":"Sun May 06 00:52:27 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
00:52:27 RSL: Pow1/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:52:27 RSL: Pow1/STATUS10 = {"StatusSNS":{"Time":"2018-05-06T00:52:27","ENERGY":{"Total":0.237,"Yesterday":0.237,"Today":0.000,"Power":1,"Factor":0.08,"Voltage":232,"Current":0.075}}}
00:52:27 RSL: Pow1/STATUS11 = {"StatusSTS":{"Time":"2018-05-06T00:52:27","Uptime":"0T00:19:05","Vcc":3.178,"POWER":"ON","Wifi":{"AP":1,"SSId":"Solaris","RSSI":52,"APMac":"38:10:D5:62:3E:77"}}}`

Frogmore42, 2018-05-06:
It looks like it isn't connected to MQTT at all.  I would turn on more logging and capture the log from power on.

I would also check the log on the MQTT broker.

curzon01, 2018-05-06:
set MQTT client to default "DVES_%06X", do not use - see #2353, #2291

curzon01, 2018-05-06:
Did you use user_config.h to make the changes?

stephan1980, 2018-05-06:
> Did you use user_config.h to make the changes?

Yes. I changed my wifi settings in user_config.h
This worked for me for all other tasmota devices.... Is this not the right way?

curzon01, 2018-05-07:
You can do it this way but be aware changes in user_config.h are not automatically used after flash on a device having previously flashed - see [Device configuration](https://github.com/arendst/Sonoff-Tasmota/wiki/Upload#device-configuration).

I only wanted to be sure your changes are active (MQTT client to default "DVES_%06X") - have a look into `Status 6` result "MqttClient" - this was not clear enough in your previous post.

closed 2018-07-06 by stale[bot]

fkemps, 2018-09-21:
Stephan did you ever solved this issue?
I'm experiencing exactly the same with my Sonoff POW (Sonoff-Tasmota 6.2.1.4).

I want to use values like RSSI, Power withing Openhab.
Do see the messages on console but not via MQTT. Power ON/OFF do show up and work to control the Sonoff.

```
08:50:26 RSL: tele/sonoff-000001/STATE = {"Time":"2018-09-21T08:50:26","Uptime":"0T11:11:05","Vcc":3.516,"POWER":"OFF","Wifi":{"AP":1,"SSId":"4xion","BSSId":"90:6C:AC:F4:B3:79","Channel":7,"RSSI":56}}
08:50:26 RSL: tele/sonoff-000001/SENSOR = {"Time":"2018-09-21T08:50:26","ENERGY":{"Total":0.002,"Yesterday":0.001,"Today":0.001,"Period":0,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
```

status 6
```
08:53:39 RSL: stat/sonoff-000001/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.254.13","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3B9E76","MqttUser":"xxxxxx","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
```

curzon01, 2018-09-21:
@fkemps - if you see MQTT messages on console but not on MQTT server check server ACL for MqttUser xxxxxx - may be your Tasmota MqttUser is allowed to write stat/... msgs but not permitted to write tele/.. messages

fkemps, 2018-09-21:
Hi Curzon01,

I do see the tele/... messages. For example when I perform a restart of the sonoff.

```Javascript
$ mosquitto_sub -h localhost -u xxxuserxxxx -P xxxpwdxxx -t tele/sonoff-000001/+ -v
tele/sonoff-000001/LWT Online
tele/sonoff-000001/LWT Offline
tele/sonoff-000001/LWT Online
tele/sonoff-000001/INFO1 {"Module":"Sonoff Pow","Version":"6.2.1.4","FallbackTopic":"DVES_3B9E76","GroupTopic":"sonoffs"}
tele/sonoff-000001/INFO2 {"WebServerMode":"Admin","Hostname":"sonoff-000001-7798","IPAddress":"192.168.254.xx"}
tele/sonoff-000001/INFO3 {"RestartReason":"Software/System restart"}
```
Both boot and tele msg would use the same credentials, right?!

curzon01, 2018-09-21:
Don't know things about your binary, make a test with a binary from Release page here to exclude issues with your own version

Frogmore42, 2018-09-21:
@fkemps the snippet of the log you provided shows that Tasmota was disconnected from your MQTT broker when it was time to publish the messages you want to see.

There are many possibilities for why this might be.  Most likely it is because you are using a core version that is known to have issues.  To help eliminate that, use one of the pre-compiled release binaries that are known to be good on many people's devices and environments.  There is some troubleshooting information on the wiki, in particular the recommendation to fully erase the flash before uploading new firmware when you have strange issues.  So, please follow the suggestion of @curzon01

fkemps, 2018-09-22:
@Frogmore42 Thanks for the feedback, although not sure how you determined the disconnect with MQTT broker. Can you provide?

To be honest this was the first Sonoff flash, not particular chosen an image release. Wanted to extend my home automation (Openhab) with MQTT.
Need to search the pre-compiled release binaries for OTA upgrading. As you notice I'm new on Tasmota and need to find my way. A pointer to a preferred pre-compiled image is appreciated.

Frogmore42, 2018-09-22:
When it is connected to MQTT, the lines are prefaced by MQT, when not they start with RSL.