arendst/Tasmota #2642: hpa to mmhg

eeak, 2018-05-06:
How can I switch the format of the pressure from hpa to mmhg?

arendst, 2018-05-06:
With your HA tool

eeak, 2018-05-06:
Sorry, what do you mean by the HA tool?

arendst, 2018-05-06:
HA = HomeAutomation tool like Domoticz, Node-Red etc. Any user front-end.

closed 2018-06-04 by ascillato

edgiru, 2019-10-18:
Hello
I have the same question. How to change the format? On screen i2C (SSD1306 OLED 128x64) hpa to mmhg? (In home automation everything is fine there I did mmhg)
[image]

edgiru, 2019-10-18:
16:56:13 MQT: tele/cabinet/SENSOR = {"Time":"2019-10-18T16:56:12","BME280":{"Temperature":23.5,"Humidity":38.8,"Pressure":754.2},"**PressureUnit":"mmHg**","TempUnit":"C"}

arendst, 2019-10-18:
OK. Little bug. Will fix now.