arendst/Tasmota #2643: KNX not responding to messages.

jeylites, 2018-05-06:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

@ascillato 

I tried to simulate [KNX Example 2 from wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/KNX-Features). It seems like Device 1 is able to control the ON/ OFF of Device 2, but not the other way round. 

Compiling with core 2.3.0  ran me into wifi disconnection issues. With 2.4.0, KNX seems responsive, no weird behavior like 2.4.1.     

Following is my current config 

Core: 2.4.0
Tasmota: 5.13.1.a
- wifi sleep mode off

Hardware: Sonoff T1, 3gang 

***Device 1***

`00:00:00 Project jey1 JEY 1 (Topic jey1, Fallback 26, GroupTopic sonoffs) Version 5.13.1a-2_4_0
00:00:00 WIF: Connecting to AP1 SUGAR1 in mode 11N as jey1-4683...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on jey1-4683 with IP address 192.168.0.26
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/jey1/LWT = Online (retained)
00:00:05 MQT: cmnd/jey1/POWER = 
00:00:05 MQT: tele/jey1/INFO1 = {"Module":"Sonoff T1 3CH","Version":"5.13.1a","FallbackTopic":"26","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/jey1/INFO2 = {"WebServerMode":"Admin","Hostname":"jey1-4683","IPAddress":"192.168.0.26"}
00:00:05 MQT: tele/jey1/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
00:00:06 MQT: stat/jey1/POWER1 = OFF
00:00:06 MQT: stat/jey1/RESULT = {"POWER2":"OFF"}
00:00:06 MQT: stat/jey1/POWER2 = OFF
00:00:06 MQT: stat/jey1/RESULT = {"POWER3":"OFF"}
00:00:06 MQT: stat/jey1/POWER3 = OFF
00:00:11 KNX: Relay 1 = 1 sent to 2.2.1
00:00:11 MQT: stat/jey1/RESULT = {"POWER1":"ON"}
00:00:11 MQT: stat/jey1/POWER1 = ON
00:00:13 MQT: tele/jey1/STATE = {"Time":"1970-01-01T00:00:13","Uptime":"0T00:00:13","POWER1":"ON","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR1","RSSI":96,"APMac":"88:D7:F6:02:18:80"}}
00:00:13 KNX: Relay 1 = 1 sent to 2.2.1
00:00:13 MQT: stat/jey1/RESULT = {"POWER1":"ON"}
00:00:13 MQT: stat/jey1/POWER1 = ON
00:00:13 MQT: stat/jey1/RESULT = {"POWER2":"OFF"}
00:00:13 MQT: stat/jey1/POWER2 = OFF
00:00:13 MQT: stat/jey1/RESULT = {"POWER3":"OFF"}
00:00:13 MQT: stat/jey1/POWER3 = OFF
00:00:16 KNX: Relay 1 = 0 sent to 2.2.1
00:00:16 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
00:00:16 MQT: stat/jey1/POWER1 = OFF
00:02:00 MQT: tele/jey1/UPTIME = {"Time":"1970-01-01T00:02:00","Uptime":"0T00:02:00"}
00:02:23 KNX: Relay 1 = 1 sent to 2.2.1
00:02:23 MQT: stat/jey1/RESULT = {"POWER1":"ON"}
00:02:23 MQT: stat/jey1/POWER1 = ON
00:02:27 KNX: Relay 1 = 0 sent to 2.2.1
00:02:27 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
00:02:27 MQT: stat/jey1/POWER1 = OFF
00:05:13 MQT: tele/jey1/STATE = {"Time":"1970-01-01T00:05:13","Uptime":"0T00:05:13","POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR1","RSSI":98,"APMac":"88:D7:F6:02:18:80"}}
00:05:13 KNX: Relay 1 = 0 sent to 2.2.1
00:05:13 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
00:05:13 MQT: stat/jey1/POWER1 = OFF
00:05:13 MQT: stat/jey1/RESULT = {"POWER2":"OFF"}
00:05:13 MQT: stat/jey1/POWER2 = OFF
00:05:13 MQT: stat/jey1/RESULT = {"POWER3":"OFF"}
00:05:13 MQT: stat/jey1/POWER3 = OFF
00:10:00 APP: Serial logging disabled
[... 87 lines cut ...]
00:10:13 MQT: stat/jey2/POWER2 = OFF
00:10:13 MQT: stat/jey2/RESULT = {"POWER3":"OFF"}
00:10:13 MQT: stat/jey2/POWER3 = OFF
00:13:54 KNX: Received from 2.2.1 Command Write: 0 to Relay 1
00:13:54 MQT: stat/jey2/RESULT = {"POWER1":"OFF"}
00:13:54 MQT: stat/jey2/POWER1 = OFF
00:15:13 MQT: tele/jey2/STATE = {"Time":"1970-01-01T00:15:13","Uptime":"0T00:15:13","POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR1","RSSI":100,"APMac":"88:D7:F6:02:18:80"}}
00:15:13 MQT: stat/jey2/RESULT = {"POWER1":"OFF"}
00:15:13 MQT: stat/jey2/POWER1 = OFF
00:15:13 MQT: stat/jey2/RESULT = {"POWER2":"OFF"}
00:15:13 MQT: stat/jey2/POWER2 = OFF
00:15:13 MQT: stat/jey2/RESULT = {"POWER3":"OFF"}
00:15:13 MQT: stat/jey2/POWER3 = OFF
00:16:50 CMD: status 0
00:16:50 MQT: stat/jey2/STATUS = {"Status":{"Module":30,"FriendlyName":["JEY 2","JEY 22","JEY 23"],"Topic":"jey2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:16:50 MQT: stat/jey2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:16:49","StartupUTC":"","Sleep":0,"BootCount":13,"SaveCount":283,"SaveAddress":"F9000"}}
00:16:50 MQT: stat/jey2/STATUS2 = {"StatusFWR":{"Version":"5.13.1a","BuildDateTime":"2018-05-06T15:51:11","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
00:16:50 MQT: stat/jey2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SUGAR1","SUGAR2"],"TelePeriod":300,"SetOption":["02008009","55818000"]}}
00:16:50 MQT: stat/jey2/STATUS4 = {"StatusMEM":{"ProgramSize":469,"Free":532,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
00:16:50 MQT: stat/jey2/STATUS5 = {"StatusNET":{"Hostname":"jey2-2150","IPAddress":"192.168.0.27","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"60:01:94:97:48:66","Webserver":2,"WifiConfig":4}}
00:16:50 MQT: stat/jey2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.200","MqttPort":1883,"MqttClientMask":"27","MqttClient":"27","MqttUser":"admin","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:16:50 MQT: stat/jey2/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:16:50 1970","Local":"Thu Jan 01 00:16:50 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1}}
00:16:50 MQT: stat/jey2/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:16:50"}}
00:16:50 MQT: stat/jey2/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:16:50","Uptime":"0T00:16:49","POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR1","RSSI":98,"APMac":"88:D7:F6:02:18:80"}}}
```

ascillato, 2018-05-06:
Hi,

The device 2 of the wiki example just send a KNX command but only pressing the physical button1.

So, if you are toggling the relay by using MQTT or the web interface, it will not do anything. Device 2 is expecting a press on the button to send the command.

The device 1 is sending the relay1 status, that's why, if a MQTT or Web interface or rule, etc change the relay 1 status, the device 1 will send the KNX Command.

jeylites, 2018-05-06:
Hmm, even pressing the physical button it works one direction.

jeylites, 2018-05-06:
@ascillato 

I apologize, my honest mistake. I think something didn't get set correctly and reconfiguring everything from scratch fixed it. Now both devices work accordingly to my needs.

I assume I can't have the same behavior via MQTT?  

I'm using Homekit which allows grouping of switches as one, which pretty much solves my issues :) but just curious to see if it's possible on the KNX end.

ascillato, 2018-05-06:
By MQTT you can have the same behaviour. Please see example 2 of rules on wiki

By KNX you can have a direct communication device to device 

From the example, you will have the desired behaviour but device to MQTT broker to device. 

But both approaches will work.

ascillato, 2018-05-06:
Happy to see your issues solved :)

jeylites, 2018-05-06:
Rules is the next best feature! Will definitely consider for other things. 
For now, I'm happy with the current setup using KNX as somewhat fail safe if MQTT broker fails on me, not to mention the space issue. As far as MQTT control, a simple grouping of switches on Homekit does the trick for me. 

Thank you!

ascillato, 2018-05-07:
Love your idea of using KNX and MQTT together as one to be the backup of the other. Thanks for sharing.

jeylites, 2018-05-08:
I encounter a problem today where Device 1 was able to turn ON Device 2 but not OFF. 

For Device 1, When the physical button is pressed, I get a sent KNX sent message and receive on Device 2. However when Device 2 is pressed , sent is seen but no receive on Device 1.

All got fix after a hard reset on Device 1.

Will try to get some logs

jeylites, 2018-05-08:
I encounter a problem today where Device 1 was able to turn ON Device 2 but not OFF. 

Issue got fixed after doing a hard reset. 

Device 1

jeylites, 2018-05-08:
***Device 1***
```
05:54:21 WIF: Checking connection...
05:54:21 WIF: Attempting connection...
05:54:22 WIF: Checking connection...
05:54:22 WIF: Attempting connection...
05:54:23 WIF: Checking connection...
05:54:23 WIF: Attempting connection...
05:54:24 WIF: Checking connection...
05:54:24 WIF: Connect failed as AP cannot be reached
05:54:25 WIF: Checking connection...
05:54:25 WIF: Connect failed as AP cannot be reached
05:54:25 WIF: Connecting to AP1 SUGAR1 in mode 11N as jey1-4683...
05:54:26 CFG: Saved to flash at FA, Count 514, Bytes 2560
05:54:26 WIF: Checking connection...
05:54:26 WIF: Attempting connection...
05:54:27 WIF: Checking connection...
05:54:27 WIF: Attempting connection...
05:54:28 RSL: tele/jey1/STATE = {"Time":"1970-01-01T05:54:28","Uptime":"0T05:54:13","POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR1","RSSI":100,"APMac":"88:D7:F6:02:18:80"}}
05:54:28 WIF: Checking connection...
05:54:28 WIF: Connected
05:54:39 MQT: Attempting connection...
05:54:39 MQT: Connected
05:54:39 MQT: tele/jey1/LWT = Online (retained)
05:54:39 MQT: cmnd/jey1/POWER = 
05:54:39 MQT: Subscribe to cmnd/jey1/#
05:54:39 MQT: Subscribe to cmnd/sonoffs/#
05:54:39 MQT: Subscribe to cmnd/26/#
05:54:48 WIF: Checking connection...
05:54:48 WIF: Connected
05:54:55 HTP: Main Menu
05:54:59 KNX: Relay 1 = 1 sent to 2.2.1
05:54:59 MQT: stat/jey1/RESULT = {"POWER1":"ON"}
05:54:59 MQT: stat/jey1/POWER1 = ON
05:54:59 CFG: Saved to flash at F9, Count 515, Bytes 2560
05:55:02 KNX: Relay 1 = 0 sent to 2.2.1
05:55:02 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
05:55:02 MQT: stat/jey1/POWER1 = OFF
05:55:03 CFG: Saved to flash at F8, Count 516, Bytes 2560
05:55:08 WIF: Checking connection...
05:55:08 WIF: Connected
05:55:28 MQT: tele/jey1/STATE = {"Time":"1970-01-01T05:55:28","Uptime":"0T05:55:13","POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR1","RSSI":100,"APMac":"88:D7:F6:02:18:80"}}
05:55:28 RSL: Received Topic cmnd/jey1/POWER1, Data Size 3, Data OFF
05:55:28 RSL: Group 0, Index 1, Command POWER, Data OFF
05:55:28 KNX: Relay 1 = 0 sent to 2.2.1
05:55:28 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
05:55:28 MQT: stat/jey1/POWER1 = OFF
05:55:28 RSL: Received Topic cmnd/jey1/POWER2, Data Size 3, Data OFF
05:55:28 RSL: Group 0, Index 2, Command POWER, Data OFF
05:55:28 MQT: stat/jey1/RESULT = {"POWER2":"OFF"}
05:55:28 MQT: stat/jey1/POWER2 = OFF
05:55:28 RSL: Received Topic cmnd/jey1/POWER3, Data Size 3, Data OFF
05:55:28 RSL: Group 0, Index 3, Command POWER, Data OFF
05:55:28 MQT: stat/jey1/RESULT = {"POWER3":"OFF"}
05:55:28 MQT: stat/jey1/POWER3 = OFF
05:55:28 WIF: Checking connection...
05:55:28 WIF: Connected
05:55:48 WIF: Checking connection...
05:55:48 WIF: Connected
05:56:08 WIF: Checking connection...
05:56:08 WIF: Connected
05:56:28 MQT: tele/jey1/STATE = {"Time":"1970-01-01T05:56:28","Uptime":"0T05:56:13","POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR1","RSSI":100,"APMac":"88:D7:F6:02:18:80"}}
05:56:28 RSL: Received Topic cmnd/jey1/POWER1, Data Size 3, Data OFF
05:56:28 RSL: Group 0, Index 1, Command POWER, Data OFF
05:56:28 KNX: Relay 1 = 0 sent to 2.2.1
05:56:28 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
05:56:28 MQT: stat/jey1/POWER1 = OFF
05:56:28 RSL: Received Topic cmnd/jey1/POWER2, Data Size 3, Data OFF
05:56:28 RSL: Group 0, Index 2, Command POWER, Data OFF
05:56:28 MQT: stat/jey1/RESULT = {"POWER2":"OFF"}
[... 182 lines cut ...]
00:01:40 MQT: stat/jey2/RESULT = {"POWER1":"OFF"}
00:01:40 MQT: stat/jey2/POWER1 = OFF
00:01:41 CFG: Saved to flash at F4, Count 752, Bytes 2560
00:01:45 KNX: Received from 2.2.1 Command Write: 1 to Relay 1
00:01:45 MQT: stat/jey2/RESULT = {"POWER1":"ON"}
00:01:45 MQT: stat/jey2/POWER1 = ON
00:01:46 CFG: Saved to flash at FB, Count 753, Bytes 2560
00:01:46 KNX: Received from 2.2.1 Command Write: 0 to Relay 1
00:01:46 MQT: stat/jey2/RESULT = {"POWER1":"OFF"}
00:01:46 MQT: stat/jey2/POWER1 = OFF
00:01:47 CFG: Saved to flash at FA, Count 754, Bytes 2560
00:01:52 WIF: Checking connection...
00:01:52 WIF: Connected
00:01:53 APP: Button1 multi-press 1
00:01:53 KNX: Button 1 = 1 sent to 2.2.2
00:01:53 MQT: stat/jey2/RESULT = {"POWER1":"ON"}
00:01:53 MQT: stat/jey2/POWER1 = ON
00:01:54 CFG: Saved to flash at F9, Count 755, Bytes 2560
00:01:55 APP: Button1 multi-press 1
00:01:56 KNX: Button 1 = 1 sent to 2.2.2
00:01:56 MQT: stat/jey2/RESULT = {"POWER1":"OFF"}
00:01:56 MQT: stat/jey2/POWER1 = OFF
00:01:56 CFG: Saved to flash at F8, Count 756, Bytes 2560
00:02:00 MQT: tele/jey2/UPTIME = {"Time":"1970-01-01T00:02:00","Uptime":"0T00:01:58"}
```

jeylites, 2018-05-08:
What is the cause of the broker turning off ?

May a retained message on your MQTT Broker? - What do you mean?

ascillato, 2018-05-08:
Depending on your broker, you can set to retain messages and be re sended.
You should check the MQTT Broker configuration.

jeylites, 2018-05-08:
`{
   "bridge":{
      "name":"Homeroom",
      "username":"",
      "port":51820,
      "pin":"031-45-150"
   },
   "description":"",
   "platforms":[

   ],
   "accessories":[
      {
         "accessory":"mqtt-switch-tasmota",
         "name":" Lamp",
         "url":"mqtt://192.168.0.200",
         "username":"admin",
         "password":"admin",
         "topics":{
            "statusGet":"stat/jey1/POWER1",
            "statusSet":"cmnd/jey1/POWER1",
            "stateGet":"tele/jey1/STATE"
         },
         "onValue":"ON",
         "offValue":"OFF",
         "activityTopic":"tele/jey1/LWT",
         "activityParameter":"Online",
         "startCmd":"cmnd/jey1/TelePeriod",
         "startParameter":"60"
      },
      {
         "accessory":"mqtt-switch-tasmota",
         "name":"LAMP 2",
         "url":"mqtt://192.168.0.200",
         "username":"admin",
         "password":"admin",
         "topics":{
            "statusGet":"stat/jey2/POWER2",
            "statusSet":"cmnd/jey2/POWER2",
            "stateGet":"tele/jey2/STATE"
         },
         "onValue":"ON",
         "offValue":"OFF",
         "activityTopic":"tele/jey2/LWT",
         "activityParameter":"Online",
         "startCmd":"cmnd/jey2/TelePeriod",
         "startParameter":"60"`

jeylites, 2018-05-08:
further testing seems like  Device 1 is able to "ON/OFF"  Device 2 via the
physical button.  However, Device 2 is not able to do the same to Device 1
after a while of running. Only a hard reset fixes the issue.

Pulling up console I see Device 2 is sending a message but Device 1 is not
receiving. And the retain kicks in depending the "ON/OFF" state of Device
1.

Assuming I have Device 1 & Device 2 on "ON" state and I intentionally turn
OFF Device 2, it will turn "OFF" but Device 1 remains "ON". A few seconds
later, the retain on Device 1 kicks in to turn "ON" Device 2

Is it possible to set both devices to be responders if that makes any
sense...

*Jey Kasi*

ascillato, 2018-05-09:
Hi,

So,

1- Only device 1 is not receiving after sometime of reboot?

2- If you try to send telegrams several times, device 1 stills no respond to any telegram?

ascillato, 2018-05-09:
3- can you post the knx configuration of device 1?

jeylites, 2018-05-09:
1- Only device 1 is not receiving after sometime of reboot?

**Yes, only device 1 is not receiving after a while** 

2- If you try to send telegrams several times, device 1 stills no respond to any telegram?

**Not at all, it only works after a reboot**

3- can you post the knx configuration of device 1?

I have included both Device 1 & Device 2, Previously my configuration was same as your Stairs light example (2)  on wiki. Had the same problem of Device 1 not receiving so I change it to the below to see if it made any difference.... guess not :(

[image]

[image]

ascillato, 2018-05-09:
mmmm,

Seems to be similar to [this issue](https://github.com/ascillato/Sonoff-Tasmota_KNX/issues/82)

I could not make my devices do that behaviour. That's why I'm looking for the difference on your devices. The key to the problem should be there.

I noticed that the firmware build time for both devices is different. The firmware for device 1 and 2 is the same?

ascillato, 2018-05-09:
> Yes same problem
> A second device also have the issue with the same config than the device
> They both compiled the same time also I have set up a second device
> to make sure... the same problem
> Core 2.40 lwIP 1.4 Prebuilt
> Compiled with Arduino 1.8.5 OSX
> Sent from Iphone Mobile

Ok. But device 2 is working ok?

jeylites, 2018-05-09:
Both Device New and Old  that I have setup is going through the same issue
where communication goes one direction. I just flashed another board with
platform io. Will report back

jeylites, 2018-05-09:
I have made some progress since 11 hours ago.

I manage to flash a T1 touch with Platform io , Core: 2.4.1, Tasmota 5.13.1a..... all of a sudden the problem I had in the past with Arduino IDE has gone away. 

The bizarre part us, the one direction communication based on Core 2.40 fixed it self after including the new 2.4.1 in the network.  To sum up currently have Core 2.4.1 & Core 2.40 on the same address communicating with each other without any issues. 

I'm new to Platform Io and will like to get an input on the following **Platform.ini file**, the uncommented, are they correct? 

I uncommented `platform = espressif8266@1.5.0   ;v2.3.0` but it's keep defaulting to `;platform = espressif8266@1.7.0  ; v2.4.1` not sure why. 

What is the difference between uncommenting `env_default = sonoff` & `env_default = sonoff-knx` & in user.config file? 

Because I realise if I comment `env_default = sonoff-knx` and uncomment in user.config file KNX doesn't take effect.

```
[platformio]
src_dir = sonoff

; *** Uncomment one of the lines below to build/upload only one environment
env_default = sonoff
;env_default = sonoff-minimal
;env_default = sonoff-classic
env_default = sonoff-knx
;env_default = sonoff-allsensors
;env_default = sonoff-BG
;env_default = sonoff-CN
;env_default = sonoff-CZ
;env_default = sonoff-DE
;env_default = sonoff-ES
;env_default = sonoff-FR
;env_default = sonoff-GR
;env_default = sonoff-HU
;env_default = sonoff-IT
;env_default = sonoff-NL
;env_default = sonoff-PL
;env_default = sonoff-PT
;env_default = sonoff-RU
;env_default = sonoff-TW

[common] ; ************************************************************
; *** Esp8266 core for Arduino version
platform = espressif8266@1.5.0   ;v2.3.0
;platform = espressif8266@1.6.0  ; v2.4.0
;platform = espressif8266@1.7.0  ; v2.4.1
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
platform = espressif8266

framework = arduino
board = esp01_1m
board_flash_mode = dout

; *** Fix espressif8266@1.7.0 induced undesired all warnings
build_unflags = -Wall

build_flags =
  -Wl,-Tesp8266.flash.1m0.ld
  -DVTABLES_IN_FLASH
  -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
;  -DUSE_CONFIG_OVERRIDE

; *** Fix Esp/Arduino core 2.4.x induced Tasmota unused floating point includes
extra_scripts = pio/strip-floats.py

; *** Serial Monitor options
monitor_baud = 115200

; *** Upload Serial reset method for Wemos and NodeMCU
;upload_speed = 115200
upload_speed = 115200
upload_resetmethod = nodemcu
[... 216 lines cut ...]

[env:sonoff-RU]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board}
board_flash_mode = ${common.board_flash_mode}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags} -DMY_LANGUAGE=ru-RU
monitor_baud = ${common.monitor_baud}
upload_speed = ${common.upload_speed}
upload_port = ${common.upload_port}
extra_scripts = ${common.extra_scripts}

[env:sonoff-TW]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board}
board_flash_mode = ${common.board_flash_mode}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags} -DMY_LANGUAGE=zh-TW
monitor_baud = ${common.monitor_baud}
upload_speed = ${common.upload_speed}
upload_port = ${common.upload_port}
extra_scripts = ${common.extra_scripts}
```

ascillato, 2018-05-09:
about the _platformio.ini_:

`env_default = sonoff` <--- you should uncomment just one `env_default`. Uncomment 2 or more make platformio just compile the last option.

if you have all commented will make platformio build all the versions.

```
[common] ; ************************************************************
; *** Esp8266 core for Arduino version
platform = espressif8266@1.5.0   ;v2.3.0
;platform = espressif8266@1.6.0  ; v2.4.0
;platform = espressif8266@1.7.0  ; v2.4.1
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
platform = espressif8266
```
the same than above. just uncomment one platform type. there you have 3 different platforms. platformio is using just the last one that (`platform = espressif8266`) that means use the espressif8266 of higher version previously downloaded.

ascillato, 2018-05-09:
> What is the difference between uncommenting env_default = sonoff & env_default = sonoff-knx & in user_config.h?

when you uncomment for example:
`env_default = sonoff-DE`

platformio will compile just:

```
[env:sonoff-DE]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board}
board_flash_mode = ${common.board_flash_mode}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags} -DMY_LANGUAGE=de-DE
```

and in the `build_flags` it adds `MY_LANGUAGE=de-DE` that it is the same of uncommenting `MY_LANGUAGE=de-DE` on the _user_config.h_

ascillato, 2018-05-09:
> I manage to flash a T1 touch with Platform io , Core: 2.4.1, Tasmota 5.13.1a..... all of a sudden the problem I had in the past with Arduino IDE has gone away.

using the exact same device configuration?

jeylites, 2018-05-09:
Thanks  for the explanation! To sum up I have to use `env_default = sonoff`

for the Core, you mentioned 3 different platforms. but the list has 5 platform options?  assuming I need to pick just Core 2.3.0, is the following correct?

> common] ; ************************************************************
> ; *** Esp8266 core for Arduino version
> platform = espressif8266@1.5.0   ;v2.3.0
> ;platform = espressif8266@1.6.0  ; v2.4.0 [...]

jeylites, 2018-05-09:
> I manage to flash a T1 touch with Platform io , Core: 2.4.1, Tasmota 5.13.1a..... all of a sudden the problem I had in the past with Arduino IDE has gone away.
using the exact same device configuration?

Yes, the `user_config.h` is the same with the rest. The only different was I used Platform io with the about` Platform.ini` file. As you can see I uncommented a few things because I wasn't sure how to work the settings. Some how it compiled without any errors.

ascillato, 2018-05-09:
> for the Core, you mentioned 3 different platforms. but the list has 5 platform options? assuming I need to pick just Core 2.3.0, is the following correct?

change to this:

```
common] ; ************************************************************
; *** Esp8266 core for Arduino version
platform = espressif8266@1.5.0 ;v2.3.0
;platform = espressif8266@1.6.0 ; v2.4.0
;platform = espressif8266@1.7.0 ; v2.4.1
;platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
;platform = espressif8266
```

comment everything you dont need and just chose one platform like above (`platform = espressif8266@1.5.0 ;v2.3.0`)

ascillato, 2018-05-09:
those errors happened also to me.

just delete .pioenv folder inside your sonoff project folder

and also do a rebuild c/c++

[image]

jeylites, 2018-05-09:
I have been going through every folder looking for the pioenvs file or folder, are you referring to the **pio** folder?

ascillato, 2018-05-09:
the .pioenvs folder is made by platformio inside the sonoff-tasmota folder

ascillato, 2018-05-09:
So, this issue get solved for you when you recompiled the firmware right?

On the first boot, did you loose the configuration?

jeylites, 2018-05-09:
the folder doesn't seem to exists in my  sonoff-tasmota folder so replaced it with a clean version on 5.13.1a and it took it. Currently testing Core 2.30, Core 2.4.0,Core 2.4.1

Will let you now you know soon. 

It seem to be an ongoing issue, at anytime do you get access to web browser? I realise using core 2.4.0  & 2.4.1 after a while web server become unresponsive. Found out that a power cycle of sonoff or reboot of router only brings it back.... The cycle continues. 

Another thing by accident I found that access is granted if MQTT broker is done after a while. My best guess will be traffic, the esp is unable to keep up.

ascillato, 2018-05-09:
Hi

I think I found the problem ! :smile: 

I could reproduce this issue. 

If the sonoff device get disconnected from the wifi router, when reconnecting to the same router it loose the multicast connection for receiving packets. That's why a reboot solve the issue because on boot it opens again the UDP.

I will make a PR to solve this.

So, The issue was not related to esp lib versions, configurations nor compilation.

For the disconnection tests, I just unplug my wifi router and plug it again.

So, then with the patched version you should not have problems, but you should look why you have those micro-disconnections from your router ( may be too many wifi devices ? )

There is a very interesting and technical video about wifi routers installation by @davidelang at https://www.usenix.org/conference/lisa12/technical-sessions/presentation/lang_david_wireless

Thanks a lot again for all the tests you made. Thanks to all your help was possible to be found the root of the problem. +1

jeylites, 2018-05-10:
@ascillato 

That explain a lot why KNX works on all Core modes.  Just saw the push, will test it out soon. Thanks for sharing @davidelang video on wireless router. Very informative!!!

ascillato, 2018-05-10:
I close the PR because it is not ready yet.

jeylites, 2018-05-11:
Nice will test it later.

jeylites, 2018-05-13:
Just tested the latest release 5.13.1a downloaded yesterday. Seem the KNX devices are not receiving messages after running a while. Please see below  logs . 

**Device 1**

```
09:25:19 MQT: stat/jey1/POWER2 = OFF
09:25:19 MQT: stat/jey1/RESULT = {"POWER3":"OFF"}
09:25:19 MQT: stat/jey1/POWER3 = OFF
09:30:18 MQT: tele/jey1/STATE = {"Time":"1970-01-01T09:30:18","Uptime":"0T09:30:14","POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR","RSSI":96,"APMac":"88:D7:F6:02:18:80"}}
09:30:18 KNX: Relay 1 = 0 sent to 2.2.1
09:30:18 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
09:30:18 MQT: stat/jey1/POWER1 = OFF
09:30:18 MQT: stat/jey1/RESULT = {"POWER2":"OFF"}
09:30:18 MQT: stat/jey1/POWER2 = OFF
09:30:18 MQT: stat/jey1/RESULT = {"POWER3":"OFF"}
09:30:18 MQT: stat/jey1/POWER3 = OFF
09:35:18 MQT: tele/jey1/STATE = {"Time":"1970-01-01T09:35:18","Uptime":"0T09:35:14","POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR","RSSI":96,"APMac":"88:D7:F6:02:18:80"}}
09:35:18 KNX: Relay 1 = 0 sent to 2.2.1
09:35:18 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
09:35:18 MQT: stat/jey1/POWER1 = OFF
09:35:18 MQT: stat/jey1/RESULT = {"POWER2":"OFF"}
09:35:18 MQT: stat/jey1/POWER2 = OFF
09:35:18 MQT: stat/jey1/RESULT = {"POWER3":"OFF"}
09:35:18 MQT: stat/jey1/POWER3 = OFF
09:40:18 MQT: tele/jey1/STATE = {"Time":"1970-01-01T09:40:18","Uptime":"0T09:40:14","POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR","RSSI":92,"APMac":"88:D7:F6:02:18:80"}}
09:40:18 KNX: Relay 1 = 0 sent to 2.2.1
09:40:18 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
09:40:18 MQT: stat/jey1/POWER1 = OFF
09:40:18 MQT: stat/jey1/RESULT = {"POWER2":"OFF"}
09:40:18 MQT: stat/jey1/POWER2 = OFF
09:40:18 MQT: stat/jey1/RESULT = {"POWER3":"OFF"}
09:40:18 MQT: stat/jey1/POWER3 = OFF
09:40:45 KNX: Relay 1 = 1 sent to 2.2.1
09:40:45 MQT: stat/jey1/RESULT = {"POWER1":"ON"}
09:40:45 MQT: stat/jey1/POWER1 = ON
09:40:47 KNX: Relay 1 = 0 sent to 2.2.1
09:40:47 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
09:40:47 MQT: stat/jey1/POWER1 = OFF
09:40:48 KNX: Relay 1 = 1 sent to 2.2.1
09:40:48 MQT: stat/jey1/RESULT = {"POWER1":"ON"}
09:40:48 MQT: stat/jey1/POWER1 = ON
09:40:50 KNX: Relay 1 = 0 sent to 2.2.1
09:40:50 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
09:40:50 MQT: stat/jey1/POWER1 = OFF
09:40:52 KNX: Relay 1 = 1 sent to 2.2.1
09:40:52 MQT: stat/jey1/RESULT = {"POWER1":"ON"}
09:40:52 MQT: stat/jey1/POWER1 = ON
09:40:56 KNX: Relay 1 = 0 sent to 2.2.1
09:40:56 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
09:40:56 MQT: stat/jey1/POWER1 = OFF
09:40:59 KNX: Relay 1 = 1 sent to 2.2.1
09:40:59 MQT: stat/jey1/RESULT = {"POWER1":"ON"}
09:40:59 MQT: stat/jey1/POWER1 = ON
09:41:00 KNX: Relay 1 = 0 sent to 2.2.1
09:41:00 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
09:41:00 MQT: stat/jey1/POWER1 = OFF
09:41:33 KNX: Relay 1 = 1 sent to 2.2.1
09:41:33 MQT: stat/jey1/RESULT = {"POWER1":"ON"}
09:41:33 MQT: stat/jey1/POWER1 = ON
09:41:43 KNX: Relay 1 = 0 sent to 2.2.1
09:41:43 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
09:41:43 MQT: stat/jey1/POWER1 = OFF
09:41:44 KNX: Relay 1 = 1 sent to 2.2.1
09:41:44 MQT: stat/jey1/RESULT = {"POWER1":"ON"}
09:41:44 MQT: stat/jey1/POWER1 = ON
09:41:45 KNX: Relay 1 = 0 sent to 2.2.1
09:41:45 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
09:41:45 MQT: stat/jey1/POWER1 = OFF
09:42:59 KNX: Relay 1 = 1 sent to 2.2.1
09:42:59 MQT: stat/jey1/RESULT = {"POWER1":"ON"}
[... 46 lines cut ...]
09:35:52 MQT: tele/jey2/STATE = {"Time":"1970-01-01T09:35:52","Uptime":"0T09:35:14","POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR","RSSI":94,"APMac":"88:D7:F6:02:18:80"}}
09:39:07 MQT: stat/jey2/RESULT = {"POWER1":"ON"}
09:39:07 MQT: stat/jey2/POWER1 = ON
09:39:08 MQT: stat/jey2/RESULT = {"POWER1":"OFF"}
09:39:08 MQT: stat/jey2/POWER1 = OFF
09:40:52 MQT: tele/jey2/STATE = {"Time":"1970-01-01T09:40:52","Uptime":"0T09:40:14","POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR","RSSI":94,"APMac":"88:D7:F6:02:18:80"}}
09:41:21 KNX: Button 1 = 1 sent to 2.2.2
09:41:21 MQT: stat/jey2/RESULT = {"POWER1":"ON"}
09:41:21 MQT: stat/jey2/POWER1 = ON
09:41:22 KNX: Button 1 = 1 sent to 2.2.2
09:41:22 MQT: stat/jey2/RESULT = {"POWER1":"OFF"}
09:41:22 MQT: stat/jey2/POWER1 = OFF
09:45:03 CMD: status 0
09:45:03 MQT: stat/jey2/STATUS = {"Status":{"Module":30,"FriendlyName":["JEY 2","JEY 22","JEY 23"],"Topic":"jey2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
09:45:03 MQT: stat/jey2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T09:44:17","StartupUTC":"","Sleep":0,"BootCount":5,"SaveCount":39,"SaveAddress":"F5000"}}
09:45:03 MQT: stat/jey2/STATUS2 = {"StatusFWR":{"Version":"5.13.1a","BuildDateTime":"2018-05-12T22:45:39","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
09:45:03 MQT: stat/jey2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SUGAR","SALT"],"TelePeriod":300,"SetOption":["02008009","55818000"]}}
09:45:03 MQT: stat/jey2/STATUS4 = {"StatusMEM":{"ProgramSize":466,"Free":536,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
09:45:03 MQT: stat/jey2/STATUS5 = {"StatusNET":{"Hostname":"jey2-2150","IPAddress":"192.168.0.27","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"60:01:94:97:48:66","Webserver" [... cut ...] }
09:45:03 MQT: stat/jey2/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T09:45:03"}}
09:45:03 MQT: stat/jey2/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T09:45:03","Uptime":"0T09:44:17","POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR","RSSI":94,"APMac":"88:D7:F6:02:18:80"}}}
09:46:00 MQT: tele/jey2/STATE = {"Time":"1970-01-01T09:46:00","Uptime":"0T09:45:14","POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR","RSSI":92,"APMac":"88:D7:F6:02:18:80"}}
```

ascillato, 2018-05-13:
Hi,

1- This is happening to both devices?

2- Are you using a wifi repeater?

3- Can you access to your wifi router configuration to see:
    - If multicast is turned on ?
    - IGMP is v3 ?
    - Multicast Snoop is off ?

jeylites, 2018-05-13:
1- This is happening to both devices?
Yes, both devices. Usually with a simple reset it fixes it self today. I had to reset the router for it to take effect. Still investigating this part.

2- Are you using a wifi repeater?
The sonoff touch are connected to the Asus router setup as AP. The main router is connected via cable to the Asus AP.

3- Can you access to your wifi router configuration to see: I just drop the RTS value to 2304 from 2346

[image]

ascillato, 2018-05-13:
Test if the issue persists changing **Enable IGMP Snooping** to **disable**

jeylites, 2018-05-13:
will do.

ascillato, 2018-05-13:
Also, please check the config on the main router as your AP is not isolated, meaning that your main router is the one that manages everything.

jeylites, 2018-05-14:
Just an update. I took your tip about moving the KNX Sonoff stuff to the main router. Everything started working right after, also solved few other issue - NTP server sync, Web browser became more reachable and KNX were receiving messages.

The puzzling yet disappointing part is the stock ISP router seem far more reliable than the mid range Asus RT- AC55UHP that I have. 

I also did some research on the Asus router, seems to be a common issue with multicast. It dropout after awhile and the only way to get it back is to do a power cycle. 

Anyway, all systems been running for 15 hours without any major issue.

closed 2018-05-14 by jeylites

ascillato, 2018-05-14:
Wow. Very good news!
Happy to hear that you found the problem :)

jeylites, 2018-05-15:
I've some networking question to ask.

SSID
I'm planing to have the following - A Main AC Router and two routers setup as AC AP MODE connection to main via LAN.  Should I connect all Sonoff devices to the AP or  both Main and AP 

Is it advisable to have two different SSID names  Tasmota device STA_SSID1 &  STA_SSID2 or can they be the same?

What is the best wifi tool to use. Currently have WIFI_RETRY                     
(WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT)

Also planing to get a new AP router, what do you have installed perhaps I could look for something similar?

ascillato, 2018-05-15:
> Should I connect all Sonoff devices to the AP or both Main and AP ?

Really, I don't know what should be the best. You should try. The only advise is (as explained on the davidelang video) be aware of the _hidden transmitters_, so try that all sonoff that are connected to a specific router, see each other.

> Is it advisable to have two different SSID names Tasmota device STA_SSID1 & STA_SSID2 or can they be the same?

Can be the same. No problem with that. I have just one wifi router by now, so I put both SSID the same. So if it disconnects from the wifi will try again.

> What is the best wifi tool to use?

That depends on what you want to do. I also use WIFI_RETRY so when it disconnects from the wifi it will immediately try to connect to the next SSID - that in my case is the same network. 

> what do you have installed ?

By now, my setup is just one TP-LINK Wifi Router provided by my ISP. Everything is connected there.
I'm planning to have another just to avoid the _hidden transmitter_ issue.

jeylites, 2018-05-15:
Thank you. I will try setting this up and see where it goes. :)