arendst/Tasmota #2645: SwitchTopic with sonoff touch

spad, 2018-05-06:
Hi,
I set up my sonoff touch with:

`Topic: libreria_s1l1`
`FullTopic: tasmota/%topic%/%prefix%`
`SwitchMode 0`
`SwitchTopic 1`

As described at [here](https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic), the module is supposed to send an mqtt cmd, instead I receive:

`tasmota/libreria_s1l1/stat/RESULT {"POWER":"ON"}`
`tasmota/libreria_s1l1/stat/POWER ON`

I tried every combination of SwitchMode and SwitchTopic, with the same result. There is something else I have to know?

spad, 2018-05-06:
Thanks @arendst.
My goal is to command one bulb by two different sonoff. At the beginning I used GroupTopic and ButtonTopic, and it worked well.

Unfortunately you cannot use this way when you are using 2 gangs devices (1 gang for the bulb and the other for something else) just because GroupTopic and ButtonTopic operate for the entire device.

So I decided to write a rule to simulate a "diverter". 
What is the best way to do that using openhab and mosquitto?

spad, 2018-05-08:
Hi @Jason2866 ,
I would like to replace light switches in my house by sonoff modules. In a common situation we have two or more switches for a single buld (i.e. in the living room I have 3 different switches). With classic bTicino switches we use diverters.

Using 2 sonoff Touch, I can use ButtonTopic and GroupTopic to force two or more sonoff modules to command the same bulb; but again, in a common situation we need 2 or 3 gangs modules where 1 gang commands a lamp and the other commands a bulb.
So I cannot use ButtonTopic/GroupTopic just because it operates for all gangs in my sonoff device (T2 or T3).

So the solution is to use rules to keep in sync different switches.
My idea was to write one rule for each topic, wiring a switch to the bulb and considering the other as a logical switch for the same bulb:
when one of the switches change status, I'll send a toggle power command to the other one.

Can this do the job? It's the correct way to do that?
Yhanks

Jason2866, 2018-05-08:
There is not just the one and only correct way... 
You can do it as you described. The rule is simple just toggle all other switches in the group if one changes the status from off to on or from on to off. It is overkill if you can change wiring and change switches. Just use momentary wall switches connect them all parallel to one free GPIO from the sonoff. With that you can switch on and off from every switch and need no HA system and just one sonoff.
So it depends what infrastructure do you have and what can be changed or not...

spad, 2018-05-21:
Hi @Jason2866 ,
I followed you suggestion using a Sonoff Basic and using a physical switch on GPIO 14. 
I can switch on and off the relay. The only issue I found is to keep in sync the physical switch. I mean:

1) I switch on the relay by the physical switch
2) I switch off it by software
Now when I press the physical switch again I would like the relay was turned on again. I think I can do that using SwitchTopic 1 and a Rule in openHab. It would be great an "SwitchMode 5" that simply toggle the relay when the GPIO change status.

Jason2866, 2018-05-21:
Hi @spad 
All you want is possible. Read https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic
Thats the reason i suggest a momentary switch. Use switchmode3 0

spad, 2018-05-21:
Yes, I already read that. And no, I dont want to use momentary switch: I
already have switches in my house and I cannot replace all of them. So I
think I will use SwitchMode 1 e SwitchTopic 1, then I'll write a rule to
toggle the relay when GPIO14 changes status.
Thanks.

spad, 2018-05-28:
I just read the code and found switchmode 7!!!!
it's what I was looking for I guess.

Maybe https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic must to be updated :)

ascillato, 2018-05-28:
Hi

Happy to know that is now working for you. Please, close this issue. Thanks :+1:

closed 2018-05-28 by spad