arendst/Tasmota #2646: Unable to compile v5.13.1

johny-mnemonic, 2018-05-06:
I wanted to update to new Tasmota, but I ended while trying to compile it:-(

My compile is crashing like this:

```
Compiling .pioenvs\sonoff\libb91\NeoPixelBus-2.2.9\internal\NeoEsp8266UartMethod.cpp.o
D:/Downloads/Sonoff-Tasmota-development/sonoff/xdrv_08_serial_bridge.ino: In function 'void SerialBridgeInit()':
D:/Downloads/Sonoff-Tasmota-development/sonoff/xdrv_08_serial_bridge.ino:75:60: error: no matching function for call to 'TasmotaSerial::begin(int)'
if (SerialBridgeSerial->begin(Settings.sbaudrate * 1200)) {  // Baud rate is stored div 1200 so it fits into one byte
^
D:/Downloads/Sonoff-Tasmota-development/sonoff/xdrv_08_serial_bridge.ino:75:60: note: candidate is:
In file included from D:/Downloads/Sonoff-Tasmota-development/sonoff/xdrv_03_energy.ino:413:0:
lib\TasmotaSerial-1.0.1\src/TasmotaSerial.h:37:10: note: bool TasmotaSerial::begin()
bool begin();
^
lib\TasmotaSerial-1.0.1\src/TasmotaSerial.h:37:10: note:   candidate expects 0 arguments, 1 provided
D:/Downloads/Sonoff-Tasmota-development/sonoff/xdrv_08_serial_bridge.ino: In function 'boolean SerialBridgeCommand()':
D:/Downloads/Sonoff-Tasmota-development/sonoff/xdrv_08_serial_bridge.ino:113:58: error: no matching function for call to 'TasmotaSerial::begin(int)'
SerialBridgeSerial->begin(Settings.sbaudrate * 1200);  // Reinitialize serial port with new baud rate
^
D:/Downloads/Sonoff-Tasmota-development/sonoff/xdrv_08_serial_bridge.ino:113:58: note: candidate is:
In file included from D:/Downloads/Sonoff-Tasmota-development/sonoff/xdrv_03_energy.ino:413:0:
lib\TasmotaSerial-1.0.1\src/TasmotaSerial.h:37:10: note: bool TasmotaSerial::begin()
bool begin();
^
lib\TasmotaSerial-1.0.1\src/TasmotaSerial.h:37:10: note:   candidate expects 0 arguments, 1 provided
Compiling .pioenvs\sonoff\libb91\NeoPixelBus-2.2.9\internal\NeoGamma.cpp.o
Compiling .pioenvs\sonoff\libb91\NeoPixelBus-2.2.9\internal\NeoPixelAnimator.cpp.o
*** [.pioenvs\sonoff\src\sonoff.ino.cpp.o] Error 1
=========================================================== [ERROR] Took 3.95 seconds ===========================================================
```

I tried disabling my settings override and also disabling USE_WS2812 to disable compiling of NeoPixel, but neither helped.

I am using VS Code with Platform.IO on Windows 10. Tasmota v5.12.0 compiled without any issue when I last tried it.

ascillato, 2018-06-02:
Hi, could you manage to compile. Your errors seems that you have a different directory structure than Tasmota. You should keep as it is in this Repository.

johny-mnemonic, 2018-06-02:
@ascillato I am sorry, but I am not sure what you mean by that. I am not aware of changing the directory structure in any way. I just downloaded the repo as a zip file, unpacked and edited config.
Could you please be more specific?