arendst/Tasmota #2649: basic r2 doesnt connect to wifi when on ac power

djiceman, 2018-05-06:
Hi

Ive received a batch of basic's r2, after flashing(env_default = sonoff), they work perfectly on the bench(usb power), but once they are powered by AC, they no longer connect to the wifi. 

Also, while on AC, the serial output is garbage with a few readable lines every now and then. So debugging is practically useless. Ive seen this a few times - "Unable to connect to AP".

Theres no additional capacitor on my models.
Ive proceeded to flash with the minimal build and the devices work as expected when on AC power.

Ive then flashed the normal build via the web upgrade, AC results in no wifi connection.
Flashed again with minimal and wifi works when on AC power.

Any ideas as what what causes this and how do I resolve?
[image]

djiceman, 2018-05-06:
Ive resolved the wifi connection on AC issue by disabling all the sensors, timers, sunrise and rules features.

w4tsn, 2018-05-07:
My Basic R2 does not only have no connection to Wifi when on AC, but also does not boot at all. It does not light up the LED on start nor respond to any button commands.

I've tried to use the minimal firmware to no success.

djiceman, 2018-05-07:
Connect your serial tx and rx lines only then power up with ac. Take the required precautions.
Open up your terminal client and connect to the com port for the sonoff, do you get any logs?

w4tsn, 2018-05-07:
I don't get any output from the serial port on AC. With connected programmer I receive logs as expected.
I tried to connect RX->RX, TX->TX as well as RX-TX, TX-RX. I've not connected GND and 3.3V, which is correct, isn't it?

w4tsn, 2018-05-07:
The flashing process succeeds. I'm flashing with baud rate 115200 and I've already flashed Tasmota 5.12.0 and 5.13.1. I tried both the release bin of 5.12.0 as well as a self-compiled one of both.

I already commented an explanation of my setup here https://github.com/arendst/Sonoff-Tasmota/issues/391#issuecomment-386997124

Frogmore42, 2018-05-07:
If you don't connect the gnd, you have no reliable reference, so you will get serial errors.

If you have a DVM and know how to use it you can measure the VCC pin and see what it is when the Sonoff is plugged in.  Since it is possible to kill yourself doing this, don't try it unless you already know how to not kill yourself when working with high voltage.

It sounds like your Sonoff's power supply is not working correctly.

w4tsn, 2018-05-08:
So this seems to be a problem with my sonoff, not with this firmware right?

Ok, so I guess my options are handing it to someone with more experience to see if they can fix the sonoffs power supply or buying a new one, or, well, both.

Thanks for the help so far :)

Frogmore42, 2018-05-10:
Well, the FW you are using is built with core 2.4.0.  Some people have had problems with that version of the core, and have had better luck with version 2.3.0.  Using 2.3.0 is easy to do if you are building your own FW with PlatformIO.  I would try that first.

closed 2018-07-09 by stale[bot]

ohadbenita, 2018-12-03:
Been having the exact same issues with all my new sonoff basics, anyone have had some success about this ?

Frogmore42, 2018-12-03:
Which issue are you having?
Are you able to flash it?
Does it work when powered by your serial converter?
Does it work when powered by the line/high voltage?
The Basic has a power supply that is known to be weak.  There were code changes relatively recently that are supposed to improve it.
What version are you using?
Did you compile it yourself or download a release from somewhere?

ascillato, 2018-12-03:
Solved. See https://github.com/arendst/Sonoff-Tasmota/issues/4523