arendst/Tasmota #2650: Enable user mode for GPIO5 on the Sonoff RF/Basic

brainnex, 2018-05-06:
Hello everybody,
I have some Sonoff RF and I need one more IO. I see in the schematics that GPIO4 and GPIO5 are unused but only GPIO4 is in Tasmota 5.12.0 configurable. Is there a reason for that? Could you make GPIO5 usable? This would be awesome. 

Many thanks to the developper and community.

brainnex, 2018-05-06:
I use the RF functionality. Will this still be usable?

Leseratte10, 2018-05-07:
RF in Sonoff devices is not handled by the firmware / by Tasmota. There is a seperate RF chip. So yes,RF will still work with the "generic" module setting. You just have to configure all the GPIOs by yourself (Relay, LED, and your own used ports).

brainnex, 2018-05-13:
Thanks guys. The workaround is working.

ascillato, 2018-05-13:
Happy to know that it is working for you. Please close the issue. Thanks.

closed 2018-05-13 by brainnex