arendst/Tasmota #2651: device not (always) using fulltopic for stat and tele responses

alegner, 2018-05-06:
Hi, using Sonoff Basic with v5.13.1 of this beautiful software.

Problem:

Client mosqsub/27074-raspberry received PUBLISH (d0, q0, r0, m0, 'lichter/garten/cmnd/fulltopic', ... (0 bytes))
lichter/garten/cmnd/fulltopic (null)
Client mosqsub/27074-raspberry received PUBLISH (d0, q0, r0, m0, 'stat/garten/RESULT', ... (41 bytes))
**stat/garten/RESULT** {"**FullTopic":"lichter/%topic%/%prefix%/**"}

I.e. I'm using self defined fulltopic  'lichter/%topic%/%prefix%,  but getting 'stat' (and 'tele') messages back to (default) %prefix%/%topic%.

What am I missing?  Or is this a bug?

BR,
 Andreas

Here my 'status 0':

Client mosqsub/27074-raspberry received PUBLISH (d0, q0, r0, m0, 'lichter/garten/cmnd/status', ... (1 bytes))
lichter/garten/cmnd/status 0
Client mosqsub/27074-raspberry received PUBLISH (d0, q0, r0, m0, 'stat/garten/STATUS', ... (189 bytes))
stat/garten/STATUS {"Status":{"Module":1,"FriendlyName":["Garten"],"Topic":"garten","ButtonTopic":"garten","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":1,"PowerRetain":1}}
Client mosqsub/27074-raspberry received PUBLISH (d0, q0, r0, m0, 'stat/garten/STATUS1', ... (273 bytes))
stat/garten/STATUS1 {"StatusPRM":{"Baudrate":115200,"GroupTopic":"lichter","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:01","StartupUTC":"2018-05-06T20:52:19","Sleep":0,"BootCount":18,"SaveCount":111,"SaveAddress":"F8000"}}
Client mosqsub/27074-raspberry received PUBLISH (d0, q0, r0, m0, 'stat/garten/STATUS2', ... (119 bytes))
stat/garten/STATUS2 {"StatusFWR":{"Version":"5.13.1","BuildDateTime":"2018.05.01 21:05:42","Boot":6,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
Client mosqsub/27074-raspberry received PUBLISH (d0, q0, r0, m0, 'stat/garten/STATUS3', ... (169 bytes))
stat/garten/STATUS3 {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["Fritzi2","indebuurt2"],"TelePeriod":300,"SetOption":["55800069","55818000"]}}
Client mosqsub/27074-raspberry received PUBLISH (d0, q0, r0, m0, 'stat/garten/STATUS4', ... (109 bytes))
stat/garten/STATUS4 {"StatusMEM":{"ProgramSize":530,"Free":472,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
Client mosqsub/27074-raspberry received PUBLISH (d0, q0, r0, m0, 'stat/garten/STATUS5', ... (202 bytes))
stat/garten/STATUS5 {"StatusNET":{"Hostname":"garten","IPAddress":"192.168.178.29","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"EC:FA:BC:10:75:D7","Webserver":2,"WifiConfig":2}}
Client mosqsub/27074-raspberry received PUBLISH (d0, q0, r0, m0, 'stat/garten/STATUS6', ... (180 bytes))
stat/garten/STATUS6 {"StatusMQT":{"MqttHost":"192.168.178.236","MqttPort":8883,"MqttClientMask":"garten","MqttClient":"garten","MqttUser":"lichter","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
Client mosqsub/27074-raspberry received PUBLISH (d0, q0, r0, m0, 'stat/garten/STATUS7', ... (205 bytes))
stat/garten/STATUS7 {"StatusTIM":{"UTC":"Sun May 06 20:55:20 2018","Local":"Sun May 06 21:55:20 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:21","Sunset":"20:12"}}
Client mosqsub/27074-raspberry received PUBLISH (d0, q0, r0, m0, 'stat/garten/STATUS10', ... (44 bytes))
stat/garten/STATUS10 {"StatusSNS":{"Time":"2018-05-06T21:55:20"}}
Client mosqsub/27074-raspberry received PUBLISH (d0, q0, r0, m0, 'stat/garten/STATUS11', ... (163 bytes))
stat/garten/STATUS11 {"StatusSTS":{"Time":"2018-05-06T21:55:20","Uptime":"0T00:03:01","Vcc":3.484,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Fritzi2","RSSI":56,"APMac":"24:65:11:7F:F7:B7"}}}

alegner, 2018-05-07:
Hmm, not seeing any similarity, why would you think this is same?

Jason2866, 2018-05-07:
Try to change your topic definitions to the standard (as described) wiki and see if it works.
If a unusual setup is choosen stat and tele response sometimes get confused or is missing...

ascillato, 2018-06-02:
Hi,

If you could manage to solve your issue, please close it. Thanks.