arendst/Tasmota #2652: Is Sonoff Pow R2 better than R1 (when used with tasmota)?

kaefert, 2018-05-06:
Hi there!

I am currently considering to buy Sonoff Pow for $10.37 or Sonoff Pow R2 for $13.83
Some sellers seem to say R2 is more accurate then R1, the comparison chart that floats around just lists a few changes that seem to be mostly software related.

Could you please state your opinion (if you have experience with both) whether R2 is really more accurate then R1, or if it's just little design & software changes?

Thank you!

tobox, 2018-05-07:
I had exactly the same question. I already have 3 R1s, and ordered a R2 that should arrive shortly. I will do some tests as soon as I receive them. I will compare the measurements with my HM-ES-PMSw1-Pl and my AVM DECT 200.

tobox, 2018-05-07:
Just received the Pow R2. Flashing was no problem, I decided to go with 5.13.1a with core version 2.3.0, because there are way too many issues with 2.4.0 and 2.4.1. 

I connected it behind the HM-ES-PMSw1-Pl and measured 1.03W with the relay turned on. Then I connected a  400W halogen lamp to the POW and calibrated it to the HM-ES-PMSw1-Pl readings minus 1.03W.

Then I tried to measure some small loads (oder Sonoffs), but the Pow always showed 0W - so not much better than the regular POW R1. Even with 3 Sonoffs in connected, with all relays turned on, it still showed 0W :-( The Homematic device showed 4.4W (for the 3 other Sonoffs plus the Pow).

Next test was with a 40W soldering iron, where the Pow showed almost the same values as the homematic device. 

So I guess, the Pow R2 is not really better suited for low power measurements than the R1. Maybe it has less noise than the R1, I was not able to test than. For most of my applications, I had to average the measurements of the R1 to remove spikes. 

I will let you know when I have more results.

eazuka, 2018-05-16:
i'm also curious about this

kaefert, 2018-06-30:
@tobox did you have a chance to test more yet? Does the Sonoff R2 produce less noise (spikes although the load should not produce spikes)?

Does anybody else have access to both an R1 and R2 and can report on the differences she perceived?

closed 2018-08-29 by stale[bot]