arendst/Tasmota #2653: TIPP: Sonoff Basic for 3,47€ / $4.20

0n3-70uch, 2018-05-06:
Hi, the second Sonoff deal from the same shop. Now the Sonoff Basic (you can order two devices in seperate orders):
https://www.rosegal.com/iphone-cases/sonoff-basic-dual-channel-wifi-wireless-smart-switch-for-diy-home-safety-2205777.html

Just use this coupon code:
DSRGMAY7

Tipp:
Change the currency to euro (or whatever you want) at the top right corner. This trick will skip the PayPal exchange costs.

Greetings Fabi

Sonoff S20 Deal:
https://github.com/arendst/Sonoff-Tasmota/issues/2630

0n3-70uch, 2018-05-07:
Sold out :-(.

0n3-70uch, 2018-05-08:
Back in stock :-).

znanev, 2018-05-08:
Thanks again!

Just ordered two - couldn't miss them for $4.09 per device, delivered!

0n3-70uch, 2018-05-09:
Yeah so crazy. An ice cream is more expensive :-).

ascillato, 2018-06-04:
Great!

closed 2018-06-04 by ascillato