arendst/Tasmota #2655: Sonoff POW - Poll Data every Second - Feasibility & Implementability with Sonoff-Tasmota (5.13.1)

oblansundar, 2018-05-07:
Hi,

I am using Sonoff POW to get the power consumption data. I have flashed it with Sonoff-Tasmota Firmware (5.13.1) - https://github.com/arendst/Sonoff-Tasmota. So far, I get the Power Consumption data under the topic tele/<MQTT_CLIENT_ID>/SENSOR automatically for minimum interval of **10 seconds**. This interval is the lower limit as suggested in https://github.com/arendst/Sonoff-Tasmota/wiki/Commands. 

But, I have a requirement that I need the information published under the topic tele/<MQTT_CLIENT_ID>/SENSOR automatically for **one second** if we meet certain thresholds as suggested here - https://github.com/arendst/Sonoff-Tasmota/issues/688 (In this thread, the telemetry interval is 30 seconds - my requirement is one second). I can get the SENSOR  information under stat/<MQTT_CLIENT_ID>/STATUS10 , but I need to publish this topic every second to Sonoff POW and I ran a Python script to do this, but the MQTT got flooded. So, I needed to find an approach to do something in firmware that publishes MQTT under SENSOR topic for every one second. 

Basically, to be accurate, I would like to have the same behavior as https://github.com/arendst/Sonoff-Tasmota/issues/688 (Checking the thresholds) except the telemetry interval I want is one second.

I have the following questions:
Is the above feasible? If so, how could it be implemented? Is it implementable with Sonoff-Tasmota Firmware?

<I also checked if it would be a hardware limitation of the sensor.  I am not sure how to check that, but i got some information about that. I checked the sensor that is used inside SONOFF for power sensing is  HLW 8012 IC. (http://tinkerman.cat/hlw8012-ic-new-sonoff-pow/ ) Datasheet (Only in chinese) - http://www.datasheetspdf.com/pdf-down/H/L/W/HLW8012-HLW.pdf>

Frogmore42, 2018-05-07:
From the link above:
For the suggested application in the datasheet a frequency of 1Hz corresponds to a ~12W active power, 10Hz to ~120W, 100Hz to ~1.2kW and so on.

That means to be able to read down to 12W, it takes at least a second.  To get lower values requires even more time.  But, if you only care about power values in the 100s of Watts and greater the sensor would not be a limitation.

What exactly are you doing that you feel you need such frequent updates?

oblansundar, 2018-05-08:
@Frogmore42  Thanks for the reply.. We have a coffee machine, where we connect the Sonoff POW for which we need live update of power consumed. So, the power consumption would be mostly 500 Watts to 800 Watts when somebody uses it. So, we don't need frequent updates exactly, but I would like to compare the sensor information to the threshold every one second and then send the data if it changes. Yeah, in worst case, the updates will be every second, but on average, it would be like every 5 or 10 seconds. 

@arendst We are measuring the power of greater than 100 Watts. Now, how to implement this in the Tasmota Firmware? (Minimum Telemetry is 10 sec by default). Also, "rules is able to monitor these values", you mean rules inside the Sonoff POW checking the thresholds right?

Frogmore42, 2018-05-08:
Doesn't really answer the question of what are you doing with the data, ie what information are you trying to extract from it?  What decisions are you going to make based on the data?

As an example I have a dehumidifier connected to a POW.  I keep track of run time and efficiency of the dehumidifier.  I use the total kwh value to determine how much power the dehumidifier used for a cycle and keep track of the number of bucket full of water output from a repurposed rain gauge.  I have multiple temp and humidity sensors that turn on the dehumidifier when there is enough moisture in the air for it to run efficiently.

There is no need to get updates more frequently than every 10-30s, since the chip keeps track of the energy used more accurately than I could by integrating the occasional power updates.  Since I know there is significant inaccuracy in all of the measurements, this method is accurate enough and got me to a solution that was easy to implement quickly.

Your needs might be different, hence the question as to why you need to know that quickly.

thantsi, 2018-05-09:
@arendst What do you mean with "he values are stored and will be presented when asked for"? Maybe @oblansundar wants the analysis to happen every one second to identify a certain operation of the device. If it is every 10-30 seconds and the operation of making a coffee for example lasts 5 seconds then most probable he will lose this reading. The interval of 10 seconds measures the average power consumption in these  10 seconds or it triggers a reading at the 10th second?

oblansundar, 2018-05-21:
Well, I found some sections of code in the firmware where the telemetry period is being set to 10 even if we give it as something between 0 and 10.

From sonoff.ino file:

>   if ((payload >= 0) && (payload < 3601)) {
>         Settings.tele_period = (1 == payload) ? TELE_PERIOD : payload;
>         if ((Settings.tele_period > 0) && (Settings.tele_period < 10)) Settings.tele_period = 10;   // Do not allow periods < 10 seconds
>         tele_period = Settings.tele_period;
>       }
> 

And somewhere down:

>  if (Settings.tele_period) {
>     tele_period++;
>     if (tele_period == Settings.tele_period -1) {
>       XsnsCall(FUNC_PREP_BEFORE_TELEPERIOD);
>     }
>     if (tele_period >= Settings.tele_period) {
>       tele_period = 0;
>       mqtt_data[0] = '\0';
>       MqttShowState();
>       MqttPublishPrefixTopic_P(TELE, PSTR(D_RSLT_STATE), MQTT_TELE_RETAIN);
>       mqtt_data[0] = '\0';
>       if (MqttShowSensor()) {
>         MqttPublishPrefixTopic_P(TELE, PSTR(D_RSLT_SENSOR), Settings.flag.mqtt_sensor_retain);
> #ifdef USE_RULES
>         RulesTeleperiod();  // Allow rule based HA messages
> #endif  // USE_RULES
>       }

So, what if we make 
`  if ((Settings.tele_period > 0) && (Settings.tele_period < 10)) Settings.tele_period = 10;   // Do not allow periods < 10 seconds`
     ` tele_period = Settings.tele_period;`

to:

`  //if ((Settings.tele_period > 0) && (Settings.tele_period < 10)) Settings.tele_period = 10;   // Do not allow periods < 10 seconds`
        `tele_period = Settings.tele_period;`

And test it by giving TELE_PERIOD in user_config.h to 5?
Would it cause some instability in other functions?

oblansundar, 2018-05-22:
@Frogmore42  Thanks for the reply. I actually changed it to 5 seconds and tested, it worked fine. It also works for one second too after setting TELE_PERIOD=1 in userconfig.h file.

oblansundar, 2018-05-22:
After getting telemetry data for 1 seecond, now, the missing part is:
-I want to check if the voltage value is same as previous value and then send the telemetry if it changes (within 1 sec).  I see that we have the commands to set the **thresholds** for current,voltage and Power. 

I require the SENSOR information only when there is a change in the previous value. How could it be done with thresholds? (I can set Low and High Thresholds) but I just want the change in the previous value to be reported.

closed 2018-07-10 by ascillato2

s0170071, 2020-04-14:
> From the link above:
> For the suggested application in the datasheet a frequency of 1Hz corresponds to a ~12W active power, 10Hz to ~120W, 100Hz to ~1.2kW and so on.
>
> That means to be able to read [...]

Well, there is a way to speed up readings at low power values. But only if the power drops from a higher value. What you would have to do is wait for the slope from the hlw signal to time-out. That means that if you expect that slope at a certain point in time and it does not appear, you can safely assume that the power is lower. And you can keep counting down the power until the next slope arrives. 

My usecase: stop an automatic lawn mower at the charging station when it has finished charging - by switching off the socket. Currently this is  too slow so that the mower leaves the station before the socket is switched off.