arendst/Tasmota #2656: Sonoff POW - Time Issue [1970-01-01] - Not updated

oblansundar, 2018-05-07:
Hello everyone,

I am using Sonoff POW to monitor Power Consumption Data. I have used the latest version (5.13.1) of Sonoff-Tasmota. I am not able to get the correct time in the data that Sonoff POW sends under the topic - tele/<MQTT_CLIENT_ID>/SENSOR. Sample data:

> {"Time":"1970-01-01T00:10:13","ENERGY":{"Total":0.025,"Yesterday":0.000,"Today":0.025,"Period":0,"Power":0,"Factor":0.00,"Voltage":223,"Current":0.000}}

output of STATUS 7:
> {"StatusTIM":{"UTC":"Thu Jan 01 00:08:38 1970","Local":"Thu Jan 01 00:08:38 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1}}

output of STATUS 5:
> {"StatusNET":{"Hostname":"SONOFF_POW_01-1538","IPAddress":"YYY.YYY.YYY.YYY","Gateway":"YYY.YYY.YYY.YYY","Subnetmask":"255.255.192.0","DNSServer":"A","Mac":"BC:DD:C2:EA:66:02","Webserver":2,"WifiConfig":2}}

-DNS server is A 

For Sonoff-Basic in same network:
> {"StatusNET":{"Hostname":"SONOFF_BASIC_01","IPAddress":"YYY.YYY.YYY.YYY","Gateway":"YYY.YYY.YYY.YYY","Subnetmask":"255.255.192.0","DNSServer":"A","Mac":"DC:4F:22:2D:C8:A7","Webserver":2,"WifiConfig":2}}

-Same DNS server - no DNS issues here.

Also, I checked the NTP servers. They are all one and the same. So, I think this issue is not something that is network related but something else.

If you could suggest any other methods to set the correct time in this device, it would be helpful.

Thanks,
Sundar.

Jason2866, 2018-05-07:
You blacklisted your private ip addresses. Got the pow a valid one? Or have you fixed values?
Is the DNS server name resolved?
Change for testing the DNS in the POW to a known and working fixed ip address

Frogmore42, 2018-05-07:
You either have a networking issue, but you have deleted all your IP addresses, so not easy to tell (and not necessary if you are using Private IP addresses).

Or, you have the NTP servers misconfigured, but you did not show what Tasmota thinks you want them to be.

Or, you have the much more unlikely issue I encounter when the Internet goes down for an extended period.

There is no output from status 0, which makes it hard to tell if there is enough Wi-Fi signal to get reliable networking.  That is the first thing to check.

Next is that the DNS server setup is correct.  There is no way to tell from the console if it is actually working or not, but you can tell what the value is.  Your output shows "A", which can't be right, so I must assume you hid it for some reason.

Assuming you have a correct IP address there, the next thing is to make sure the values of the time servers are correct.  They can get corrupted, so you have to look at the actual values reported by the device and not just the values you configured at some point in time.

Only AFTER you have verified ALL of that is correct AND you have already done a power cycle, you can try the final step.  Remove ALL power from the device for at least 30min.  Then turn it back on.

I have to do this to some of my devices after an outage because the underlying code in the SDK which is where the real NTP code lives gets confused and will never get the correct time.

Also, you didn't include status 0 info, so we can't tell which framework version you are using.  Each version has different issues, which is why there is a request for that information in the issue template.

oblansundar, 2018-05-08:
@Jason2866 But the Sonnoff POW and Sonoff Basic has the same DNS. Not sure why Sonoff POW is only not able to resolve. But as per your suggestion, I will try changing the DNS of Sonoff POW to 8.8.8.8 and see what happens.

jeylites, 2018-05-12:
I'm having the same issue NTP not sync wit local clock. I tried a few combination of servers, reboot boot AP router, 

Hardware: Sonoff T1 Touch UK
Tasmota: 5.13.1a
Core: 2.4.1

```
/ -- Time - Up to three NTP servers in your region
#define NTP_SERVER1            "pool.ntp.org"       // [NtpServer1] 
#define NTP_SERVER2            "my.pool.ntp.org"    // [NtpServer2]
#define NTP_SERVER3            "3.asia.pool.ntp.org"  // [NtpServer3]
```

```
// -- Application ---------------------------------
#define APP_TIMEZONE           8                 // [Timezone]
```

```
00:00:00 Project ves1 VES 1 (Topic ves1, Fallback 24, GroupTopic sonoffs) Version 5.13.1a-2_4_1
00:00:00 WIF: Connecting to AP1 SUGAR in mode 11N as ves1-1256...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on ves1-1256 with IP address 192.168.0.24
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/ves1/LWT = Online (retained)
00:00:05 MQT: cmnd/ves1/POWER = 
00:00:05 MQT: tele/ves1/INFO1 = {"Module":"Sonoff T1 3CH","Version":"5.13.1a","FallbackTopic":"24","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/ves1/INFO2 = {"WebServerMode":"Admin","Hostname":"ves1-1256","IPAddress":"192.168.0.24"}
00:00:05 MQT: tele/ves1/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: stat/ves1/RESULT = {"POWER1":"OFF"}
00:00:06 MQT: stat/ves1/POWER1 = OFF
00:00:06 MQT: stat/ves1/RESULT = {"POWER2":"OFF"}
00:00:06 MQT: stat/ves1/POWER2 = OFF
00:00:06 MQT: stat/ves1/RESULT = {"POWER3":"ON"}
00:00:06 MQT: stat/ves1/POWER3 = ON
00:00:13 MQT: tele/ves1/STATE = {"Time":"1970-01-01T00:00:13","Uptime":"0T00:00:13","POWER1":"OFF","POWER2":"OFF","POWER3":"ON","Wifi":{"AP":1,"SSId":"SUGAR","RSSI":100,"APMac":"88:D7:F6:02:18:80"}}
00:00:13 MQT: stat/ves1/RESULT = {"POWER1":"OFF"}
00:00:13 MQT: stat/ves1/POWER1 = OFF
00:00:13 MQT: stat/ves1/RESULT = {"POWER2":"OFF"}
00:00:13 MQT: stat/ves1/POWER2 = OFF
00:00:13 MQT: stat/ves1/RESULT = {"POWER3":"ON"}
00:00:13 MQT: stat/ves1/POWER3 = ON
00:00:40 CMD: status 0
00:00:40 MQT: stat/veshinu1/STATUS = {"Status":{"Module":30,"FriendlyName":["VES 1","VES 12","VES 13"],"Topic":"ves1","ButtonTopic":"0","Power":4,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:00:40 MQT: stat/ves1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:40","StartupUTC":"","Sleep":0,"BootCount":4,"SaveCount":26,"SaveAddress":"FA000"}}
00:00:40 MQT: stat/ves1/STATUS2 = {"StatusFWR":{"Version":"5.13.1a","BuildDateTime":"2018-05-12T19:31:18","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
00:00:40 MQT: stat/ves1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SUGAR","SALT"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
00:00:40 MQT: stat/ves1/STATUS4 = {"StatusMEM":{"ProgramSize":466,"Free":536,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
00:00:40 MQT: stat/ves1/STATUS5 = {"StatusNET":{"Hostname":"ves1-1256","IPAddress":"192.168.0.24","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"60:01:94:97:44:E8","Webserver":2,"WifiConfig":4}}
00:00:40 MQT: stat/ves1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.200","MqttPort":1883,"MqttClientMask":"24","MqttClient":"24","MqttUser":"admin","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:40 MQT: stat/ves1/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:40 1970","Local":"Thu Jan 01 00:00:40 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":8}}
00:00:40 MQT: stat/ves1/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:40"}}
00:00:40 MQT: stat/ves1/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:40","Uptime":"0T00:00:40","POWER1":"OFF","POWER2":"OFF","POWER3":"ON","Wifi":{"AP":1,"SSId":"SUGAR","RSSI":100,"APMac":"88:D7:F6:02:18:80"}}}
00:00:46 CMD: timezone 99
```

Jason2866, 2018-05-12:
Try to build with Arduino v.2.3.0. There are many users with problems with v.2.4.0 and 2.4.1...
You have disabled many options. Why? Your programm size is very small with 466kb.
May you have killed needed feature.

jeylites, 2018-05-12:
@Jason2866 I did a 2.3.0 build as well, still same issue. I have disabled things I'm not using in `user_config.h`

**STATUS 0**
```
00:00:00 Project jey1 JEY 1 (Topic jey1, Fallback 26, GroupTopic sonoffs) Version 5.13.1a-2_3_0
00:00:00 WIF: Connecting to AP1 SUGAR in mode 11N as jey1-4683...
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on jey1-4683 with IP address 192.168.0.26
00:00:08 MQT: Attempting connection...
00:00:08 MQT: Connected
00:00:08 MQT: tele/jey1/LWT = Online (retained)
00:00:08 MQT: cmnd/jey1/POWER = 
00:00:08 MQT: tele/jey1/INFO1 = {"Module":"Sonoff T1 3CH","Version":"5.13.1a","FallbackTopic":"26","GroupTopic":"sonoffs"}
00:00:08 MQT: tele/jey1/INFO2 = {"WebServerMode":"Admin","Hostname":"jey1-4683","IPAddress":"192.168.0.26"}
00:00:08 MQT: tele/jey1/INFO3 = {"RestartReason":"Software/System restart"}
00:00:09 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
00:00:09 MQT: stat/jey1/POWER1 = OFF
00:00:09 MQT: stat/jey1/RESULT = {"POWER2":"OFF"}
00:00:09 MQT: stat/jey1/POWER2 = OFF
00:00:09 MQT: stat/jey1/RESULT = {"POWER3":"ON"}
00:00:09 MQT: stat/jey1/POWER3 = ON
00:00:10 MQT: stat/jey1/RESULT = {"POWER3":"OFF"}
00:00:10 MQT: stat/jey1/POWER3 = OFF
00:00:11 MQT: stat/jey1/RESULT = {"POWER3":"ON"}
00:00:11 MQT: stat/jey1/POWER3 = ON
00:00:12 MQT: stat/jey1/RESULT = {"POWER3":"OFF"}
00:00:12 MQT: stat/jey1/POWER3 = OFF
00:00:14 MQT: stat/jey1/RESULT = {"POWER1":"ON"}
00:00:14 MQT: stat/jey1/POWER1 = ON
00:00:17 MQT: tele/jey1/STATE = {"Time":"1970-01-01T00:00:17","Uptime":"0T00:00:16","POWER1":"ON","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR","RSSI":88,"APMac":"88:D7:F6:02:18:80"}}
00:00:17 MQT: stat/jey1/RESULT = {"POWER1":"ON"}
00:00:17 MQT: stat/jey1/POWER1 = ON
00:00:17 MQT: stat/jey1/RESULT = {"POWER2":"OFF"}
00:00:17 MQT: stat/jey1/POWER2 = OFF
00:00:17 MQT: stat/jey1/RESULT = {"POWER3":"OFF"}
00:00:17 MQT: stat/jey1/POWER3 = OFF
00:00:21 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
00:00:21 MQT: stat/jey1/POWER1 = OFF
00:00:23 KNX: Received from 2.2.1 Command Write: 1 to Relay 1
00:00:23 MQT: stat/jey1/RESULT = {"POWER1":"ON"}
00:00:23 MQT: stat/jey1/POWER1 = ON
00:00:42 KNX: Received from 2.2.1 Command Write: 0 to Relay 1
00:00:42 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
00:00:42 MQT: stat/jey1/POWER1 = OFF
00:02:00 MQT: tele/jey1/UPTIME = {"Time":"1970-01-01T00:02:00","Uptime":"0T00:01:59"}
00:05:17 MQT: tele/jey1/STATE = {"Time":"1970-01-01T00:05:17","Uptime":"0T00:05:16","POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR","RSSI":82,"APMac":"88:D7:F6:02:18:80"}}
00:05:17 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
00:05:17 MQT: stat/jey1/POWER1 = OFF
00:05:17 MQT: stat/jey1/RESULT = {"POWER2":"OFF"}
00:05:17 MQT: stat/jey1/POWER2 = OFF
00:05:17 MQT: stat/jey1/RESULT = {"POWER3":"OFF"}
00:05:17 MQT: stat/jey1/POWER3 = OFF
00:10:01 APP: Serial logging disabled
00:10:06 CMD: status 0
00:10:06 MQT: stat/jey1/STATUS = {"Status":{"Module":30,"FriendlyName":["JEY 1","JEY 12","JEY 13"],"Topic":"jey1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:10:06 MQT: stat/jey1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:10:04","StartupUTC":"","Sleep":0,"BootCount":3,"SaveCount":19,"SaveAddress":"F9000"}}
00:10:06 MQT: stat/jey1/STATUS2 = {"StatusFWR":{"Version":"5.13.1a","BuildDateTime":"2018-05-12T20:24:53","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:10:06 MQT: stat/jey1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SUGAR","SALT"],"TelePeriod":300,"SetOption":["02008009","55818000"]}}
00:10:06 MQT: stat/jey1/STATUS4 = {"StatusMEM":{"ProgramSize":456,"Free":544,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
00:10:06 MQT: stat/jey1/STATUS5 = {"StatusNET":{"Hostname":"jey1-4683","IPAddress":"192.168.0.26","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"60:01:94:98:B2:4B","Webserver":2,"WifiConfig":4}}
00:10:06 MQT: stat/jey1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.200","MqttPort":1883,"MqttClientMask":"26","MqttClient":"26","MqttUser":"admin","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:10:06 MQT: stat/jey1/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:10:06 1970","Local":"Thu Jan 01 00:10:06 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":8}}
00:10:06 MQT: stat/jey1/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:10:06"}}
00:10:06 MQT: stat/jey1/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:10:06","Uptime":"0T00:10:04","POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"SUGAR","RSSI":88,"APMac":"88:D7:F6:02:18:80"}}}

```

**user_config.h**

```
[... 308 lines cut ...]
/*********************************************************************************************\
 * Select KNX without Emulation to save space
\*********************************************************************************************/

#define USE_KNX_NO_EMULATION                     // Create sonoff-knx with KNX but without Emulation (See sonoff_post.h)

/*********************************************************************************************\
 * Compile a minimal version if upgrade memory gets tight ONLY TO BE USED FOR UPGRADE STEP 1!
 *   To be used as step 1 during upgrade.
[... cut ...]
//#define BE_MINIMAL                               // Minimal version if upgrade memory gets tight (-45k code, -2k mem)

/*********************************************************************************************\
 * No user configurable items below
\*********************************************************************************************/

#if defined(USE_MQTT_TLS) && defined(USE_WEBSERVER)
  #error "Select either USE_MQTT_TLS or USE_WEBSERVER as there is just not enough memory to play with"
#endif

#endif  // _USER_CONFIG_H_

```

oblansundar, 2018-05-22:
The time in my Sonoff POW got fixed automatically. I did not do anything except using the new firmware - 5.14.0.

@jeylites You can close this after your issue is fixed.

jeylites, 2018-05-23:
Sorry, I didn’t report back. Mine was a AP router issue. I moved all the devices to the main router and it worked right after.

closed 2018-06-04 by ascillato