arendst/Tasmota #2658: Tasmota 5.12.0 infinite loop after having set GPIO14 to 4

yann1420, 2018-05-07:
I have a number of SonOff basic's and decided to connect a DS18B20 to one of them using GPIO14.
I used MQTT to set GPIO14 to 4 and then the device started action weird and thus I started ate zero again, i.e. reprogramming/reconfiguring the device but still is in some infinite loop that takes about 15-20 seconds and produces every cycl the following output on my MQTT server and clicks the relay and LED accordingly:

_tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
stat/sonoff/RESULT {"POWER":"OFF"}
stat/sonoff/POWER OFF
tele/sonoff/INFO1 {"Module":"Sonoff Basic","Version":"5.12.0","FallbackTopic":"DVES_AF04A6","GroupTopic":"sonoffs"}
tele/sonoff/INFO2 {"WebServerMode":"Admin","Hostname":"sonoff","IPAddress":"192.168.178.94"}
tele/sonoff/INFO3 {"RestartReason":"Software/System restart"}
**homeassistant/switch/sonoff/config (null)**
stat/sonoff/RESULT {"GPIO1":"0 (None)","GPIO3":"0 (None)","GPIO4":"0 (None)","GPIO14":"0 (None)"}
stat/sonoff/RESULT {"POWER":"ON"}
stat/sonoff/POWER ON
stat/sonoff/RESULT {"POWER":"ON"}
stat/sonoff/POWER ON_

Any idea where this comes from (even after starting from scratch) and what does this line **homeassistant/switch/sonoff/config (null)** come from?

Per the request how the status ) looks like, see below
_cmnd/sonoff/status 0
stat/sonoff/RESULT {"GPIO1":"0 (None)","GPIO3":"0 (None)","GPIO4":"0 (None)","GPIO14":"0 (None)"}
tele/sonoff/INFO1 {"Module":"Sonoff Basic","Version":"5.12.0","FallbackTopic":"DVES_AF04A6","GroupTopic":"sonoffs"}
tele/sonoff/INFO2 {"WebServerMode":"Admin","Hostname":"sonoff","IPAddress":"192.168.178.94"}
tele/sonoff/INFO3 {"RestartReason":"Software/System restart"}
homeassistant/switch/sonoff/config (null)
stat/sonoff/RESULT {"POWER":"ON"}
stat/sonoff/POWER ON
stat/sonoff/STATUS {"Status":{"Module":1,"FriendlyName":"Sonoff","Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
stat/sonoff/STATUS1 {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0T00:00:12","Sleep":0,"BootCount":678,"SaveCount":824,"SaveAddress":"F4000"}}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-09T14:54:51","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/STATUS3 {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"titania","SSId2":"openwrt","TelePeriod":300,"SetOption":"00000009"}}
stat/sonoff/STATUS4 {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
stat/sonoff/STATUS5 {"StatusNET":{"Hostname":"sonoff","IPAddress":"192.168.178.94","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.60","Mac":"5C:CF:7F:AF:04:A6","Webserver":2,"WifiConfig":2}}
stat/sonoff/STATUS6 {"StatusMQT":{"MqttHost":"MQTT","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AF04A6","MqttUser":"DVES_USER","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
stat/sonoff/STATUS7 {"StatusTIM":{"UTC":"Mon May 07 10:46:19 2018","Local":"Mon May 07 11:46:19 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
stat/sonoff/STATUS10 {"StatusSNS":{"Time":"2018-05-07T11:46:19"}}
stat/sonoff/STATUS11 {"StatusSTS":{"Time":"2018-05-07T11:46:19","Uptime":"0T00:00:12","Vcc":3.194,"POWER":"ON","Wifi":{"AP":1,"SSId":"titania","RSSI":88,"APMac":"70:5A:9E:27:9F:1D"}}}
stat/sonoff/RESULT {"POWER":"ON"}
stat/sonoff/POWER ON
tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
stat/sonoff/RESULT {"POWER":"OFF"}
stat/sonoff/POWER OFF
tele/sonoff/INFO1 {"Module":"Sonoff Basic","Version":"5.12.0","FallbackTopic":"DVES_AF04A6","GroupTopic":"sonoffs"}
tele/sonoff/INFO2 {"WebServerMode":"Admin","Hostname":"sonoff","IPAddress":"192.168.178.94"}
tele/sonoff/INFO3 {"RestartReason":"Software/System restart"}
homeassistant/switch/sonoff/config (null)
stat/sonoff/RESULT {"GPIO1":"0 (None)","GPIO3":"0 (None)","GPIO4":"0 (None)","GPIO14":"0 (None)"}
stat/sonoff/STATUS {"Status":{"Module":1,"FriendlyName":"Sonoff","Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
stat/sonoff/STATUS1 {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0T00:00:13","Sleep":0,"BootCount":679,"SaveCount":825,"SaveAddress":"FB000"}}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-09T14:54:51","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/STATUS3 {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"titania","SSId2":"openwrt","TelePeriod":300,"SetOption":"00000009"}}
stat/sonoff/STATUS4 {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
stat/sonoff/STATUS5 {"StatusNET":{"Hostname":"sonoff","IPAddress":"192.168.178.94","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.60","Mac":"5C:CF:7F:AF:04:A6","Webserver":2,"WifiConfig":2}}
stat/sonoff/STATUS6 {"StatusMQT":{"MqttHost":"MQTT","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AF04A6","MqttUser":"DVES_USER","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
stat/sonoff/STATUS7 {"StatusTIM":{"UTC":"Mon May 07 10:46:36 2018","Local":"Mon May 07 11:46:36 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
stat/sonoff/STATUS10 {"StatusSNS":{"Time":"2018-05-07T11:46:36"}}
stat/sonoff/STATUS11 {"StatusSTS":{"Time":"2018-05-07T11:46:36","Uptime":"0T00:00:13","Vcc":3.195,"POWER":"OFF","Wifi":{"AP":1,"SSId":"titania","RSSI":84,"APMac":"70:5A:9E:27:9F:1D"}}}
stat/sonoff/RESULT {"POWER":"ON"}
stat/sonoff/POWER ON
stat/sonoff/RESULT {"POWER":"ON"}
stat/sonoff/POWER ON
tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
stat/sonoff/RESULT {"POWER":"OFF"}
stat/sonoff/POWER OFF
tele/sonoff/INFO1 {"Module":"Sonoff Basic","Version":"5.12.0","FallbackTopic":"DVES_AF04A6","GroupTopic":"sonoffs"}
tele/sonoff/INFO2 {"WebServerMode":"Admin","Hostname":"sonoff","IPAddress":"192.168.178.94"}
tele/sonoff/INFO3 {"RestartReason":"Software/System restart"}
homeassistant/switch/sonoff/config (null)
stat/sonoff/RESULT {"GPIO1":"0 (None)","GPIO3":"0 (None)","GPIO4":"0 (None)","GPIO14":"0 (None)"}
stat/sonoff/STATUS {"Status":{"Module":1,"FriendlyName":"Sonoff","Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
stat/sonoff/STATUS1 {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0T00:00:16","Sleep":0,"BootCount":680,"SaveCount":826,"SaveAddress":"FA000"}}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-09T14:54:51","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/STATUS3 {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"titania","SSId2":"openwrt","TelePeriod":300,"SetOption":"00000009"}}
stat/sonoff/STATUS4 {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
stat/sonoff/STATUS5 {"StatusNET":{"Hostname":"sonoff","IPAddress":"192.168.178.94","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.60","Mac":"5C:CF:7F:AF:04:A6","Webserver":2,"WifiConfig":2}}
stat/sonoff/STATUS6 {"StatusMQT":{"MqttHost":"MQTT","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AF04A6","MqttUser":"DVES_USER","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
stat/sonoff/STATUS7 {"StatusTIM":{"UTC":"Mon May 07 10:46:56 2018","Local":"Mon May 07 11:46:56 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
stat/sonoff/STATUS10 {"StatusSNS":{"Time":"2018-05-07T11:46:56"}}
stat/sonoff/STATUS11 {"StatusSTS":{"Time":"2018-05-07T11:46:56","Uptime":"0T00:00:16","Vcc":3.191,"POWER":"OFF","Wifi":{"AP":1,"SSId":"titania","RSSI":86,"APMac":"70:5A:9E:27:9F:1D"}}}
stat/sonoff/RESULT {"POWER":"ON"}
stat/sonoff/POWER ON
stat/sonoff/RESULT {"POWER":"ON"}
stat/sonoff/POWER ON_

Jason2866, 2018-05-07:
The change of setting GPI0 14 failed via mqtt. I think the devices crashes with the sended command 
Now HA tries after every reboot to send this command again (retained messages) 
You have to delete the retained messages in you HA broker.
Do a search in issues for "retain message" This is not a Tasmota Issue!

yann1420, 2018-05-08:
The fun thing is that I never an Home Automation instance in my environment. But yes, clearing the MQTT brooker did help.

closed 2018-05-08 by yann1420