arendst/Tasmota #2659: MHZ19 after random time shows 0 ppm

orimate, 2018-05-07:
I've installed 5.12 / 5.13, since after boot randomly (5 minutes - 2-3 hours) the sensor shows 0 ppm. I have flashed on Sonoff TH10, NodeMCU and Wemos mini d1 with 5.11 everything is normal since 3-4 months. With hyperterminal I analyzed the query FF 01 86 00 00 00 00 00 79 comes every ~ 2 sec, but answer from the MHZ19 nothing. Software restart does not help, I have to unplug the power plug that it communicates again. Power supply 5 volts extra included. Idea?
Sorry about bad english.

[image]
[image]
[image]

tobox, 2018-05-07:
I had exactly the same problems. I got the feeling that the MHZ19 is very picky with software serial ports. Running it on the hardware serial ports of devices seems to much better. After patching Tasmota to call Serial.Swap I was able to get it running stable with a Wemos using the alternative RX/TX ports.

orimate, 2018-05-07:
Hi, thanks for the answer, then I'm not alone I calmed down. :) I have always used the RX / TX line. Where should I perform the patch?

arendst, 2018-05-08:
Observed too.

tobox, 2018-05-08:
If you are using the Sonoff TH, you should already be connected to the hardware serial, if I remember correctly. The problem occurred for me when using the wemos, where hardware serial is mapped to the USB port. On the wemos, I solved the problem by calling serial.swap. 

Note that you need to deactivate logging so serial out completely if you connect the MHZ!

FabianHM, 2018-05-08:
I also noticed the behavior

orimate, 2018-05-08:
Now the test runs with 5.13 disabled SerialLog on NodeMCU board via TX / RX port. (GPIO1/3)
@tobox 
"The problem occurred for me when using the wemos, where hardware serial is mapped to the USB port."<= I understood.
"On the wemos, I solved the problem by calling serial.swap." <= Unfortunately, I'm not a programmer (not noob, but beginner) and I still do not know where I would like the serial.swap calling if I do not want to use sonoff.

tobox, 2018-05-08:
I modified the Tasmota Source code to call Serial.Swap before initializing the MHZ driver. I don't remember exactly how I did it. I just compiled the firmware and it worked, so I did not bother to keep the source code of that file - sorry...

arendst, 2018-05-08:
Wait! I will release an updated version today which might solve the lost info. Longer term testing is needed.

orimate, 2018-05-08:
Ok I'll wait. After 2 hours result is equal to = 0ppm. :(

arendst, 2018-05-08:
Give it a try let me know your results.

arendst, 2018-05-08:
Still fails on me :-(

orimate, 2018-05-08:
On me since 4 hours is working. I let it run at night.

orimate, 2018-05-09:
Afer 10 hours 0 ppm :(

arendst, 2018-05-09:
Currently working on an optional hardware serial solution and see how that performs...

arendst, 2018-05-09:
You might want to try the hardware serial option which is auto-enabled when the below settings are used:

[image]

FabianHM, 2018-05-09:
Are the settings identical for the model MHZ-19B as well? In the past I often had the problem that my "B" model was wrongly recognized as MHZ-19.

orimate, 2018-05-09:
I have B.

arendst, 2018-05-09:
The settings are the same. The software should detect the kind of MHZ, either 19 or 19B depending on result of serial data.

orimate, 2018-05-10:
After 6 hours = 0 ppm :(
I send 4  pictures what is my config. I hope that is not my fault.

[image]
[image]
[image]
[image]

tobox, 2018-05-10:
I updated the firmware, rewired the Wemos and it is running for 3 hours without problems. I will keep you updated.

orimate, 2018-05-10:
OHH....swapped.

FabianHM, 2018-05-10:
After 3 Tests it took less than 15 minutes and it was 0 ppm....
My settings are, as before, this:
[image]

orimate, 2018-05-10:
@FabianHM You need to swap the TX/RX to use Hardware serial!

FabianHM, 2018-05-10:
I have swapped the cable connections (Rx. ->Tx) and thought that the same effect was achieved. 
I test again with changed settings

orimate, 2018-05-10:
I forgot to swap too.
[image]

arendst, 2018-05-10:
See here (https://github.com/arendst/Sonoff-Tasmota/wiki/What's-New#20180510---5131a-with-tasmotaserial-200) for other sensor information

FabianHM, 2018-05-10:
After I clarified my understanding problem with the configuration I tested with a Wemos. After 1,0 hours, a completely new problem has occurred. The GUI is frozen and the web interface is no longer accessible. Multiple on and staking of the wemos has brought nothing, the Webgui is no longer available.

tobox, 2018-05-10:
@FabianHM I saw similar problems when compiling with Core 2.4.0, does the problem persist when using 2.3.0?

@arendst My CO2-Sensor is still running fine with your latest changes. With the previous firmware it has never lasted that long without problems.

tobox, 2018-05-12:
Still running without any problems. One less custom firmware that can be replaced with stock tasmota! Thanks for the great work!

grad-michal, 2020-01-19:
Anybody has this problem now?
I have exact problem as described in this thread.

I have MH-Z19B module connected through hardware serial with configuration like this:
[image]. 
Relay and switch configuration are always enabled as I removed relay module and power sensor from 5V line there.

According to link provided in previous post this should enable hardware serial and judging from console log it does.
> See here (https://github.com/arendst/Sonoff-Tasmota/wiki/What's-New#20180510---5131a-with-tasmotaserial-200) for other sensor information

Sensor was working fine on custom firmware on esp8266 using MHZ library for over a week.
On tasmota and Sonoff Basic R1 it stop working after a couple of hour. 
After powering down and up the sensor (ex. using relay) it is working again until next freeze.

tobox, 2020-01-20:
A friend of mine had similar problems. The reason was:

- the MH-Z19B was connected to the 5V pin of a Wemos D1 Mini
- his USB Power supply delivered approx. 4,5V (which is within the specs, IIRC)
- the diode inn the 5V line of the Wemos D1 Mini lead to approx. 4.0V on the 5V pin
- the MH-Z19B does not run properly with undervoltage

He switched to a better USB power supply and the problem disappeared.

grad-michal, 2020-01-20:
Thanks, I will invastigate it further.
Sonoff has nice 5V±3%, but when sensor is connected voltage drop to 4.5V and even to 4.0V when IR diod is firing at sensor. 
I was not expecting that as avg power consumption of sensor should be 19mA (yeah I know - avarage) and relay coil is rated at 200-400mA depending on model.

I will try to add some capacitors and let you know. 

Nevertheless I will try to modify tasmota so that sensor read interval could be changed. 
Currently it is 8s and default MQTT push is 300s so in many scenarios there no point to read data this often (I imagine this could improve expected life time of sensor).
Also as understand sensor specification there should measurement should be taken after few minutes after sensor heats up.

SchiFrank, 2020-10-30:
Hi all,
my config is exact like here: [https://tasmota.github.io/docs/MH-Z19B/](url)
same problem. 0° and 0ppm from the start. I tried different things:
- Check contacts of cables -> no problem
- Tried 3 different USB Power -> all with no better result
- Changed Tx/Rx (cable and entrys in configuration, all combinations)

I used the newest tasmota_sensors.bin

Can I somehow check, if the sensor is working? Perhaps the sensor is defect....

Any further Ideas? What should I try?

Best regards,
Frank

SchiFrank, 2020-11-01:
Can anybody help? I feal alone...

GerhardGrasinger, 2020-12-04:
I am not using tasmota but an ESP32 Module and i have the same problem.
I purchased two MMZ 19 sensors. One is working very well.
The other one is crashing afer some hours of working. Sometimes after 8 hours , sometime after two days.
The MHZ 19 does not answer and that results in showing 0ppm by the driver.

What i tried without success:
-Support capacitors 10u Tantal and 470u Aluminium 
-An external power supply
-Pullup resistors 2k2 on TxD and RxD
-Calling the CO2 reading function only in quiet periods with no other Software activity running in parallel

At the moment i give up.

robert777robert777, 2021-03-16:
> I am not using tasmota but an ESP32 Module and i have the same problem.
> I purchased two MMZ 19 sensors. One is working very well.
> The other one is crashing afer some hours of working. Sometimes [...]

I had the same problem.
I;ve tested with arduino with the library https://github.com/WifWaf/MH-Z19 And the sensor DOES work.
For Tasmota the solution was to connect the sensor to the hardware serial pins (sensor TX->RX2, sensor RX->TX2) and in tasmota select 
IO GPIO16 MHZ Rx
IO GPIO17 MHZ Tx

I've also used the console command Sensor15 0 
So I'm not sure what actually solved the problem, but it DOES work now

roblomq, 2021-04-10:
I have also problems with the MHZ-19. I use also a ESP32 board, the sensor has run for a few hours and gave expected readings but after that it went to 5000ppm and after a reboot it got 0ppm and stayed there. 
- Tried swapping TX and RX on hardware and software (and tried all combinations.
- Powered the MHZ-19 with external 5V power supply.
- Tried other GPIO pins.
Somebody an idea?

wytr, 2022-07-23:
> I have also problems with the MHZ-19. I use also a ESP32 board, the sensor has run for a few hours and gave expected readings but after that it went to 5000ppm and after a reboot it got 0ppm and [...]

Did you resolve the problem? I already tried the calibration on it. I got two sensors shipped from the same seller on aliexpress. One works fine, and the other one is only giving me the 0ppm co2 value. The temperature is "fine" though, quite inaccurate but within 5 degrees. I remember this one drifting to 5000ppm before the calibration attempt.

jjifa01, 2025-02-17:
I have the same problem with the MH-Z19B I bought from AliExpress on ESP32. I read 0ppm without getting over 24 hours, and if I only RESET the power of the MH-Z19B, it comes back to normal right away. I delayed the request from 4 seconds to 9 seconds, but it didn't get better. Did I also buy a defective one?

orimate, 2025-02-17:
I made this module many years ago. It just run without any errors. Important is: 
-5V stable supply, not from ESP. Sonoff has stable 5V for the relay.
-soldered contact, jumper wire connection are not so good, 
-and of course communication goes over Hardware serial.  

[image]

[image]