arendst/Tasmota #2660: Unable to upload via web v5.13

rajkedda, 2018-05-07:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ X] What hardware you are using  Sonoff Basic
- [X ] Provide the output of command ``status 0``
Upload Failed
Upload buffer miscompare

- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

ascillato, 2018-05-08:
Hi,

Buffer Miscompare is when your firmware is bigger than the free space on your device.

Please, upload the **sonoff_minimal.bin** file and after upload the firmware one you need.

rajkedda, 2018-05-08:
Hi
  I followed your recommendation, uploaded minimal and the 5.13.  Upload successful and rebooted.  It never came back online.  I waited for 5 minutes, and manually restarted the device, still no luck.  Any thoughts on how to recover and get a copy of v5.12  bin file?

rajkedda, 2018-05-10:
Thank you.  I was able to flash it back with hardwire to 5.12.  But on Sonoff Basic, even after doing minimal and then 5.13.1, it gives me same error and will not upload the latest version.  It is same issue with POW as well.

closed 2018-07-09 by stale[bot]