arendst/Tasmota #2661: Bot failure

davidelang, 2018-05-07:
the bot is attempting to close issues and failing, and looping around and trying again.

please fix bot permissions or disable bot

arendst, 2018-05-08:
I think the bot is doing fine. There are just a lot of issues to close.

ascillato, 2018-06-04:
Hi all,

The bot is now running ok.

What do you think of changing the stale time from 120 days to 45 and then, the closing to 15?
So, if an issue has no activity from 45 days, the bot will call the attention to it, and if nothing happens in 2 months, the bot will close it. Now, an issue has to be without activity for 134 days (4 months and half) to be closed.

I think that if an issue if left behind without activity for 2 months, it means that has been solved, has no easy solution or is forgotten and no more interest is on that issue. It is just an idea to clean issues and make a better issue following.

What do you think?

Modification proposed:
```
# Number of days of inactivity before an Issue or Pull Request becomes stale
daysUntilStale: 45

# Number of days of inactivity before a stale Issue or Pull Request is closed.
# Set to false to disable. If disabled, issues still need to be closed manually, but will remain marked as stale.
daysUntilClose: 15
```

closed 2018-06-05 by ascillato