arendst/Tasmota #2663: Sonoff Dual R2 and VPN

opendore, 2018-05-08:
Thanks everyone in advance for the help.
I have a Dual R2 Sonoff connected to my house shutter and it works perfectly (last Tasmota Firmware).
I would like to be able to manage it from my mobile phone via VPN connection with Router.
I have a Netgear that allows me to make a VPN connection that I already use for something else.
The problem is that from my mobile phone I can ping the Sonoff but the display is not perfect .... you can see lines of code and some pieces of the menu. Why? What am I doing wrong? Even if I try from my laptop connected via VPN I get the same result.
Any suggestion?

wongnam, 2018-05-08:
Please use the Tasmota source code with esp8266 core version 2.3.0 and compile yourself. Sonoff web will display correctly via VPN.

opendore, 2018-05-08:
Thanks for help...Can you explain the steps....
It's not possibile to find .bin and import by OTA?

wongnam, 2018-05-10:
Also refer to below link there is the same situation.
https://github.com/arendst/Sonoff-Tasmota/issues/2685

gd-99, 2018-05-24:
Thanks @wongnam. Following your advice I can confirm my Sonoff Dual R2 works correctly over VPN.
My device had been flashed with:
*Tasmota development-5.14.0a
*Arduino IDE v1.8.5
*ESP8366 Community Core v2.4.1
* IwIP v1.4
Every thing worked very well while on my local network. When I tried accessing my local network remotely over VPN I was unable to access the Tasmota web interface. I was able to access other devices on the same local network via the VPN.

Returning to the local network (both tablet and sonoff on the same network) I was still unable to connect to the Sonoff Dual - incomplete packet I think was the error. Fortunately I had implemented a restart rule, to kick in at midnight. The following day I was then able to connect to the Sonoff Dual web interface.

I have now re-flashed the Sonoff Dual R2 with:
*Tasmota development-5.14.0a
*Arduino IDE v1.8.5
*ESP8366 Community Core v2.3.0

No other changes made other than the ESP8266 core version. Everything works well on the local network including the reboot /restart rule. I have since tested the set-up remotely over the VPN and this too now works as expected.

From my experience using the Sonoff Dual R2 I think I would recommend compiling using the ESP8266 core v2.3.0. I do not know what deference there are between the ESP8266 core versions. The only observable difference I notice as a user - apart from the VPN working - is it takes a touch longer for the Sonoff to connect to my AP. Serial output shows it usually misses the first attempt but then after a 3 min pause connects correctly.

Hope this is of help so someone.

wongnam, 2018-05-24:
I glad to hear you can solve the vpn issue. Enjoy!

BTW, remember to close the case. :)

opendore, 2018-05-24:
Yes!!!!, With the last Firmware is all OK!

closed 2018-06-08 by ascillato2

peret2000, 2018-06-30:
I am experiencing the same issue. Sadly, with the last firmawe (I guess 6.0.0b) and using last expressif8266 available (1.7.3) does not work.
Following this issue, I have linked with ESP8266 2.3.0, but I get a compiler error:
"undefined reference to `__ieee754_sqrt'"

I have removed all the packages, as I have understood in some forums that maybe they are not downloaded again although its hash changes, but with no success.

peret2000, 2018-07-01:
I noticed I was using development branch. Once I used v5.14 tag, I could compile using "platform = espressif8266" in platformio.ini with success.