arendst/Tasmota #2664: [Update] SDS011 PM2.5 PM10 Laser Dust Sensor added to Tasmota Wemos Wiki

RaymondMouthaan, 2018-05-08:
Hai All,

Just to inform you guys, the [Tasmota Wemos Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini) has been update for Nova Fitness SDS011 PM2.5 PM10 Laser Dust Sensor. 

Thanks to @FabianHM for providing the screenshots and info about this sensor. 👍

joba-1, 2018-05-09:
Cool, just in time for me. My SDS011 just arrived. Hooked it up and it works perfectly.

Just one question: It looks like PM2.5 and PM10 values are prepared to be delivered to domoticz, but what kind of virtual senor do I need to create in domoticz? I could not find something that matches, but maybe my domoticz (Version 3.4834 (March 2nd 2016) ) is simply not new enough?

arendst, 2018-05-10:
Using Domoticz 3.8153 I'm using Virtual Sensors of type Custom Sensor for each value and select ug/m3 as unit.

joba-1, 2018-05-11:
Hi @RaymondMouthaan 

Although the measurement works fine, I have a suggestion:

according to the manufacturer, the laserdiode in the SDS011 has a lifetime of only 8000h. This is less than a year if running non-stop. I think it would be a good idea to only start the sensor if needed, e.g. for the tele interval. Or let it run for ~10s, to get fresh air, then taking the mean value of 5 samples over the next 5s or so and then stop again for 1 minute and just serve a cached value should be enough to extend the lifetime to a few years.

Those guys near my hometown have nice code that does something like that, if you need an example:

https://github.com/opendata-stuttgart/sensors-software/blob/master/airrohr-firmware/airrohr-firmware.ino#L534

Drawback: you need rx _and_ tx to send commands and receive results...

RaymondMouthaan, 2018-05-11:
@joba-1 just to be clear, I am not the developer of the code for this sensor, @arendst is. I do not even have this sensor (yet). I will order it soon. I only added it to the Wiki base upon the information provide by @FabianHM. 

Maybe Theo can have a look at the link you posted to improve the code for longer usage of this sensor of you might consider a PR yourself? As for me, I think it's a nice suggestion you wrote. 👍

FabianHM, 2018-05-12:
I also think this is a great Idea. Currently, I control the power supply of the SDS011 via an additional Sonoff Basic module, which delivers 5 minutes of measurements on the hour, which is not a nice solution.
Would it perhaps be possible to use the new timers from Tasmota?

closed 2018-06-04 by ascillato