arendst/Tasmota #2666: Sonoff 4CH PRO Delay

TheCellMC, 2018-05-08:
Hi,
I wan't to use external 2-way switches with my Sonoff 4CH PRO. I went ahead and soldered wires to the buttons so that my 2-way switches. However, when I turn on my light, I get a 1 second delay. If I leave the switch on the a while, I don't get a delay when turning it off.
How can this be fixed?

Leseratte10, 2018-05-08:
I guess that is due to the RF chip interpreting the button hold as a pair signal. If you solder directly to the contacts of the buttons, you have to use momentary push buttons, no 2-way switch.

TheCellMC, 2018-05-08:
I don’t think that this has to go with RF learning as that is only triggered when you double press the button. However this delay occurs anytime I change the state of the switch.

Before, the switch would act weirdly with it’s multifunctions. So I changed th switches from KEY to SWT in the configuration.

Now I am able to only use TOGGLE but I always get a delay. Is there a setting to stop this? Has this got to do with a HOLD function?

TheCellMC, 2018-05-09:
Any Ideas? Is this related to the 4CH PRO only?
Could something like ESPEASY solve this?

Jason2866, 2018-05-09:
Upload status 0 and a log when the device is switched

closed 2018-06-23 by TheCellMC