arendst/Tasmota #2667: [Update] RCWL-0516 Microwave Radar Sensor added to Tasmota Wemos Wiki

RaymondMouthaan, 2018-05-08:
Hai All,

Just to inform you guys, the [Tasmota Wemos Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos%20D1%20Mini) has been updated for RCWL-0516 Microwave Radar Sensor.

Greetz!

MatzeJoerling, 2018-05-09:
working well for me! 
I dont log false positives, also dont care. 
it makes the light on and off, and if it is on one more time ...  

i would not use it for alarm system!!!  

check this: https://github.com/jdesbonnet/RCWL-0516

RaymondMouthaan, 2018-05-17:
No false positives here. I prefer this accurate sensor over the HC-SR501 (PIR).

closed 2018-05-17 by RaymondMouthaan

aihysp, 2018-07-02:
can someone please tell me what sensor to put under this GPIO?
i am getting allot of issues with Switch1-4