arendst/Tasmota #2669: Rule to set relay ON between sunset and sunrise ешьу after power loss.

dmitrybabeshko, 2018-05-08:
Hello,

I use Sonoff basic as automatic standalone day/night switch for street lightning using approach with Sunset/Sunrise for timers.
The problem that after power loss I need to set correct relay state. For example if power loss happened at night I need to understand whether it restored at night or at day to switch relay ON or OFF.
Is it possible to create Rule which will trigger between sunset and sunrise to switch relay On?

Thank you in advance!

curzon01, 2018-05-08:
If you set PowerOnState to default Sonoff will restore the last state after power loss automatically, no necessary to add something

dmitrybabeshko, 2018-05-09:
PowerOnState won't work as expected.
For example:
Sunrise 5:00
Sunset 20:00
1) 5:00 Sunrise timer triggers - OFF
2) 19:55 Power loss
3) 20:05 Power restored - Last state OFF
I need street lights ON during nighttime.

Or viseversa
1) 20:00 Sunset timer triggers - ON
2) 4:55 Power loss
3) 5:30 Power restore - Last state ON
No need in ON street lights during daytime.

The problem that we have several power loss/restore combinations and PowerOnState can't cover all of them.
Loss/Restore

Daytime/Daytime
Daytime/Nighttime
Nighttime/Nighttime
Nighttime/Daytime

curzon01, 2018-05-09:
What you need is a retrigger on startup for all last non-trigger.

Well, as enhancements like that are more or less special expections and would be better handled by a third system. I don't think that all these exceptional user requirements can be handled by such small device in future (for your case the system needs to "remember" which trigger are already triggered), my suggestion for your need is a workaround: use a home system like Fhem which can handle retrigger of non-triggered events.

ascillato, 2018-06-08:
Hi,

Now, in rules there are the commands `VAR1` to `VAR5` and `MEM1` to `MEM5` to store temporary and permanent values. And then, you can use them as `%var1%` and `%mem1%` inside rules.

What about adding new values that report :

* `%time%` : actual elapsed time since midnight in minutes
* `%sunrise%` : sunrise in minutes 
* `%sunset%` : sunset in minutes
* `%uptime%` : actual uptime in minutes

So, with those, we could make a rule like:

For example: If we want a light to be on from 20:00 to 8:00 and to be off from 8:00 to 20:00
So, _8 hr = 8 x 60 = 480 minutes_, and _20 hr = 20 x 60 = 1200 minutes_

Rule:

```
on Time#Initialized do backlog power1 off; event check=%time% endon
on event#check>1200 do power1 on endon
on event#check<480 do power1 on endon
```

What do you think?

ascillato, 2018-06-08:
@arendst what do you think?

arendst, 2018-06-25:
@ascillato Don't you think these time related values are already served by the the timer options already available? This seems to duplicate the timer functionality.

ascillato, 2018-06-25:
Hi

The **timers** will be excecuted as a **trigger**, while `%time%` will be as a **conditional**.

So, for the issue explained by dmitrybabeshko, if the sonoff is not powered while the timer should trigger, then, when it is powered up, it will not trigger the output or rule that was intended.

With a rule with `%time%` at `Time#Initialized` for example, the condition of reaching or passing that trigger time can be checked, so the output or rule can be trigger and it is not going to be missed.

I think that adding this conditions to the actual timers will make them more complex with more options and with more memory/flash use. That's why, I think that having some extra internal time variables exposed to rules will be more flexible and less code hungry.

Another use, besides turning lights at night and do not miss it if the energy was down during the trigger time, can be change the behavior of the sonoff in different hours, also without missing the trigger time.

closed 2018-06-27 by ascillato2