arendst/Tasmota #2670: How to configure mqtt and wifi using mqtt?

aaamitsingh, 2018-05-08:
Hi guys,
For some security reason, I have disabled the web server in sonoff devices and it is not possible to reflash the devices or configure the device every time using serial because the device is going to be placed where access is not allowed at every time. So I want to know is there any possible way to configure the below thing using mqtt:

//Mqtt setting
- [ ] MQTT_HOST
- [ ] MQTT_PORT

//Static IP Wifi
- [ ] WIFI_IP_ADDRESS
- [ ] WIFI_GATEWAY
- [ ] WIFI_SUBNETMASK
- [ ] WIFI_DNS

//Alternet Wifi 1
- [ ] STA_SSID1
- [ ] STA_PASS1

//Alternet Wifi 2
- [ ] STA_SSID2
- [ ] STA_PASS2

arendst, 2018-05-08:
How about the wiki - commands.

arendst, 2018-05-13:
look for command ipaddres1

Keep in mind that changing the ip address you will loose your connection at the next restart for obvious reason.

You probably also need to change the gateway and netmask addresses so again take a look at the other ipaddress options.

All commands can be given in one go using starting with command backlog. Again, see wiki command usage webpage.

ascillato, 2018-06-04:
Hi

If your issue is solved, please close it. Thanks :+1:

closed 2018-06-13 by ascillato2