arendst/Tasmota #2671: Non-Blocking MQTT Fails

dennyfmn, 2018-05-08:
Hardware: NodeMCU 4MB
Arduino IDE 1.8.5 Core 2.4.1 Generic ESP8266 Module 1MB no spiffs

15:09:01 CMD: status 0
15:09:01 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
15:09:01 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.3:80/api/arduino/sonoff.ino.bin","RestartReason":"External System","Uptime":"0T00:02:21","StartupUTC":"2018-05-08T20:06:40","Sleep":0,"BootCount":49,"SaveCount":145,"SaveAddress":"FB000"}}
15:09:01 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.13.1a","BuildDateTime":"2018-05-08T14:59:29","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
15:09:01 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"192.168.1.29","LogPort":514,"SSId":["microwap2",""],"TelePeriod":60,"SetOption":["00008129","55818000"]}}
15:09:01 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":495,"Free":508,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
15:09:01 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1350","IPAddress":"192.168.1.166","Gateway":"192.168.1.240","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.240","Mac":"60:01:94:7B:C5:46","Webserver":2,"WifiConfig":3}}
15:09:01 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.4","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7BC546","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:09:01 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue May 08 20:09:01 2018","Local":"Tue May 08 15:09:01 2018","StartDST":"Sun Mar 11 02:00:00 2018","EndDST":"Sun Nov 04 02:00:00 2018","Timezone":99,"Sunrise":"05:52","Sunset":"20:25"}}
15:09:01 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-05-08T15:09:01"}}
15:09:01 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-05-08T15:09:01","Uptime":"0T00:02:21","Vcc":2.993,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"microwap2","RSSI":38,"APMac":"10:C3:7B:D1:FE:D0"}}}

Both non-blocking libraries

MQTT_LIBRARY_TYPE      MQTT_TASMOTAMQTT and
MQTT_LIBRARY_TYPE      MQTT_ESPMQTTARDUINO

allow Tasmota to start properly if the MQTT server is not running. Once the MQTT server is started Tasmota connects to it publishes topics. If the MQTT server is stopped, Tasmota senses the disconnect and reconnects when the MQTT server is started again. However, this happens only once or maybe twice. The second time the MQTT server is stopped, Tasmota just keeps publishing topics as if the MQTT server were connected. So with the two non-blocking drivers, startup is non-blocking, but reconnecting only works once or maybe twice.

I have searched the GIT Hub issues and found a couple of postings that had to do with MQTT. I have changed from using the name of the MQTT server to using the IP address. I have also put
#undef  USE_DISCOVERY
#undef  MQTT_HOST_DISCOVERY
in my user_config_override.h file
These things did not help.

I have a couple other MQTT servers on my network, and I checked to make sure that the NodeMCU was not somehow connecting to them. It is not. If I reset or power down the unit, it works as described. Detects MQTT disconnect once, then does not. The cycle is very repeatable.

These messages are from the Console after the MQTT server was stopped at 15:38. The driver is MQTT_TASMOTAMQTT

15:38:48 MQT: stat/sonoff/RESULT = {"POWER2":"ON"}
15:38:48 MQT: stat/sonoff/POWER2 = ON (retained)
15:38:49 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
15:38:49 MQT: stat/sonoff/POWER2 = OFF (retained)
15:39:03 MQT: tele/sonoff/STATE = {"Time":"2018-05-08T15:39:03","Uptime":"0T00:01:13","Vcc":2.993,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"microwap2","RSSI":52,"APMac":"10:C3:7B:D1:FE:D0"}}

If I use MQTT_LIBRARY_TYPE      MQTT_PUBSUBCLIENT
Tasmota will not start properly on power up or reset if the MQTT server is not running. However, once the MQTT server is started, it can be stopped and started as many times as you wish and Tasmota will sense the disconnect, and reconnect appropriately. So with MQTT_PUBSUBCLIENT startup is blocking, but reconnecting works properly.

I am carefully testing this because I am working on an ESP8266 based project for a dual channel thermostat that must retain its full local functionality whether the network or the MQTT service are present or not. If the network or MQTT server are up, MQTT topics will be published to the broker. If the network or broker are down, the thermostat must still function normally. I am hoping to use one of the non-blocking MQTT libraries in my sketch. I have researched non-blocking MQTT on the net, and the libraries in Tasmota are the closest thing that I have found that may be workable.

The non-blocking WiFi code I found at 
http://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/generic-examples.html
works well. Now I have to figure out a way to do non-blocking MQTT.

Any help with this will be greatly appreciated!

ascillato, 2018-05-09:
Hi,

Tasmota has 3 MQTT Library to choose. The only that have the blocking behaviour is PUBSUBCLIENT.

Please, select any other in the file user_config.h.

ascillato, 2018-05-09:
I use TasmotaMQTT library and it works really well.

line 186 on _user_config.h_

```
/*-------------------------------------------------------------------------------------------*\
 * Select ONE of possible three MQTT library types below
\*-------------------------------------------------------------------------------------------*/
  // Default MQTT driver for both non-TLS and TLS connections. Blocks network if MQTT server is unavailable.
#define MQTT_LIBRARY_TYPE      MQTT_PUBSUBCLIENT   // Use PubSubClient library
  // Alternative MQTT driver does not block network when MQTT server is unavailable. No TLS support
//#define MQTT_LIBRARY_TYPE      MQTT_TASMOTAMQTT    // Use TasmotaMqtt library (+4k4 code, +4k mem) - non-TLS only
  // Alternative MQTT driver does not block network when MQTT server is unavailable. No TLS support
//#define MQTT_LIBRARY_TYPE      MQTT_ESPMQTTARDUINO // Use (patched) esp-mqtt-arduino library (+4k8 code, +4k mem) - non-TLS only
```

dennyfmn, 2018-05-09:
Thanks for your comment I am well aware of how to select which library to select.

 

I am reporting that the two libraries that are supposed to be non-blocking are not reconnecting properly after the first disconnect.

 

Please read my posting again.

ascillato, 2018-05-09:
oh !,

Sorry !

when I first see the issue, the page started at

>  If I use MQTT_LIBRARY_TYPE MQTT_PUBSUBCLIENT .......

Now I saw all your explanation. Today I had some problems with my internet connection.

I tested the TasmotaMQTT Library for reliability for my home. In my case is working fine for connection and reconnection. For both: a reset of Tasmota and for a reset of broker. I'm using the HBMQTT broker (the embedded broker in HomeAssistant)

Which mqtt broker are you using?

One issue I found is that sometimes this broker takes 1 or 2 minutes to connect after a tasmota reset. But this issue happened to me on all 3 libraries and I think is an issue of the broker because if I reset the broker all Tasmota's devices are connected at once.

I compiled with esp libraries v2.3 with lwIP 1.4. I don't know if the problem you are experiencing is because you are using v2.4.1? On my tests, several problems show up with this new library.

dennyfmn, 2018-05-09:
I am using mosquitto Version: 1.4.15-0mosquitto1 running on a Raspberry Pi 3B with Stretch.

 

For testing purposes I am using “/etc/init.d/mosquitto stop” and “/etc/init.d/mosquitto start” to take the broker off and on line. There are no other MQTT clients connecting to this broker on my network, so I can shut it down without disrupting anything that is in production.

 

I am monitoring topics with a python script I wrote that uses paho.mqtt.client. This script reports disconnects when I shut down mosquitto, and does take a little while to reconnect when I start mosquitto back up.

 

The issue with both non-blocking MQTT libraries in Tasmota is that they both fail to see the broker go off-line a second time. In the Console you can see Tasmota is still sending topics to MQT even though the broker is off-line. I have let this go for many minutes and it never sees the broker is gone.

 

Your comment about core 2.4.1 being buggy is interesting. I will set up a 2.3 environment to test.

 

How can you tell what version of lwIP is being used?

dennyfmn, 2018-05-09:
Very interesting...
I downgraded the esp8266 core to 2.3 and recompiled. Now tasmota 5.13.1a with the tasmota non-blocking MQTT library works properly. I stopped and started the broker four times and tasmota saw the disconnect and reconnected perfectly when I started the broker back up.
I had to use a different board definition since there was not a choice for generic 1MB no spiffs. I used nodemcu 1.0, 4M (1Mspiffs). The option to directly upload to the http OTA server was also gone. I had to manually copy the new .bin file to the OTA server.
I'll try it with 2.4.0 later today and report on the results.
Thanks for your help!

Jason2866, 2018-05-09:
Another reason to stay at 2.3.0  -> wifi handling is buggy in 2.4.0 too