arendst/Tasmota #2672: Garbled output on Power Status command

miraixyz, 2018-05-09:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
 **Sonoff Basic**
- [ ] Provide the output of command ``status 0``
01:37:29 RSL: stat/kitchen/STATUS = {"Status":{"Module":1,"FriendlyName":["kitchen light"],"Topic":"kitchen","ButtonTopic":"���������������������������������sonoffs","Power":0,"PowerOnState":255,"LedState":255,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
01:37:29 RSL: stat/kitchen/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.12.0/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:50:46","StartupUTC":"2018-05-08T23:46:43","Sleep":0,"BootCount":248,"SaveCount":538,"SaveAddress":"F9000"}}
01:37:29 RSL: stat/kitchen/STATUS2 = {"StatusFWR":{"Version":"5.13.1","BuildDateTime":"2018-05-01T20:59:22","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
01:37:29 RSL: stat/kitchen/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"127.0.0.1","LogPort":514,"SSId":["Mirai_NetSB","MiraiSB2"],"TelePeriod":65535,"SetOption":["00008029","55818000"]}}
01:37:29 RSL: stat/kitchen/STATUS4 = {"StatusMEM":{"ProgramSize":529,"Free":472,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
01:37:29 RSL: stat/kitchen/STATUS5 = {"StatusNET":{"Hostname":"Sonoff 1","IPAddress":"192.75.42.37","Gateway":"192.75.42.252","Subnetmask":"255.255.255.0","DNSServer":"192.75.42.252","Mac":"DC:4F:22:2D:3A:2B","Webserver":2,"WifiConfig":2}}
01:37:29 RSL: stat/kitchen/STATUS6 = {"StatusMQT":{"MqttHost":"192.75.42.16","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2D3A2B","MqttUser":"mirai","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
01:37:29 RSL: stat/kitchen/STATUS7 = {"StatusTIM":{"UTC":"Wed May 09 00:37:29 2018","Local":"Wed May 09 01:37:29 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:16","Sunset":"20:16"}}
01:37:29 RSL: stat/kitchen/STATUS10 = {"StatusSNS":{"Time":"2018-05-09T01:37:29"}}
01:37:29 RSL: stat/kitchen/STATUS11 = {"StatusSTS":{"Time":"2018-05-09T01:37:29","Uptime":"0T00:50:46","Vcc":3.351,"POWER":"��������������������������������������������������������������������������������������������������������������������������������������","Wifi":{"AP":1,"SSId":"xxxxxx","RSSI":80,"APMac":"xxxxxxxx"}}}
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

As you can see by my output results, the power status of my Sonoff Basic on Tasmota firmware version 5.13.1 shows strange illegible characters.   As a result, my home assistant is not able translate the result and report back the accurate state of the switch.   Has anyone else experienced this?  If so, what is the fix.  I how I don't need to reflash.

thanks

miraixyz, 2018-05-09:
Managed to resolve it on my own....thanks

curzon01, 2018-05-09:
Fine, so pls help the issue list keeping clean as possible and please close the issue

closed 2018-05-09 by miraixyz

miraixyz, 2018-05-10:
Sorry....I forgot to write how I fixed it.

I simply used the console to send a StateTopic command.
I found that both the "ON" and "OFF" values in the firmware were somehow corrupted,  Not sure why or how it happened.  To confirm this I typed "StateText1" and "StateText2", which should normally return "OFF" and "ON", but it was returning the above mentioned garbled characters instead.  Hence, my HomeAssistant server was unable to read the power state of my Sonoff Basic.

The fix was easy.   Just type "StateText1 OFF" and "StateText2 ON" at the console prompt, and it changed/corrected these values to what they should be.

So with this known, if you have a home automation system that requires a different value other than "ON" and "OFF" to read it's power state (like "0" and "1", for example) you can change the output value using the StateText command.

thanks

Frogmore42, 2018-05-10:
That certainly is strange.  Glad you were able to figure it out, and a good reminder to check everything, since corruption is a possibility.