arendst/Tasmota #2673: Less wifi sensibility with latest firmware

gabryk91, 2018-05-09:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
Sonoff basic

I updated the firmware to the last one and I noticed an important reduction of WiFi sensibility. I have a sonoff pretty far away from the router, with the previous firmware I didn't have a problem, it was connecting fine. Now it doesn't, I have to plug it almost in the same room of the router...

There is something I can try.
As least resource, how can I downgrade the firmware? Thanks

JF62, 2018-05-09:
Same here
5.13.1.a -> 54 %
5.12.xxx -> 82%

arendst, 2018-05-09:
So compile with ESP8266/Arduino core 2.3.0 and let us know.

gabryk91, 2018-05-09:
@arendst I used Sonota.exe and I don't know how to do that

ajanulis, 2018-05-09:
Same with my devices. Went from 5.12.0 to 5.13.1a - almost impossible to connect to WiFi. Described functionality with 3 or 4 short presses did not bring up AP. I was lucky enough to connect to wifi only when powered through USB FTDI few times but newer on AC (using SONOFF basic). Still have one working (based on Wimos mini) but on 5.12.0 - if it can help - I can reflash it as well. Using Atom & Platformio.

gabryk91, 2018-05-09:
@ajanulis what is the procedure to flash the recompiled firmware? Serial connection is needed?
Also, is possible to revert to 5.12 from the web server or sonota.exe?

ajanulis, 2018-05-09:
@gabryk91, than it's easy fix :) Do you know what address it gets from the your router? If yes - open web browser, enter that address and you will be provided with an option for "Firmware upgrade"

gabryk91, 2018-05-09:
@ajanulis yes ok, just I didn't know if "Firmware upgrade" would also replace it with an older release. With Firmware upgrade can I also load the firmware compiled through Arduino IDE? I followed the wiki to compile the latest firmware with ESP8266 v2.4.1 that you suggested working fine

ajanulis, 2018-05-09:
@gabryk91, I am pretty sure that there is no upgrade/downgrade restrictions as in an Apple world :) At least I strongly believe in such a freedom :)

If you already compiled and have firmware.bin file - just pass it to web browser and you are happiest guy! :) Just try it

gabryk91, 2018-05-09:
It returns "Upload Failed:Upload buffer miscompare". There's a chance you can share your build?

gabryk91, 2018-05-09:
Ok, maybe you can help me flying by myself :-)
I setup the Arduino environment for compiling and send the firmware OTA but I get the error 
  File "C:\Program Files (x86)\Arduino\portable\packages\esp8266\hardware\esp8266\2.4.1/tools/espupload.py", line 16, in <module>
    import pycurl
ModuleNotFoundError: No module named 'pycurl'
ModuleNotFoundError: No module named 'pycurl'

Do you know what it means? I already installed python and pycurl in the right directory

ajanulis, 2018-05-09:
@gabryk91, I'd love to teach you to fly, but I am a learner as well. I am not a big fan of Arduino and use it only when there are no other options. And with Tasmota I learned (at least a bit) how to use Atom and Platformio and I like it. So if you'll decide to do the same, most probably I can guide you to the same result where I am now.

But have you tried:

pip -v install pycurl --upgrade

gabryk91, 2018-05-09:
Ok, I managed to do it (in the end it was missing the arduino folder inside the webserver)
But doesn't feels like much of an improvement compared to the official 5.13.1
Wifi signal is at 62%

Tomorrow I will try to revert back to 5.12 to see the difference

ajanulis, 2018-05-09:
Hmm, happy you did it and strange - I got signal fully at the previous level, and I suppose it's not 5.12 or 5.13 - it's just an Android core version you are using. Anyway, continue to investigate and post your results - it might help others to overcome it faster and without a lot of frustrations.

gabryk91, 2018-05-09:
Ok, I didn't manage to wait until tomorrow, I reverted succesfully to 5.12
Now the sonoff connect instantly without any problem, placed in the exact same location than before.

I think that's something to look into for @arendst

ajanulis, 2018-05-10:
@reloxx13 , can you try 2.4.1?

ajanulis, 2018-05-16:
@Winni66, send me your "user_config.h" and I will build it for you with 2.3.0 Arduino core

Winni66, 2018-05-16:
@ajanulis 
that's a great offer, but what is user_config.h, where can I find it?
I'm using sonoff_knx.bin.

reloxx13, 2018-05-16:
pls make an own issue and read the wiki.

ajanulis, 2018-05-16:
@Winni66, ok, let's make it simpler - send me your wifi SSID, pswrd and gateway address to a.janulis at gmail tod com. @reloxx13, not everyone want's to mess with Atom, platformio and so on - some guys just want to use it straight. I can understand why @arendst still use last Arduino core if it's known for creating issues

closed 2018-07-10 by ascillato2