arendst/Tasmota #2674: Gpio2 status at boot

Mazzokun, 2018-05-09:
Hi guys! 
I have sonoff 4ch and I want use a tcrt 5000 to track the pulse of a watermeter.
If I connect the tcrt5000 to gpio2 after the boot everything works fine; but if the tcrt5000 is connected at boot time the sonoff doesn't boot up.
Is there a way to set gpio2=0 at boot time?

MatzeJoerling, 2018-05-09:
dont use water when your sonoff boots ;-) 
perhaps a pull down resistor? 
http://www.instructables.com/id/ESP8266-Using-GPIO0-GPIO2-as-inputs/

Jason2866, 2018-05-09:
GPIO2 has to be High at boot time!! If not Esp does not start. 
You have to take care for that if you connect anything to this GPIO

MatzeJoerling, 2018-05-09:
13! 1 ant 3 used for serial ...

Jason2866, 2018-05-09:
Gpio13 is used for led in ch4
You can use GPIO 1 and 3. You have to disable serial logging

Mazzokun, 2018-05-09:
Used Gpi3 and work like a charm! thank you 👍

closed 2018-06-04 by ascillato