arendst/Tasmota #2675: Documentation for setting lattitude, longitude and timezone?

apastuszak, 2018-05-09:
Make sure these boxes are checked before submitting your issue - Thank you!

- [X ] Sonoff Basic
- [] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

Is there documentation anywhere on how to set your geographic location and timezone?  Can this be added to the wiki?

Jason2866, 2018-05-09:
Is described in wiki "Commands"
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

apastuszak, 2018-05-09:
I don't see a way in the provided command to cofigure when DST starts and when it ends.  Can that only be done at compile time?

Jason2866, 2018-05-09:
Yes

apastuszak, 2018-05-09:
Hmm...  It would be nice if you could set that from the console.  With rules now in Tasmota, having proper time settings might be a kind of important.

ascillato, 2018-05-13:
Hi @apastuszak I also thinks that DST and STD should be modified by commands. Can you make a PR for that?

apastuszak, 2018-05-13:
I just did that!

https://github.com/arendst/Sonoff-Tasmota/issues/2721

ascillato, 2018-05-13:
A PR is a Pull Request that means code to be added.

You have just open another issue. 

I agree with you about adding those commands 

I was asking if you have or you can program the code to add those commands, so as to make a Pull Request?

closed 2018-05-13 by apastuszak