arendst/Tasmota #2680: AM2301/AM2302/Itead SI7021 what mudule type shoul i chose. Wiki confuse!

wongnam, 2018-05-10:
Hi 

I have 3 kinds of sensor AM2301/AM2302/Itead SI7021,  i want to chose the correct module type for them but i am very confusing by looking info on Wiki.

Module parameters:
- the  AM2301 i chose "**02 AM2301**" : it is easy and make sense. 
- the AM2302 i chose "**03 SI7201**" or "**02 AM2301**" : it also work. But i don't know it's correct or not. what kind is the best and correct i should chose?
- Itead SI7201 (one Wire), i chose SI7201: but on Wiki advise to chose AM2301,  so what kind i should chose?

Base on the wiki is very confusing:
https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-Configuration#i2c-sensor
https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini#dht22-dht21-am2301-am2302-am2321-shield-humidity--temperature-single-pin

**Info from Wiki " Devices - Additional Info" Wemos D1 Mini**:
[image]

**Info from Wiki "Sensor Configuration"**:
[image]

**Info from Tasmota "Changelog":**
[image]

Regards

Frogmore42, 2018-05-10:
I agree, it is confusing.  This is because things have changed over time and the wiki has not been updated to reflect that.  The simple answer is you should use the one that matches the sensor you are using.  

The longer answer.  The DHT21 is also known as the AM2301.  The DHT22 is also known as the AM2302.  They are basically the same.  The AM2301 has wires, while the AM2302 has pins to which you can attach/solder wires.  To get a better sensor, iTead took the Si7021 (which is normally an I2C sensor) and added a very small microcontroller and implemented the same single wire protocol, but it is a little different. 

The text strings have changed over time (which is why this is more confusing), but the numbers in parenthesis have stayed the same.  So, for both the 2301 and the 2302, select the 2nd one which is called AM2301.  For the S17021 (single wire version form iTead) select the 3rd option which is called Si7021

If you want to read about one person's quest for what is the best hygrometer.
www.kandrsmith.org/RJS/Misc/Hygrometers/calib_many.html

wongnam, 2018-05-10:
Thank you so much for your advice and  detail explaination.

wongnam, 2018-05-10:
I hope that the wiki will update soon for reducing the confusion for everyone.

Thanks

RaymondMouthaan, 2018-05-10:
I can image it is confusing, but I created the wemos wiki from a wemos point of view as a reminder how to connect sensors but also to help others. I believe it's quite clear, but feel free to update the wemos page and or the main Tasmota page now you know how it should be connected. I think everyone has access to update the wiki in general, and if not you can do a PR (pull request) and send it to Theo, who can accept your changes if he agrees.

closed 2018-05-12 by wongnam