arendst/Tasmota #2681: Reconfigure 4CH Pro without serial flash

Leseratte10, 2018-05-10:
I wanted to update my 4CH Pro to the newest Tasmota firmware (compiled myself, not the released file). I copied my user_config_override.h into the sonoff folder, but forgot the "-DUSE_CONFIG_OVERRIDE" flag (why isn't that enabled automatically when the file is present?). 

Thus, my 4CH Pro now has a firmware with no Wifi config (SSID and Password are empty in the user_config.h). And because this is a 4CH Pro, none of the Buttons on the device are connected to GPIO0. 

Do I have any way to reconfigure / reflash the device without completely disconnecting it from all the cables plugged into it, opening it and using the serial interface to reflash a working firmware?

Would it be possible to add some kind of "fallback" SSID and password which the device uses when the provided one is empty, to prevent this kind of issue in the future?

arendst, 2018-05-10:
I know. See  #2228 why.

Both arduino ide and platformio compilers fail to test on files presence.

The current solution is a selection of two bad choices...

I'm afraid the only solution is serial upload...

arendst, 2018-05-10:
The once default ssid indebuurt1 was removed by user request too.....

arendst, 2018-05-10:
I  hate security....

Leseratte10, 2018-05-10:
For the future, could Tasmota somehow detect if it is running on a 4CH Pro, and then accept one of the four present buttons as GPIO0? Or is the module setting (which got reset to "Basic" as well) the only way to detect the model?

arendst, 2018-05-10:
It can't detect the module and none of the pro buttons are connected directly to the esp8266 as they are serviced by another microcontroller.

Once you select the correct model it's features become available.

ascillato, 2018-06-04:
Hi

If your issue is solved, please close it. Thanks :+1:

closed 2018-06-11 by ascillato2