arendst/Tasmota #2684: Timezone with half an hour?

OmgImAlexis, 2018-05-10:
Is there currently any way to set a timezone with 30 min increments?
I live in South Australia which uses +930/+1030 depending on DST.

Jason2866, 2018-05-10:
See #1192

ascillato, 2018-05-11:
Hi,

I tested your case and works without problems.

I just change on user_config.h file the following:
```
// -- Time - Start Daylight Saving Time and timezone offset from UTC in minutes
#define TIME_DST  South, First, Sun, Oct, 2, +630

// -- Time - Start Standard Time and timezone offset from UTC in minutes
#define TIME_STD  South, First, Sun, Apr, 3, +530
```

I compile it and flash to a Sonoff.

Then, on the console I write timezone 99.

Console Output:

```
00:00:02 WIF: Intentando conectar...
00:00:03 WIF: Probando conección...
00:00:03 WIF: Conectado
00:00:03 HTP: Servidor web activo en sonoff-1757 con dirección IP 192.168.1.2
00:00:03 KNX: Iniciar
00:00:03 HTP: Consola
00:00:04 RSL: INFO1 = {"Module":"Generic","Version":"5.13.1a","FallbackTopic":"DVES_F7A6DD","GroupTopic":"sonoffs"}
00:00:04 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-1757","IPAddress":"192.168.1.2"}
00:00:04 RSL: INFO3 = {"RestartReason":"External System"}
00:00:06 RSL: RESULT = {"POWER":"OFF"}
00:00:06 RSL: POWER = OFF
00:00:10 APP: (UTC) Fri May 11 03:22:00 2018, (DST) Sun Oct 07 02:00:00 2018, (STD) Sun Apr 01 03:00:00 2018
13:52:03 RSL: STATE = {"Time":"2018-05-11T13:52:03","Uptime":"0T00:00:12","Vcc":2.785,"POWER":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","RSSI":70,"APMac":"18:D6:C7:80:38:2C"}}
13:52:16 WIF: Probando conección...
13:52:16 WIF: Conectado
```

And checked using the website https://24timezones.com/

OmgImAlexis, 2018-05-11:
Okay that's the issue. I was adding +9.5 not +930. My bad.

ascillato, 2018-05-12:
Hi,

You are right!

* Using the following config:

```
// -- Time - Start Daylight Saving Time and timezone offset from UTC in minutes
#define TIME_DST  South, First, Sun, Oct, 2, +630     // ( 630 minutes / 60 minutes ) = +10.5 hours

// -- Time - Start Standard Time and timezone offset from UTC in minutes
#define TIME_STD  South, First, Sun, Apr, 3, +570      // ( 570 minutes / 60 minutes ) = +9.5 hours
```

Results in:
```
00:00:06 APP: (UTC) Sat May 12 00:21:06 2018, (DST) Sun Oct 07 02:00:00 2018, (STD) Sun Apr 01 03:00:00 2018
10:51:12 RSL: STATE = {"Time":"2018-05-12T10:51:12","Uptime":"0T00:00:12","Vcc":2.788,"POWER":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","RSSI":36,"APMac":"18:D6:C7:80:38:2C"}}
10:51:47 WIF: Checking connection...
10:51:47 WIF: Connected
```

**That is wrong**

* But using the following configuration:
```
// -- Time - Start Daylight Saving Time and timezone offset from UTC in minutes
#define TIME_DST  North, First, Sun, Oct, 2, +630     // ( 630 minutes / 60 minutes ) = +10.5 hours

// -- Time - Start Standard Time and timezone offset from UTC in minutes
#define TIME_STD  North, First, Sun, Apr, 3, +570      // ( 570 minutes / 60 minutes ) = +9.5 hours
```

Results in:
```
00:00:06 APP: (UTC) Sat May 12 00:25:23 2018, (DST) Sun Oct 07 02:00:00 2018, (STD) Sun Apr 01 03:00:00 2018
09:55:29 RSL: STATE = {"Time":"2018-05-12T09:55:29","Uptime":"0T00:00:12","Vcc":2.784,"POWER":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","RSSI":50,"APMac":"18:D6:C7:80:38:2C"}}
09:55:40 WIF: Checking connection...
09:55:40 WIF: Connected
```

**That is Ok.**

So,

Looking at https://24timezones.com/world_directory/time_in_adelaide.php, the second configuration will give the correct time for you.

@OmgImAlexis, **Please use North instead of South.**

@arendst, please, review this because in the code uses the North and South hemisphere but seems that there is no need for that because DST and STD times already contains the start date and time shift and DST is around the same months everywhere. The same for STD.

As Argentina don't use DST and STD, I can not put that as example but looking at Chile, they have DST and STD.

So,

For Netherlands you **add** 1 hour for the months around july
For Australia you **substract** 1 hour for the months around july
For Chile you **substract** 1 hour for the months around july

And in all those countries, all called DST (Day Light Saving Time) to that period.

[image]

[image]

[image]

ascillato, 2018-05-13:
Today Chile changed from DST to STD. But testing Tasmota on a Sonoff stays on DST. **Is not correct.**

Found that the code that checks for DST and STD for Southern Hemisphere had a bug on `RTCSecond()` routine in _support.ino_ file

Made the PR #2714 to fix this.

Tested new code with settings for Australia, Chile and Netherlands and works fine.

----------------------------------------------------------------------------------------------------------

Chile Example, Tasmota's console:

Sonoff without the proposed fix:

```
23:59:34 WIF: Probando conección...
23:59:34 WIF: Conectado
23:59:56 WIF: Probando conección...
23:59:56 WIF: Conectado
00:00:18 WIF: Probando conección...
00:00:18 WIF: Conectado
00:00:41 WIF: Probando conección...
00:00:41 WIF: Conectado
00:00:50 RSL: STATE = {"Time":"2018-05-13T00:00:50","Uptime":"0T00:43:47","Vcc":2.784,"POWER":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","RSSI":56,"APMac":"18:D6:C7:80:38:2C"}}
00:01:03 WIF: Probando conección...
00:01:03 WIF: Conectado
00:01:24 WIF: Probando conección...
00:01:24 WIF: Conectado
00:01:42 APP: (UTC) Sun May 13 03:01:42 2018, (DST) Sun Aug 12 00:00:00 2018, (STD) Sun May 13 00:00:00 2018
00:01:46 WIF: Probando conección...
00:01:46 WIF: Conectado
00:02:01 RSL: UPTIME = {"Time":"2018-05-13T00:02:01","Uptime":"0T00:44:58"}
00:02:08 WIF: Probando conección...
00:02:08 WIF: Conectado
```

Sonoff with the proposed fix:

(time expressed as https://24timezones.com/world_directory/time_in_santiago.php)

```
23:59:37 WIF: Probando conección...
23:59:37 WIF: Conectado
23:59:59 WIF: Probando conección...
23:59:59 WIF: Conectado
23:00:13 MQT: tele/living/STATE = {"Time":"2018-05-12T23:00:13","Uptime":"0T00:43:38","Vcc":2.696,"POWER":"OFF","Wifi":{"AP":2,"SSId":"NetWireless","RSSI":62,"APMac":"18:D6:C7:80:38:2C"}}
23:00:20 WIF: Probando conección...
23:00:20 WIF: Conectado
23:00:42 WIF: Probando conección...
23:00:42 WIF: Conectado
23:01:04 WIF: Probando conección...
23:01:04 WIF: Conectado
23:01:25 WIF: Probando conección...
23:01:25 WIF: Conectado
23:01:42 APP: (UTC) Sun May 13 03:01:42 2018, (DST) Sun Aug 12 00:00:00 2018, (STD) Sun May 13 00:00:00 2018
23:01:47 WIF: Probando conección...
23:01:47 WIF: Conectado
23:02:00 MQT: tele/living/UPTIME = {"Time":"2018-05-12T23:02:00","Uptime":"0T00:45:25"}
23:02:09 WIF: Probando conección...
23:02:09 WIF: Conectado
```

ascillato, 2018-05-13:
Hi @OmgImAlexis

The fix was applied. Please, test it again.

And if the problem is solved for you, please close this issue. Thanks.

closed 2018-05-13 by OmgImAlexis