arendst/Tasmota #2685: Routed traffic to sonoff gui webserver is not working for me

GregoryOrciuch, 2018-05-10:
Hi,

Sonoff Basic. Status 0 log: https://ghostbin.com/paste/vpmxc

I'm accessing Sonoff Basic using two different methods:

1. Routed traffic from vpn host like 192.168.37.2 -> openVpn -> pfsense -> nat -> homenetwork (192.168.99.0) -> sonoff via wifi on 192.168.99.71
(basically traffic is routed from 192.168.37.0/24 to 192.168.99.0)

2. Via port forwading using SSH directly to sonoff http port.

With method 2 I can very well access sonoff webgui on http port, and also all rest command using curl work very well.
The problem is method 1, here observations:
- Can not access sonoff web page from web browser
- Curl to web page is hanging (need to hit ctrl+c, example: https://ghostbin.com/paste/crfnn )
- Curl to sonoff REST services is working well.

There are other devices like qnap NAS I'm used to access via routed network, there web gui works very well. There are no specific firewall exceptions for sonoff etc. It's just sonoff web page which has a problems here, but REST is not. This is weird.

Wifi is very good, REST communication working all the time.

Is this separate webserver for REST and Webpage ?

( I will try later capturing tcpdump)

Thanks!

wongnam, 2018-05-10:
Try to re-compile the code with esp8266 core version 2.3.0, you can access it by method 1.

GregoryOrciuch, 2018-05-10:
@wongnam can you elaborate why it's like that ? I mean why 2.3.0 helps here.

wongnam, 2018-05-10:
Because core ver 2.4.0 have bug

benzino77, 2018-05-10:
I'm basically using method 1 (more, less) and I do had  problems with Esp8266 core 2.4.0 and 2.4.1. All of my devices are based (manually recompiled) on esp8266 core 2.3.x and they work fine.
Just change the `platfomio.ini` from:
```
; *** Esp8266 core for Arduino version
;platform = espressif8266@1.5.0  ; v2.3.0
;platform = espressif8266@1.6.0  ; v2.4.0
;platform = espressif8266@1.7.0  ; v2.4.1
;platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
platform = espressif8266
```
to
```
; *** Esp8266 core for Arduino version
platform = espressif8266@1.5.0  ; v2.3.0
;platform = espressif8266@1.6.0  ; v2.4.0
;platform = espressif8266@1.7.0  ; v2.4.1
;platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
;platform = espressif8266
```

GregoryOrciuch, 2018-05-11:
confirm that it's working very well using 5.13.1a and core version **2.3.0** .

Thanks for help. Closing the issue.

closed 2018-05-11 by GregoryOrciuch