arendst/Tasmota #2688: Sonoff RF keeps turning off when using RF remote

sleezy100, 2018-05-10:
I have upgraded my Sonoff RF to 5.13.1 and all was fine. Now, every time i use the rf remote to switch on, after a few seconds it switches itself of again. I then switch on again and after a few seconds it switches off again. If i switch it on by using HA everything works as it should. Anyone have any ideas ?????

Leseratte10, 2018-05-10:
Sounds like your home automation system is overriding the state of the sonoff. You need to fix that setting (probably "persistence" or something like that) in your HA system.

Jason2866, 2018-05-10:
It looks that your HA has stored a retain message an sends this again and agian.
This is a know behaviour from HA.
See #2140

Jason2866, 2018-05-10:
I am out here, i dont use HA. I am a OpenHab user

sleezy100, 2018-05-11:
I have down graded to 5.12.0 and all is fine again.  There is obviously something within 5.13.1 not quite right.

closed 2018-07-10 by ascillato2