arendst/Tasmota #2689: Web Interface not available after upgrade/flash of version

TylerDurden23, 2018-05-10:
After installing Version v5.13.1 using either OTA or serial flash, I am no longer able to access the web Interface on the following devices:

-Lingan SWA1
-Sonoff Touch
-Sonoff POW

The port 80 seems open but the device is not sending back any data. MQTT still works.

TylerDurden23, 2018-05-10:
Also: The REST commands still seem to work - the static pages seem to be the Problem.

Frogmore42, 2018-05-10:
If you look at many of the other issues, you will see people are having issues with core 2.4.x.  You did not provide the standard information requested for all issues, so it is not possible to tell for sure what you are using, but based on your results you are likely one of the people who have an environment that doesn't work well with 2.4.x.

TylerDurden23, 2018-05-10:
I've used the pre-compiled binaries from the release page and from http://sonoff.maddox.co.uk/tasmota/sonoff.bin,

The issue is quite specific to be related to my specific the environment, since everything except the static web pages seem to work.

Frogmore42, 2018-05-10:
I believe the latest pre-compiled binaries use 2.4.1, which was supposed to address the issues from 2.4.0, but does not seem to have actually done that.

You can either use an older version of the pre-compiled binaries (I think 5.12 used 2.3.0 core) or compile the code yourself with PlatformIO.

TylerDurden23, 2018-05-10:
Are the binaries using Core 2.4.1 available anywhere?

Frogmore42, 2018-05-10:
see https://github.com/arendst/Sonoff-Tasmota/issues/2678 for issues with 2.4.1

Frogmore42, 2018-05-10:
I would use 5.12.0 if you need precompiled binaries.  Lots of people have had good luck with it.

ajanulis, 2018-05-10:
check #2673 as well

TylerDurden23, 2018-05-11:
Version 5.12 works (thats the version I upgraded from) - but does not solve the issue that the latest version has on all my devices. ;)

ajanulis, 2018-05-11:
@TylerDurden23, I guess it's not 5.12 or 5.13 - issue is with Arduino core - try to compile 5.13.1a using Arduino 2.3.0 - if success - go up with 2.4.0 and again, if success - with 2.4.1.

ajanulis, 2018-05-11:
@TylerDurden23, if you want, I can built it for you, I just keen to know where problem is hidden - in 5.13 or Arduino core

TylerDurden23, 2018-05-11:
I've built it now: Yes, is seems it was the 2.4.0 core, everything seems to work on 2.4.1 on all of the devices.

wongnam, 2018-05-11:
Use 2.3.0 to build your own fw. 2.4.0 and 2.4.1 are still have problem with webserver.

TylerDurden23, 2018-05-11:
@wongnam  I will if the need arises, at the moment everything seems to work.

ajanulis, 2018-05-11:
@wongnam, I am not sure what Arduino core version is the last, but I've tried with 2.3.0, 2.4.0, as well as 2.4.1, and everything works perfectly (not sure which version 5.13.1a uses by default, but with this default I was rarely able to connect to wifi)

rlust, 2018-05-12:
I have recompiled 5.1.3a with 2.4.1 and 2.3 and still having issues with web pages not working.

Pcdejong, 2018-05-12:
I have what looks like the same issue. Upgraded my basics from 5.12.0 to 5.13.1 and now the two far from my accesspoint works very bad. MQTT has a long delay or doesn't work. The web interface is very slow or even not reachable. Only the one that is 1 meter from my accesspoint is reachable. My sonoff S20 that I also upgraded and is twice as far as my basics has no problem.

ajanulis, 2018-05-12:
@Pcdejong, it's definitely an issue with Android core - just recompile 5.13.1a with 2.3.0 core and you will get back the range with all new functionality

mokshmridul, 2018-05-13:
Can someone guide me as to how to recompile for 2.3.0.

I am trying this guide (https://github.com/arendst/Sonoff-Tasmota/wiki/Beginner-Guide---Create-your-own-Firmware-Build) , but im failing (as the uploaded file is causing the same issue)

Or could someone please compile it for me with 2.3.0

ajanulis, 2018-05-13:
@mokshmridul, I can try to guide you. Are you using Atom and Platformio?

ajanulis, 2018-05-13:
@mokshmridul, where you are stuck? What board or device you are using? Sonoff or other? What you are getting at the hitting "Build"?

mokshmridul, 2018-05-14:
@ajanulis I am trying to build for the sonoff basic. Atom builds it sucessfully and saves the file, however when i flash using the said file, the webserver becomes unreachable and has massive packet loss. I cant even use MQTT to trigger on/off commands.

Below is a copy/paste of the platformio.ini file"

`; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter, extra scripting
;   Upload options: custom port, speed and extra flags
;   Library options: dependencies, extra library storages
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/en/stable/projectconf.html

[platformio]
src_dir = sonoff

; *** Uncomment one of the lines below to build/upload only one environment
;env_default = sonoff
env_default = sonoff-minimal
;env_default = sonoff-classic
;env_default = sonoff-knx
;env_default = sonoff-allsensors
;env_default = sonoff-BG
;env_default = sonoff-CN
;env_default = sonoff-CZ
;env_default = sonoff-DE
;env_default = sonoff-ES
;env_default = sonoff-FR
;env_default = sonoff-GR
;env_default = sonoff-HU
;env_default = sonoff-IT
;env_default = sonoff-NL
;env_default = sonoff-PL
;env_default = sonoff-PT
;env_default = sonoff-RU
;env_default = sonoff-TW

[common] ; ************************************************************
; *** Esp8266 core for Arduino version
;platform = espressif8266@1.5.0  ; v2.3.0
;platform = espressif8266@1.6.0  ; v2.4.0
;platform = espressif8266@1.7.0  ; v2.4.1
;platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
platform = espressif8266@1.5.0

framework = arduino
board = esp01_1m
board_flash_mode = dout

; *** Fix espressif8266@1.7.0 induced undesired all warnings
build_unflags = -Wall

build_flags =
  -Wl,-Tesp8266.flash.1m0.ld
  -DVTABLES_IN_FLASH
  -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
;  -DUSE_CONFIG_OVERRIDE

; *** Fix Esp/Arduino core 2.4.x induced Tasmota unused floating point includes
extra_scripts = pio/strip-floats.py

; *** Serial Monitor options
monitor_baud = 115200

; *** Upload Serial reset method for Wemos and NodeMCU
;upload_speed = 115200
upload_speed = 512000
upload_resetmethod = nodemcu
upload_port = COM5

[... 218 lines cut ...]
board = ${common.board}
board_flash_mode = ${common.board_flash_mode}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags} -DMY_LANGUAGE=ru-RU
monitor_baud = ${common.monitor_baud}
upload_speed = ${common.upload_speed}
upload_port = ${common.upload_port}
extra_scripts = ${common.extra_scripts}

[env:sonoff-TW]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board}
board_flash_mode = ${common.board_flash_mode}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags} -DMY_LANGUAGE=zh-TW
monitor_baud = ${common.monitor_baud}
upload_speed = ${common.upload_speed}
upload_port = ${common.upload_port}
extra_scripts = ${common.extra_scripts}
`

I am also attaching a screenshot of my atom post successful compilation

[image]

ajanulis, 2018-05-14:
@mokshmridul, everything looks correct. How you upload your build? Using built in Platformio functionality or using something else? Looks like you are using Windows machine. Can you try  ESP8266Flasher? Looking at your platformio.ini I would say you should find firmware.bin file under your project /.pioenvs/sonoff-minimal/firmware.bin. Please remember to choose SPI mode as DOUT under Advanced settings in esp8266Flasher.

mokshmridul, 2018-05-14:
@ajanulis let me start by thanking you for your quick replies :)

Yes, i am using windows 10. I am performing the upload using the tasmota webserver and going into "Firmware Upgrade", by first flashing the minimal-config and then the complete version.

However, when i flash the minimal version, the webserver becomes very unstable, even on the bin i have compiled.

ajanulis, 2018-05-14:
@mokshmridul but even with this "very unstable" webserver you still able to reflash firmware? What signs on webserver instability you observe?

mokshmridul, 2018-05-14:
@ajanulis i managed to flash it, but with extreme difficultly. I took me over an hour to revert back to 5.12.

By instability, I mean that it has extreme instability and massive massive packet loss. I cant even use the mqtt commands to trigger it.

ajanulis, 2018-05-14:
@mokshmridul, have you an ability to connect your Sonoff directly with any FTDI? If you want, of course, I can try to build you 5.13.1a with 2.3.0 in my environment, but that you should provide me your user_config.h or user_config_override.h (if you are using it) with your wifi credentials (or maybe it's not necessary, but I did not tested 5.13.1a in case both AP not configured). We can try, of course and see how it turns out

mokshmridul, 2018-05-14:
@ajanulis at this point, i cant connect using the FTDI connector. It would be great if you could build 5.13.1a on 2.3.0 on your environment. How should i share the details with you?

ajanulis, 2018-05-14:
@mokshmridul if you feel safe, just send me your SSID and pswd as well as Wifi gateway details (if not using DHCP) and list of sensors you are going to use. Or you can sent me user_config.h with all details. Use a.janulis @ gmail tod com

mokshmridul, 2018-05-14:
@ajanulis I've sent you an email with all the details. Let me know if there is anything i missed out.
Thanks

ajanulis, 2018-05-14:
@mokshmridul, got your email. Are you using user_config.h or user_config_override.h ?

mokshmridul, 2018-05-14:
@ajanulis im using user_config.h

ajanulis, 2018-05-14:
@mokshmridul , check email

mokshmridul, 2018-05-14:
@arendst sent you an email.

ascillato, 2018-06-04:
Hi

If your issue is solved, please close it. Thanks :+1:

closed 2018-06-11 by ascillato2