arendst/Tasmota #2690: Errors while using D1 mini with Tasmota and WS2812B RGB light in Home Assistant

jascdk, 2018-05-10:
Hi.

I have just flashed a D1 mini with the newest tasmota firmware. Afterwards I have put on a RGB (ws2812B) shield and then configured it in Home Assistant like the Wiki here says:)

The fact is that I get some errors in HA like this:

#######################################################

Thu May 10 2018 22:35:29 GMT+0200 (CEST)

Error doing job: Exception in callback <bound method MQTT._mqtt_handle_message of <homeassistant.components.mqtt.MQTT object at 0x760ec490>>
Traceback (most recent call last):
  File "uvloop/cbhandles.pyx", line 49, in uvloop.loop.Handle._run
  File "/usr/lib/python3.6/site-packages/homeassistant/components/mqtt/__init__.py", line 647, in _mqtt_handle_message
    msg.topic, payload, msg.qos)
  File "/usr/lib/python3.6/site-packages/homeassistant/core.py", line 248, in async_run_job
    target(*args)
  File "/usr/lib/python3.6/site-packages/homeassistant/components/light/mqtt.py", line 266, in rgb_received
    templates[CONF_RGB](payload).split(',')]
  File "/usr/lib/python3.6/site-packages/homeassistant/components/light/mqtt.py", line 265, in <listcomp>
    rgb = [int(val) for val in
ValueError: invalid literal for int() with base 10: '7fff00'

#######################################################

The lights actually works OK, but something must be wrong:/

Best regards

Jacob

ssirkakriss, 2018-05-10:
I have a Wemos mini and some ws2812's attached with no errors.
Here's my working config for it in configuration.yaml, for reference :

```
# sonff thermo wemos 1 2812b
  - platform: mqtt
    name: "Temp-pixel_mod test"
    retain: true
    state_topic: "stat/sw-temps/POWER"
    command_topic: "cmnd/sw-temps/POWER"
    rgb_command_mode: hex
    rgb_command_template: "{{ '%02x%02x%02x' | format(red, green, blue)}}"
    rgb_state_topic: "stat/sw-temps/Color"
    rgb_command_topic: "cmnd/sw-temps/Color"
    effect_state_topic: "stat/sw-temps/Scheme"
    effect_command_topic: "cmnd/sw-temps/Scheme"
    brightness_state_topic: "stat/sw-temps/Dimmer"
    brightness_command_topic: "cmnd/sw-temps/Dimmer"
    brightness_value_template: "{{ value_json.Dimmer }}"
    brightness_scale: 100
    effect_value_template: "{{ value_json.scheme }}"
    effect_list:
        - 0
        - 1
        - 2
        - 3
        - 4
        - 5
        - 6
        - 7
        - 8
        - 9
        - 10
        - 11
        - 12
    payload_on: "ON"
    payload_off: "OFF"
    optimistic: false
```
Hope it's of help.

ascillato, 2018-05-11:
Hi,

I use the following:

Configure any of the pins of the module as "WS2812B" and on the Console run `SetOption15 1`.

```yaml
light:
# LED WS2812b
    - platform: mqtt
      name: "WS2812b"
      retain: true
      state_topic: "stat/led/POWER"
      command_topic: "cmnd/led/POWER"
      availability_topic: "tele/led/LWT"
      qos: 1
      payload_on: "ON"
      payload_off: "OFF"
      payload_available: "Online"
      payload_not_available: "Offline"
      optimistic: false
      brightness_state_topic: "stat/led/Dimmer"
      brightness_command_topic: "cmnd/led/Dimmer"
      brightness_value_template: "{{ value_json.Dimmer }}"
      brightness_scale: 100      
      rgb_state_topic: "stat/led/Color"
      rgb_command_topic: "cmnd/led/Color2"
      rgb_command_mode: hex
      rgb_command_template: "{{ '%02x%02x%02x' | format(red, green, blue)}}"
      effect_state_topic: "stat/led/Scheme"
      effect_command_topic: "cmnd/led/Scheme"
      effect_value_template: "{{ value_json.Scheme }}"
      effect_list:
          - 0
          - 1
          - 2
          - 3
          - 4
          - 5
          - 6
          - 7
          - 8
          - 9
          - 10
          - 11
          - 12
```

jascdk, 2018-05-11:
Wow - great - thanks for the quick replys:) Maybe the wiki should be updated???:)

ascillato, 2018-05-11:
Done.

ascillato, 2018-05-12:
Speed for the schemes? It is already there in commands. See the commands in the wiki

ascillato, 2018-05-13:
Hi @jascdk
If your problem is solved, please close the issue.

Thanks

closed 2018-06-08 by ascillato2