arendst/Tasmota #2692: Relay5 to relay8 not working on GPIO5

gittyhubbyfrankybobby, 2018-05-11:
i'm running 5.13.1a on a Wemos D1 mini.
I have a relay on GPIO5 (D1).
If I assign GPIO5 to relay 1, 2, 3, or 4 it works and can be controlled via switch 1,2,3, 4 or webpage.

If I assign GPIO5 to relay 5, 6, 7, or 8 it does not change state via webpage, or MQTT, and there's no corresponding switch 5, 6, 7, or 8 to test with.

....

21:51:02 CMD: status 0
21:51:02 MQT: stat/Coop/STATUS = {"Status":{"Module":18,"FriendlyName":["Coop","RDWH12","RDWH13","RDWH14"],"Topic":"Coop","ButtonTopic":"0","Power":3,"PowerOnState":0,"LedState":7,"SaveData":1,"SaveState":0,"ButtonRetain":0,"PowerRetain":0}}
21:51:02 MQT: stat/Coop/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:32","StartupUTC":"2018-05-11T02:49:30","Sleep":0,"BootCount":131,"SaveCount":357,"SaveAddress":"3FA000"}}
21:51:02 MQT: stat/Coop/STATUS2 = {"StatusFWR":{"Version":"5.13.1a","BuildDateTime":"2018-05-09T23:52:29","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:51:02 MQT: stat/Coop/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":2,"LogHost":"192.168.99.7","LogPort":514,"SSId":["GAMERA!",""],"TelePeriod":300,"SetOption":["00008108","55818000"]}}
21:51:02 MQT: stat/Coop/STATUS4 = {"StatusMEM":{"ProgramSize":528,"Free":492,"Heap":14,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2}}
21:51:02 MQT: stat/Coop/STATUS5 = {"StatusNET":{"Hostname":"Coop","IPAddress":"192.168.xxx.199","Gateway":"192.168.xxx.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.xxx.7","Mac":"A0:20:A6:23:4A:11","Webserver":2,"WifiConfig":3}}
21:51:02 MQT: stat/Coop/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.xxx.7","MqttPort":1883,"MqttClientMask":"Coop","MqttClient":"Coop","MqttUser":"s20-6","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:51:02 MQT: stat/Coop/STATUS7 = {"StatusTIM":{"UTC":"Fri May 11 02:51:02 2018","Local":"Thu May 10 21:51:02 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":-5,"Sunrise":"05:36","Sunset":"19:04"}}
21:51:02 MQT: stat/Coop/STATUS10 = {"StatusSNS":{"Time":"2018-05-10T21:51:02","Switch1":"ON","Switch2":"ON","Switch3":"ON","Switch4":"OFF","DS18B20":{"Temperature":32.0},"TempUnit":"F"}}
21:51:02 MQT: stat/Coop/STATUS11 = {"StatusSTS":{"Time":"2018-05-10T21:51:02","Uptime":"0T00:01:32","Vcc":2.769,"POWER1":"ON","POWER2":"ON","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":1,"SSId":"xxx!","RSSI":68,"APMac":"C4:3D:C7:48:1A:A8"}}}
21:51:05 WIF: Checking connection...

curzon01, 2018-05-11:
In general, there should be no gap in relay definition, e.g. if you define only relay 4 as the only one it also does not work.

You need to define relay 1-4 before you can use relay 5 and so on.

The following works with current 5.13.1a on a Wemos:
```
CMD: Backlog SerialLog;Gpio
MQT: stat/tasmota/sonoff/RESULT = {"SerialLog":"0 (Active 0)"}
MQT: stat/tasmota/sonoff/RESULT = {"GPIO0":"0 (None)","GPIO1":"17 (Relay1)","GPIO2":"18 (Relay2)","GPIO3":"19 (Relay3)","GPIO4":"20 (Relay4)","GPIO5":"21 (Relay5)","GPIO12":"0 (None)","GPIO13":"0 (None)","GPIO14":"0 (None)","GPIO15":"0 (None)","GPIO16":"0 (None)"}
```

gittyhubbyfrankybobby, 2018-05-11:
That seems to have corrected the issue. Thanks

closed 2018-05-11 by gittyhubbyfrankybobby