arendst/Tasmota #2693: Problem with GPIO14 and built-in relay

Cjcr, 2018-05-11:
Hello,

I'm using Sonoff Basic with the lastest version of Tasmota.
I'm developing a Telegram Bot that works with Tasmota firmware communicating together via HTTP calls. That's the idea.

Well, what I want to do is read the gpio14 input status to do some actions in the telegram gateway, that's seems ok, because I can read the status of the gpio14 with http calls.

The problem I have that I can't find a way to make the build-in relay independent of the input status of GPIO14. I'm not using MQTT, so commands for this protocol aren't valid in this case.

Seems that the magic are in SwitchMode(X) command, but I miss an option to disable the relay control. Can you implement it in your code or there is any other solution for this?

Thank you!

Jason2866, 2018-05-11:
Define GPIO 14 as Switch2 (10) , and SwitchMode2  (1)
Now you have an independent virtual second relay in your Sonoff Basic.
You can check the Power2 ON / OFF Status (=GPIO 14) to do the magic with "telegram"

Cjcr, 2018-05-11:
@Jason2866 Same behavior ... when I put gnd<->gpio14, the build-in relay turn on & off everytime that gnd-gpio14 are connected. I don't know why.

Jason2866, 2018-05-11:
Please post:  Status 0

Cjcr, 2018-05-11:
{"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:10","StartupUTC":"2018-05-11T06:38:50","Sleep":0,"BootCount":55,"SaveCount":276,"SaveAddress":"F8000"},"StatusFWR":{"Version":"5.13.1a","BuildDateTime":"2018-05-10T19:30:34","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"},"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SEALAND_HOME",""],"TelePeriod":300,"SetOption":["0000C80B","55818000"]},"StatusMEM":{"ProgramSize":536,"Free":464,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3},"StatusNET":{"Hostname":"sonoff-6619","IPAddress":"10.0.0.112","Gateway":"10.0.0.11","Subnetmask":"255.0.0.0","DNSServer":"8.8.8.8","Mac":"5C:CF:7F:68:XXXX","Webserver":2,"WifiConfig":3},"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6879DB","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15},"StatusTIM":{"UTC":"Fri May 11 06:45:00 2018","Local":"Fri May 11 07:45:00 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:13","Sunset":"20:19"},"StatusSNS":{"Time":"2018-05-11T07:45:00","Switch2":"ON"},"StatusSTS":{"Time":"2018-05-11T07:45:00","Uptime":"0T00:06:10","Vcc":3.517,"POWER":"ON","Wifi":{"AP":1,"SSId":"XXXXXXXX","RSSI":58,"APMac":"EC:F4:51:B2:C9:XX"}}}

Jason2866, 2018-05-11:
Nothing special here...

Please post: Result of command:  gpio

curzon01, 2018-05-11:
@Jason2866 - Cjcr has disabled MQTT functionality and if you do so `SwitchTopic` other than `0` does not work as expected anymore.

@Cjcr 
Switching additonal input independently to relay output currently does not work if you disable the MQTT (even not with the latest 5.13.1a).

- The following setup works for Gpio14 as Jason2866 described:
```
CMD: Backlog Module;Gpio;SwitchMode1;SwitchMode2;SwitchTopic;SetOption3
MQT: stat/tasmota/sonoff/RESULT = {"Module":"1 (Sonoff Basic)"}
MQT: stat/tasmota/sonoff/RESULT = {"GPIO1":"0 (None)","GPIO3":"0 (None)","GPIO4":"0 (None)","GPIO14":"10 (Switch2)"}
MQT: stat/tasmota/sonoff/RESULT = {"SwitchMode1":0}
MQT: stat/tasmota/sonoff/RESULT = {"SwitchMode2":1}
MQT: stat/tasmota/sonoff/RESULT = {"SwitchTopic":"sonoff"}
MQT: stat/tasmota/sonoff/RESULT = {"SetOption3":"ON"}
```
- It **does not work if your disable MQTT** (`SetOption3 OFF`).

My suggestion as a workaround: 
Compile a version using non-blocking MQTT driver (user_config.h or user_config_override.h: `#define MQTT_LIBRARY_TYPE      MQTT_TASMOTAMQTT`) and enable MQTT even you are not using them.

Jason2866, 2018-05-11:
@curzon01 Help me, where did you see that mqtt is disabled?

Cjcr, 2018-05-11:
@curzon01 Hi, thank you, but I still got the same behaivor ... built-in relay turns on/off everytime I connect the gnd-gpio14.

```
16:45:52 CMD: Backlog Module;Gpio;SwitchMode1;SwitchMode2;SwitchTopic;SetOption3
16:45:53 RSL: stat/sonoff/RESULT = {"Module":"1 (Sonoff Basic)"}
16:45:53 RSL: stat/sonoff/RESULT = {"GPIO1":"0 (Ninguno)","GPIO3":"0 (Ninguno)","GPIO4":"0 (Ninguno)","GPIO14":"10 (Llave2)"}
16:45:53 RSL: stat/sonoff/RESULT = {"SwitchMode1":0}
16:45:53 RSL: stat/sonoff/RESULT = {"SwitchMode2":1}
16:45:53 RSL: stat/sonoff/RESULT = {"SwitchTopic":"sonoff"}
16:45:54 RSL: stat/sonoff/RESULT = {"SetOption3":"ON"}
```
@Jason2866 
Gpio

```
16:46:20 CMD: gpio
16:46:20 RSL: stat/sonoff/RESULT = {"GPIO1":"0 (Ninguno)","GPIO3":"0 (Ninguno)","GPIO4":"0 (Ninguno)","GPIO14":"10 (Llave2)"}
```

Cjcr, 2018-05-11:
> Switching additonal input independently to relay output currently does not work if you disable the MQTT (even not with the latest 5.13.1a).

I hope this should be fixed because not all we use mqtt ...

Cjcr, 2018-05-11:
@curzon01 @Jason2866 Initially it was disabled (first post) but later I enabled it again because seems that is needed to make it work ... but no luck yet :-(

Cjcr, 2018-05-11:
Finally I fixed it myself modding a bit the code. If you want the same behavior, you can archive it doing this (no mtqq needed, you can disable it):

**sonoff.h**

Before:
`enum SwitchModeOptions {TOGGLE, FOLLOW, FOLLOW_INV, PUSHBUTTON, PUSHBUTTON_INV, PUSHBUTTONHOLD, PUSHBUTTONHOLD_INV, PUSHBUTTON_TOGGLE, MAX_SWITCH_OPTION};`

After:

`enum SwitchModeOptions {TOGGLE, FOLLOW, FOLLOW_INV, PUSHBUTTON, PUSHBUTTON_INV, PUSHBUTTONHOLD, PUSHBUTTONHOLD_INV, PUSHBUTTON_TOGGLE, DONOTHING, MAX_SWITCH_OPTION};`

**sonoff.ino**

Just add another 'case' with this code in the SwitchHandler() function:

```
        case DONOTHING:
          switchflag = 6;
          break;
        }
```

And when you load the code, use the `SwitchMode2 8` in the console

Now works perfect :-)

Jason2866, 2018-05-11:
Mhh, i dont understand why the solution that @curzon01 and i suggest didnt worked for you.
Finally you found a solution. That counts :-)

curzon01, 2018-05-13:
@Jason2866 - as we do not know what side effects other settings might have Cjcr could reproduce the suggestion by using a plain Tasmota without own settings changes (e.g. without own user_config.h) and Reset config - that's what I did and my suggestion is reproducable.

closed 2018-06-04 by ascillato

beamzer, 2020-10-24:
I got here because i have the same problem, Sonoff basic with a PIR sensor on GPIO14, i don't want the PIR output to control the relay directly, but via a rule: e.g. 
```
Rule1 on Switch2#state=1 DO Backlog Power1 ON; RuleTimer1 30 ENDON ON Rules#Timer=1 DO Power1 OFF ENDON
```
The proposed solution from @curzon01 works half way, the PIR sensor doesn't actuate the Relay, but the Rule if have doesn't work either :(
I can see the detection of the PIR sensor in the Console log (i set the topic to dummy)
```
16:24:18 MQT: cmnd/dummy/POWER2 = ON
16:24:22 MQT: cmnd/dummy/POWER2 = OFF
```
Also it doesn't seem to send any actual MQTT messages to my MQTT server (domoticz, i do see those when i operate the button on the Sonoff)
```
2020-10-24 17:30:20.174  MQTT: Topic: domoticz/in, Message: {"idx":235,"nvalue":1,"svalue":"","Battery":100,"RSSI":4}
2020-10-24 17:30:21.787  MQTT: Topic: domoticz/in, Message: {"idx":235,"nvalue":0,"svalue":"","Battery":100,"RSSI":4}
```
Any idea?

thanks in advance,
Ewald...