arendst/Tasmota #2697: Omoton wifi smartplug with two LEDs, a button, and a relay.

bears14, 2018-05-11:
I have an Omoton wifi smartplug which I flashed with this firmware.  With GPIO assignments as follows:
GPIO 16 = Relay; 
GPIO 13 = LED in button;
TX = Button; and 
GPIO 14 = LED on front of switch.  
I configured it as follows:
Module type = 18 Generic
D3 GPIO0  = None
TX GPIO1 = Button1
D4 GPIO2  = None
RX GPIO3  = None
D2 GPIO4  = None
D1 GPIO5  = None
D6 GPIO12 = None
D7 GPIO13 = Led1i
D5 GPIO14 = Relay1
D8 GPIO15 = None
D0 GPIO16 = Led1

However, this does not seem to work with both LEDs only GPIO 16 led works.  If I change GPIO 16 to 'None' the the GPIO 13 LED works but obviously GPIO 16 led does not.  Is there any way to configure this device so that both LEDs will work.  Id prefer to have GPIO 13 illuminate when the relay is off and GPIO 16 illuminate when the relay is on.

Thanks

Geitde, 2018-05-17:
The two LED situation is kind of an issue for me, too. 

I modified the Sonoff Touch in a way that the LED original connected parallel to the relay gpio is now connected to gpio4. That way I was able to have the touch ring light, when the light is off. (Having it only light when the light is on, is quite a hardware design flaw).

This works fine after doing some custom config for the touch, to show the inverted relay status on gpio4.

However the original Wifi LED is now kind of useless, as there is no way to define some output as "error state", "packets received" or what not.

closed 2018-07-10 by ascillato2

austinlg96, 2018-08-17:
@bears14 Which Omoton plug were you working with?