arendst/Tasmota #2699: problem wifi sonoff t1 2buttons

carlosribeiro, 2018-05-11:
after write firmware(5.13.1a) not connect to wifi and show message on terminal:
00:00:11 WIF: Connect failed with AP timeout
00:00:11 WIF: Connecting to AP1  in mode 11N as sonoff-2897...
00:00:22 WIF: Connect failed with AP timeout
00:00:22 WIF: WPSConfig active for 3 minutes
00:00:43 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

reloxx13, 2018-05-12:
So what?
Your wifi config failed and you need to connect by WPS on default or build the firmware yourself, change to WIFI_WAIT and enter your wifi credentials.

OR

4x btn pressing:
https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage

carlosribeiro, 2018-05-14:
problem resolve. the problem occurred in the merge between the different versions.

closed 2018-05-14 by carlosribeiro