arendst/Tasmota #2700: Sonoff Basis delivered with soldered GPIO

gcb2018, 2018-05-11:
Yesterday I received a new batch of Sonoff Basics and to my surprise all the holes are soldered and now I am no longer able to flash it... Is there anyone else who has had the same? Is this something new? And more importantly: do you know what to do? Can I desolder it (I have no experience with this at all).

Thanks!

Make sure these boxes are checked before submitting your issue - Thank you!

- [x] What hardware you are using? Multiple Sonoff Basics w/ Domoticz and Raspberry Pi 3

Frogmore42, 2018-05-11:
There are two main ways of removing solder, braid or solder sucker (aka desoldering pump).  The braid is usually a cheaper first cost and pretty easy to use.  The solder sucker is generally cheaper over the long run and sometimes works better.  It should be pretty easy to remove the solder to add a header with either method.  I am sure there are lots of YouTube videos on how to do it, if you need guidance.

tobox, 2018-05-12:
If you have neither braid nor a solder sucker, you can try to solder each pin individually by heating up the solder and inserting the pin at the same time. Not pretty, but it should work.

You can also try SonOTA, if you just want to flash the firmware.

Leseratte10, 2018-05-12:
If its a very new batch with even hardware changes (soldered holes) I doubt the firmware is low enough for SonOTA.

Jason2866, 2018-05-12:
May you could post a picture off the new version?

localhost61, 2018-05-12:
@gcb2018,  to empty the hole, try this trick... Heat a hole with the tip on an iron and once the solder is liquid shock the PCB on the table in order to let it go away due to centrifugal force.
It works not so well if there is too little solder, or if it's not warm enough then try to add more solder and retry.
Take care with heavy components like the transformer they don't like to be shocked, keep them in the palm of the hand.

gcb2018, 2018-05-12:
SonOTA won't work as the FW is too new. I attached an image. I will try your advice. 

has anyone been successful with desoldering AND is it still flashable?
[image]

Jason2866, 2018-05-12:
This is not a new sonoff basic version. It will work... just holes with solder.  @localhost61 methode will work
@volkerrichert This is a know issue. Many users have problems with 2.4.0 (1.6) and 2.4.1 (1.7) Version 2.3.0 (1.5) will always work.

MYLE-01, 2018-05-16:
@gcb2018 
Bugger  Just order some more
they must done for a reason
would love to know the count they have sold and the number that are using there software. i bet there are more hack ones than using there software.

volkerrichert, 2018-05-16:
@Jason2866 : ok, good to know in detail. At this point i just want to give a hint to @gcb2018 not to run into the same issue.

gcb2018, 2018-05-18:
Today I received the soldering iron (fine tip). And I think I already messed up two Sonoffs... so I guess I should practice some more or just ask a professional to fix it for me. It's kind of discouraging though.

After soldering I tried flashing it and Arduino gives me ZERO errors. It goes up to 100% just like the other Sonoffs I flashed. I was really happy but after hooking it up to the wall outlet there is no led-light and the sonoff does not show up on the network. Did I mess it up or is this expected behaviour and am I missing something? 

Thanks for your help!

gcb2018, 2018-05-18:
Do you mean attaching it with the FTDI to my Mac -> open Arduino -> Tools -> Serial Monitor? If yes, it's empty. 

I used the following settings:
[image]

I tried both 1M (no spiffs) as 1M (64K spiffs)

arendst, 2018-05-18:
No QIO, No v2 Lower Memory.

pls  re-read wiki - arduino ide

gcb2018, 2018-05-18:
THANK YOU. This worked! You are my hero :)

regoobgib, 2018-05-28:
I received 12 basics with the soldered holes.  I was able to get them flashed without desoldering, placed jumper wires from ftdi to 90degree header pins, held them on soldered points and managed to flash using ArduinoIDE.  It was a pain, but it did work.

ascillato, 2018-05-28:
Happy to know that now it is working for you! :+1: Please, close this issue. Thanks.

loudmicro, 2018-06-08:
🤔 I opened the holes but there's no way my pc can see the board, tried with arduino Ide and espflasher.exe and py but they can't find the sonoff

Frogmore42, 2018-06-08:
@loudmicro you'll need to provide more information.  Please review the basics of flashing, if you still don't get it to work, open a new issue and provide complete information about what you tried.  You said the PC can't see your board, what does that mean, ie what is the exact error message you got that leads you to believe this.

semiautomatix, 2018-06-11:
@loudmicro have you tried using Termite to view the terminal output from the connected device?

closed 2018-06-13 by ascillato2

kelvinn, 2018-12-08:
Belated comment on closed issue for anybody finding this...

Friend asked to solder on headers, yet found out the holes were filled in. Unable to get solder wick to suck it up, so had to put in each pin individually, and it took a while.

Next time I do it I'm going to print one of these to just hold the programmer to the solder-filled pins:

https://www.thingiverse.com/thing:2980893

meingraham, 2018-12-08:
One potential issue is that with the holes filled in, the pogo pins on the clip may not align on the contacts. You can use a 0.9mm drill bit to "cut" a bit of an indentation (I just spin the bit in with my hand - not a electric drill!) into the solder for the pins to seat. Of course, the drill bit can completely drill out the solder too.

kelvinn, 2018-12-09:
Good idea - probably worthwhile I look at getting one... something like this?

https://www.adafruit.com/product/2120

meingraham, 2018-12-09:
@kelvinn

I got these - https://www.ebay.com/itm/392124866294

kelvinn, 2018-12-09:
Bought! Now I can expect to do the next Sonoff on... Feb 6th when the bits arrive.

Cheers.