arendst/Tasmota #2701: DHT Scan Rate

kingphil123, 2018-05-11:
I’m using a D1 Mini with a DHT11 sensor attached and it is happily sending data to my mqtt broker. 

The issue I have is that data is being sent as different frequencies - sometime it will sent the temp every 5 minutes, others it will send it ever 4 hours.

Is there a way to set the scan frequency?

Frogmore42, 2018-05-11:
There is a wealth of information in the wiki, especially the commands section:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

There you will find teleperiod, which sets the frequency (actually the inverse of it) with which Tasmota sends sensor readings.

The default is 5 min.  You can look at the log to see what is going on, when you are not getting readings.

You didn't provide any of the requested information in the template, so it is hard to say why it is not working consistently.

ascillato, 2018-06-04:
Hi

If your issue is solved, please close it. Thanks :+1:

closed 2018-06-11 by ascillato2