arendst/Tasmota #2705: Device going online/offline every 15sec

matrottier, 2018-05-12:
Make sure these boxes are checked before submitting your issue - Thank you!

- [x] What hardware you are using
- [x] Provide the output of command ``status 0``
- [x] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [x] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

Hardware :  
2 Sonoff Basic with v5.13.1 (was version 5.12.0 before it start bugging)
1 Sonoff SV with version v5.12.0
1 Sonoff Touch (US version) v5.12.0

Problem : 
After a restart of my Router (TP-Link Archer C5 AC1200), all the devices started going online and offline each 15-30 seconds. 

log of mosquitto subscription : 
```
11.05.2018 23:22:02 tele/sonoffswitchext/LWT Offline 
11.05.2018 23:22:08 tele/sonoffswitch1/LWT Offline 
11.05.2018 23:22:12 tele/sonoffswitchext/LWT Online 
11.05.2018 23:22:18 tele/sonoffswitch1/LWT Online 
11.05.2018 23:22:24 cmnd/sonoffswitchext/POWER (null) 
11.05.2018 23:22:39 tele/sonoffswitch1/LWT Offline 
11.05.2018 23:22:42 tele/sonoffswitchext/LWT Offline 
11.05.2018 23:22:43 tele/sonoffswitchext/LWT Online 
11.05.2018 23:23:05 tele/sonoffswitchext/LWT Offline 
11.05.2018 23:23:08 tele/sonoffswitchext/LWT Online 
11.05.2018 23:23:10 tele/sonoffswitch1/LWT Offline 
``` 
mosquitto log : 
```
1526095563: Client DVES_2DE9AD has exceeded timeout, disconnecting.
1526095563: Socket error on client DVES_2DE9AD, disconnecting.
1526095565: New connection from 192.168.0.104 on port 1883.
1526095565: New client connected from 192.168.0.104 as DVES_2DE9AD (c1, k15, u'DVES_USER').
1526095566: New connection from 192.168.0.113 on port 1883.
1526095566: New client connected from 192.168.0.113 as DVES_29FB47 (c1, k15, u'DVES_USER').
1526095587: Client DVES_2DE9AD has exceeded timeout, disconnecting.
1526095587: Socket error on client DVES_2DE9AD, disconnecting.
1526095587: Client DVES_29FB47 has exceeded timeout, disconnecting.
1526095587: Socket error on client DVES_29FB47, disconnecting.
1526095589: New connection from 192.168.0.104 on port 1883.
1526095589: New client connected from 192.168.0.104 as DVES_2DE9AD (c1, k15, u'DVES_USER').
1526095597: New connection from 192.168.0.113 on port 1883.
```

try: 
i changed the channel of the wifi. nothing
i restarted the devices, router, mosquitto many times. nothing
i upgraded to v5.13.1 the 2 sonoff basic. nothing

Magic: 
the sonoff SV seem to be OK (24hr without problem)
the sonoff touch going online/offline on a 1 hour interval compare to 15 second with the sonoff Basic

re-flashing the device, is it the only solution ? (PLEASE NO!)

PS: at the same time, i got some issues with Plex Media Server. Maybe it's just my shitty router. =/

Frogmore42, 2018-05-12:
Status 0 of your devices?

matrottier, 2018-05-12:
the result of 'status 0':
### 1 devices not working. can't get the other one. 
is there any other method than the web? (IP_SONOFF/cm?cmnd=status 0)
```
  |  
-- | --
Status |  
Module | 1
FriendlyName |  
0 | "Sonoff-switch1"
Topic | "sonoffswitch1"
ButtonTopic | "0"
Power | 0
PowerOnState | 3
LedState | 1
SaveData | 1
SaveState | 1
ButtonRetain | 0
PowerRetain | 0
StatusPRM |  
Baudrate | 115200
GroupTopic | "sonoffs"
OtaUrl | "http://domus1:80/api/arduino/sonoff.ino.bin"
RestartReason | "Software/System restart"
Uptime | "0T08:35:31"
StartupUTC | "2018-05-12T03:48:00"
Sleep | 0
BootCount | 353
SaveCount | 893
SaveAddress | "FA000"
StatusFWR |  
Version | "5.13.1"
BuildDateTime | "2018-05-01T20:59:22"
Boot | 31
Core | "2_4_0"
SDK | "2.1.0(deb1901)"
StatusLOG |  
SerialLog | 2
WebLog | 2
SysLog | 0
LogHost | "domus1"
LogPort | 514
SSId |  
0 | "ritachabotte24"
1 | "indebuurt2"
TelePeriod | 300
SetOption |  
0 | "00000009"
1 | "55818000"
StatusMEM |  
ProgramSize | 529
Free | 472
Heap | 16
ProgramFlashSize | 1024
FlashSize | 1024
FlashMode | 3
StatusNET |  
Hostname | "sonoffswitch1-6983"
IPAddress | "192.168.0.113"
Gateway | "192.168.0.1"
Subnetmask | "255.255.255.0"
DNSServer | "192.168.0.1"
Mac | "DC:4F:22:29:FB:47"
Webserver | 2
WifiConfig | 3
StatusMQT |  
MqttHost | "192.168.0.102"
MqttPort | 1883
MqttClientMask | "DVES_%06X"
MqttClient | "DVES_29FB47"
[... 258 lines cut ...]
MqttPort | 1883
MqttClientMask | "DVES_%06X"
MqttClient | "DVES_959EE4"
MqttUser | "DVES_USER"
MAX_PACKET_SIZE | 1000
KEEPALIVE | 15
StatusTIM |  
UTC | "Sat May 12 12:31:24 2018"
Local | "Sat May 12 13:31:24 2018"
StartDST | "Sun Mar 25 02:00:00 2018"
EndDST | "Sun Oct 28 03:00:00 2018"
Timezone | 1
StatusSNS |  
Time | "2018-05-12T13:31:24"
StatusSTS |  
Time | "2018-05-12T13:31:24"
Uptime | "0T06:26:22"
Vcc | 3.462
POWER | "OFF"
Wifi |  
AP | 1
SSId | "ritachabotte24"
RSSI | 80
APMac | "60:E3:27:3C:1E:25"
```

matrottier, 2018-05-12:
i can see that "RSSI" is not great but the 2 sonoffs working are the farthest.  
the electric cables can disrupt the signal ?  
distance between router and device :  
sonoff basic 1 (not working) : ~5m
sonoff basic 2 (not working) : ~8m (1 floor lower)
sonoff SV ( working) : ~10m (in the garage)
sonoff SV ( working ) : ~15m (1 floor lower)
sonoff touch (working 75%) : ~13m (1 floor lower)

Jason2866, 2018-05-12:
Flash a version with arduino sdk 2.3.0.  Version 2.4.0 do often make problems with wifi.
The strange thing. Some devices works flawessly others have problems.

reloxx13, 2018-05-12:
do you see the disconnect in the log of your router?

for me it looks like the devices get too much requests in a too short time from mqtt.

Jason2866, 2018-05-12:
Maybe the mqtt broker is flooding the devices because he tries to send retained messages.
And this is too much and the get unresponsive in combination with laggy arduino v 2.4.x

alton76, 2018-05-12:
If MQTT broker is flooding, how to resolve that issue?

Jason2866, 2018-05-12:
See #2140

matrottier, 2018-05-12:
i done a quick test : 
i stopped HASS.
stop mosquitto
delete /var/lib/mosquitto/mosquitto.db to delete all retained message
restart mosquitto
same thing =/

matrottier, 2018-05-13:
### more info : 
i use mosquitto 1.4.15

Jason2866, 2018-05-13:
Do you have used arduino sdk 2.3.0?
If not do this.

daballiemo, 2018-05-13:
I see the same behavior, only thing changed was an update from 5.12 -> 5.13.1 via http://sonoff.maddox.co.uk/tasmota/sonoff.bin:

`2018-05-13 10:48:28.002 [vent.ItemStateChangedEvent] - sonoff_BD6C7F_LWT changed from Online to Offline
2018-05-13 10:48:28.917 [vent.ItemStateChangedEvent] - sonoff_BD6C7F_Reachable changed from OFF to ON
2018-05-13 10:48:28.919 [vent.ItemStateChangedEvent] - sonoff_BD6C7F_LWT changed from Offline to Online
2018-05-13 10:48:40.920 [vent.ItemStateChangedEvent] - sonoff_BC6459_Reachable changed from ON to OFF
2018-05-13 10:48:40.922 [vent.ItemStateChangedEvent] - sonoff_BC6459_LWT changed from Online to Offline
2018-05-13 10:48:41.020 [vent.ItemStateChangedEvent] - sonoff_0F3EAA_Reachable changed from ON to OFF
2018-05-13 10:48:41.025 [vent.ItemStateChangedEvent] - sonoff_0F3EAA_LWT changed from Online to Offline
2018-05-13 10:48:41.115 [vent.ItemStateChangedEvent] - sonoff_0F3EAA_LWT changed from Offline to Online
2018-05-13 10:48:41.117 [vent.ItemStateChangedEvent] - sonoff_0F3EAA_Reachable changed from OFF to ON
2018-05-13 10:48:49.503 [vent.ItemStateChangedEvent] - powerbar_meter_watts5 changed from 0.9 to 1.1
2018-05-13 10:48:50.998 [vent.ItemStateChangedEvent] - sonoff_BD6C7F_Reachable changed from ON to OFF
2018-05-13 10:48:50.999 [vent.ItemStateChangedEvent] - sonoff_BD6C7F_LWT changed from Online to Offline
2018-05-13 10:48:51.906 [vent.ItemStateChangedEvent] - sonoff_BD6C7F_Reachable changed from OFF to ON
2018-05-13 10:48:51.909 [vent.ItemStateChangedEvent] - sonoff_BD6C7F_LWT changed from Offline to Online
2018-05-13 10:48:58.501 [vent.ItemStateChangedEvent] - powerbar_meter_watts5 changed from 1.1 to 0.9
2018-05-13 10:49:02.985 [vent.ItemStateChangedEvent] - sonoff_0F3EAA_Reachable changed from ON to OFF
2018-05-13 10:49:02.988 [vent.ItemStateChangedEvent] - sonoff_0F3EAA_LWT changed from Online to Offline
2018-05-13 10:49:06.472 [vent.ItemStateChangedEvent] - sonoff_BC6459_Reachable changed from OFF to ON
2018-05-13 10:49:06.476 [vent.ItemStateChangedEvent] - sonoff_BC6459_LWT changed from Offline to Online
2018-05-13 10:49:13.950 [vent.ItemStateChangedEvent] - sonoff_BD6C7F_Reachable changed from ON to OFF
2018-05-13 10:49:13.954 [vent.ItemStateChangedEvent] - sonoff_BD6C7F_LWT changed from Online to Offline
2018-05-13 10:49:13.955 [vent.ItemStateChangedEvent] - sonoff_077697_LWT changed from Online to Offline
2018-05-13 10:49:13.956 [vent.ItemStateChangedEvent] - sonoff_077697_Reachable changed from ON to OFF
2018-05-13 10:49:16.502 [vent.ItemStateChangedEvent] - powerbar_meter_watts5 changed from 0.9 to 1.1
2018-05-13 10:49:18.091 [vent.ItemStateChangedEvent] - sonoff_077697_Reachable changed from OFF to ON
2018-05-13 10:49:18.095 [vent.ItemStateChangedEvent] - sonoff_077697_LWT changed from Offline to Online
2018-05-13 10:49:24.939 [vent.ItemStateChangedEvent] - sonoff_BD6C7F_LWT changed from Offline to Online
2018-05-13 10:49:24.941 [vent.ItemStateChangedEvent] - sonoff_BD6C7F_Reachable changed from OFF to ON`

lets see if a downgrade will help

daballiemo, 2018-05-13:
downgrade to 5.12.0 solved the problems

matrottier, 2018-05-13:
i will try somewhere in the week : 
- downgrade sdk 2.3.0 
- upgrade to 2.4.1

matrottier, 2018-05-21:
re-flash with tasmota 5.12 + esp8266 2.4.1.
15 minute without offline. 
going to see tomorrow if it really working :pray:

Jason2866, 2018-05-21:
Just wanna know... Is there a reason that you want to use 2.4.1? 
With 2.3.0 there is no problem at all. Everything works (Hue Emulation!) and sleep command.
And compiled binary is the smallest at all.

matrottier, 2018-05-21:
there is no reason, just keeping up to date =) 
do you think 2.3.0 is really better ? 
tasmota 5.12 + esp8266 2.4.1 seems ok. 
test on 24 hours.
but i will downgrade to 2.3.0 on your advice

matrottier, 2018-05-26:
i'm closing it.
5 devices on tasmota 5.12.0, 2 of 5 with eps8266 2.4.0 SDK 2.1.0
```
{"Version":"5.12.0","BuildDateTime":"2018-05-20T21:58:28","Boot":6,"Core":"2_4_1","SDK":"2.2.1
{"Version":"5.12.0","BuildDateTime":"2018-05-20T21:33:28","Boot":31,"Core":"2_4_1","SDK":"2.2.1
{"Version":"5.12.0","BuildDateTime":"2018-05-20T21:58:28","Boot":6,"Core":"2_4_1","SDK":"2.2.1
{"Version":"5.12.0","BuildDateTime":"2018-04-30T14:16:05","Boot":6,"Core":"2_4_0","SDK":"2.1.0
{"Version":"5.12.0","BuildDateTime":"2018-03-24T14:55:29","Boot":6,"Core":"2_4_0","SDK":"2.1.0
```

everything is ok.
maybe a network problem ...

closed 2018-05-26 by matrottier