arendst/Tasmota #2707: Support multiple I²C-buses or I²C-devices with the same I²C address

renne, 2018-05-12:
- [x] What hardware you are using
   generic

Devices with the same I²C-Address cannot be used on the same I²C-bus. So I suggest to implement the [use of virtual I²C-buses](https://www.esp8266.com/viewtopic.php?f=29&t=13174#p60404) in Tasmota. Only one common SCL-line/GPIO is needed for all I²C-buses. A SDA-line/GPIO has to be exclusive to a bus.

To be able to use the same device or different devices with the same I²C address multiple times I suggest a user-configurable table to assign libraries and I²C-addresses to buses.

e.g.
```
GPIO0:  I2C-SCL
GPIO2:  I2C-SDA-1
GPIO4:  I2C-SDA-2
GPIO5:  I2C-SDA-3
```
```
I2C-SDA-1(0x38) -> lib-VEML6070
I2C-SDA-1(0x39) -> lib-VEML6070
I2C-SDA-2(0x39) -> lib-TSL2561
I2C-SDA-3(0x39) -> lib-APDS9960
```

Frogmore42, 2018-05-12:
Why?
What particular use case do you see for this?
It sounds like a lot of complexity for not so much gain.  For most sensors there is only support for a single one, even when they have configurable addresses.  So, that problem would also need to be addressed.

renne, 2018-05-12:
I want to use e.g. a VEML6070 and a TSL2561 in my weather station.

andrethomas, 2018-06-22:
`Is this the reason a BME280 at 0x76 does not show if a BME680 is at 0x77 at the same time?`

Yes, its because the driver only looks for one instance within its address range and once it finds it, it uses that particular one.

I saw an interesting way around this over here
https://github.com/stefanbode/Sonoff-Tasmota/blob/master/sonoff/xdrv_20_pcf8574.ino

@stefanbode added support for multiple instances of the same I2C device on different I2C addresses to expand his requirements to up to 64 relays I think.

This could be done on the bme sensor library I think but I'm not sure to which point @arendst  would commit changes in this way as it may distract from the lightness of the current I2C model... but certainly something you could do in your own repo/fork.

The above was done on a port expander though and although I've thought of adding support for MCP23008/MCP23017 port expanders this type of I2C interaction/addition seems to be frowned upon as it is not something that one would generally use on a sonoff device.

I only have a few sonoff devices and the rest are all custom ESP8266 circuits which have a combination of other stuff not generally used for sonoff devices so in that case, a separate fork such as the one maintained by @stefanbode makes sense.

closed 2018-08-21 by stale[bot]

Littlesloths, 2021-04-05:
as said in #1049: same here
My usecase is: to connect two SI7021 or AM2301 to read outside and inside abs humidity to control ventilation.

s-hadinger, 2021-04-05:
You need esp32 if you want two physical distinct buses

arendst, 2021-04-06:
> My usecase is: to connect two SI7021 or AM2301 to read outside and inside abs humidity to control ventilation.

Both devices are NOT I2C devices so your remark here makes no sense.

Littlesloths, 2021-04-06:
Please excuse my noob questions, but from similiar sites like
http://www.esp8266learning.com/esp8266-si7021-temperature-sensor-example.php
I got the impression that the sensors has SDA + SCL and hence is I2C.
Sonoff built added an interface to one wire instead of I2C, but on the board is also SDA + SCL available. If not I2C, what protocol is it?

joba-1, 2021-04-06:
@Littlesloths , not as noob as you might think: si7021 really is an i2c sensor and I use an am2301 (One wire protocol) with tasmota. I see no reason why it wont work for you.

Littlesloths, 2021-04-07:
> @Littlesloths , not as noob as you might think: si7021 really is an i2c sensor and I use an am2301 (One wire protocol) with tasmota. I see no reason why it wont work for you.

They work all fine for me (unsing Sonoff one wire), but I like to use multiple sensors preferably on only one GPIO. 
Tasmote seems not to support multiple sensors on one wire, so I was thinking about using I2C (the native I2C interface of the sonsoff sensors (SDA+SCL), but I don't know if that's possible at all?)