arendst/Tasmota #2708: How to connect and setup DS18B20 with Sonoff basic

budisek, 2018-05-12:
Hi,

I am using tasmotized Sonoff basic with DHT22. Works well. 

But i want to measure the temperature in swimming pool with another Sonoff so i need to connect waterproof temperature senzor. I have just bought DS18B20 [here](https://www.aliexpress.com/item/MK00241-Stainless-steel-waterproof-package-type-DS18b20-temperature-probe-temperature-sensor-18B20/32658611153.html) 

Please is there any tutorial how to connect and setup this sensor with tasmota?

Thanks for any advice. 

Budis

Jason2866, 2018-05-12:
Yes as in wiki described
https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-Configuration

Jason2866, 2018-05-12:
Ok. It is easy. Just connect Gnd to Gnd vcc to 3.3V and data to gpio 14
See here https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Basic
You have to solder a 10k Ohm resistor between data and vcc.
In module webpage select on gpio14 ds18b20 sensor
Thats it

budisek, 2018-05-12:
Wow it seems really easy. But i dont understand about resistor. Should i solder it on data wire and another to vcc wire? Or connect both wires with one resistor. 

Also i read here that someone uses 4.7k resistor. Is there any difference?

And i read about change some line in tasmota firmware before upload to sonoff. Is this also needed?

Jason2866, 2018-05-12:
You have to solder the resistor between vcc and data. 
4k7 or 10k take what you have both will work.
But just use one!

budisek, 2018-05-12:
[image]

Is this right?

budisek, 2018-05-12:
Cool. Thank you. I will get some resistors and try it.

Frogmore42, 2018-05-12:
There are lots of tutorials on the web that cover the basics of using sensors.  Both Sparkfun and Adafruit have good ones that cover both the hardware basics and details of the software.  The software details will be different, and you don't really need to follow that to use Tasmota (since the software is already written and tested).  You just need to understand how to hook up the hardware.

budisek, 2018-05-13:
😃 my master-piece draw on ipad? Sure, you are welcome 🤣

Chaot1, 2018-05-14:
I connectet my Sensor as described, but i get 0.0 °C as result from the Sensor.
Do you have any ideas what can be wrong?

Jason2866, 2018-05-14:
@Chaot1 
Plz post output -> status 0
and a photo where can be seen how and where you connected the sensor

Chaot1, 2018-05-14:
`20:05:05 MQT: tele/sonoff/SENSOR = {"Time":"2018-05-14T20:05:05","DS18B20":{"Temperature":0.0},"TempUnit":"C"}`

`20:10:11 MQT: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["AQ_Mond"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
20:10:11 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:20:20","StartupUTC":"2018-05-14T18:49:51","Sleep":0,"BootCount":12,"SaveCount":22,"SaveAddress":"F6000"}}
20:10:11 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.13.1a","BuildDateTime":"2018.05.14 19:44:50","Boot":6,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
20:10:11 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxbrawe",""],"TelePeriod":300,"SetOption":["00008009","55A18000"]}}
20:10:11 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":535,"Free":468,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
20:10:11 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3394","IPAddress":"192.168.8.xxx","Gateway":"192.168.8.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.8.1","Mac":"68:C6:3A:CD:8D:42","Webserver":2,"WifiConfig":4}}
20:10:11 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.8.xxx","MqttPort":1500,"MqttClientMask":"AQ_Mond","MqttClient":"AQ_Mond","MqttUser":"xxxx","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:10:11 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon May 14 19:10:11 2018","Local":"Mon May 14 20:10:11 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:29","Sunset":"19:53"}}
20:10:11 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-05-14T20:10:11","DS18B20":{"Temperature":0.0},"TempUnit":"C"}}
20:10:11 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-05-14T20:10:11","Uptime":"0T00:20:20","Vcc":3.472,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxawe","RSSI":100,"APMac":"78:8A:20:D4:C7:3E"}}}`

Sensor is connected to GPIO 14 and a 4,7 kOhm Resistor between Vcc and GPIO14
Red wire Vcc
black wire Gnd
yellow wire GPIO 14

I also tried to connect to GPIO 3 or 4 with the same result.
Sensor wire is 1m.

Jason2866, 2018-05-14:
Do you have changed something in sensor setup part in userconfig.h for the ds18b20?

Chaot1, 2018-05-14:
No,
i mad no changes.

On a second sonoff Basic i tried some variations:

second Sensor : same Result

activate "#define USE_DS18x20" in user_config.h : no temperature displayed -  no sensor recognized.

budisek, 2018-05-14:
Mine sensor works well right after I added resistor. There is correct temp. value on web interface. 

But i have HAPnodeJS connected for homekit integration and it is not working. I copied the same accessory JS file which i use for DHT22 sensor and I edited necessary fields but it is not giving right values in homekit only default 38’C. 

Is there any homekit user who can help me?

Frogmore42, 2018-05-14:
0 deg C means the pull-up is not attached correctly and/or the data line is shorted to ground.  The CRC will pass for all zeros, which is an unfortunate choice.

85 deg C is the default reading when the sensor has not yet completed a measurement.  You will get this if your connection for VCC is not good.

Either/both problems can cause the sensor to be not found if you restart Tasmota.

Double and triple check your connections.

Chaot1, 2018-05-16:
shame on me.
I wanted to use a 4,7 kOhm Resistor = yellow, violet, red...
but i used a 47 kOhm Resistor = yellow, violet, **orange** ...

Thanks a lot for your help. That way i got the idea to measure all the connections and i found that mistatke.

closed 2018-07-10 by ascillato2

riddik14, 2018-10-21:
Tasmota read multiple ds18b20 on unique gpio with different address?

Frogmore42, 2018-10-21:
Don't use a closed issue to ask a new question.  Use the discord line for questions.  The link is in the template when you create a new issue.

Also your question is not complete enough for me to understand what you are asking.  I believe you are asking if you can have many DS18B20 sensors.  The answer to that is yes.  It supports 8 I believe, but you have to select the option that supports more than one.  All 8 work on a single GPIO.

kevinbaluha, 2020-01-18:
I recently bought a pack of these DS18x20s and tried to get them working. When I hooked them up and powered on the Tasmota (ESP01) board The device would start IF i did not have the DS18x20 connected first. Determined to get them working without waiting for the return cycle for a replacement sensor, I decided I would make it work somehow. I played with diff resisters, the internal pull up etc. What I ended up doing was connecting the vcc for the resistor through a second pin. This worked, but I am relying on the boot process to power on that pin by resetting its pre boot state. (i have to manually turn it on the first time). It's been very reliable, and predictable. Weird, but kept me from having to risk the same behaviour in a replacement.