arendst/Tasmota #2711: choose bssid

Mazzokun, 2018-05-12:
Hi, I have 3 access point into my house and they have the same ssid/password.
Is there a way to say tasmota to connect to a specific bssid?
Thank you

reloxx13, 2018-05-13:
nope

ascillato, 2018-05-13:
Hi, I didn't find any command to do that on the wifi library for esp8266. So, I think It is not possible to do that.

By the way, why do you want to do that? 
And, do you have any device that can do that?

Mazzokun, 2018-05-13:
Because I have noticed that many times the sonoffs become very slow to answer if not unattainable sometimes. so I assumed that having three access points with the same ssd and with the same password the tasmota would randomly connect to one of these (I think it always chose the farthest). So I chose a different ssd to associate with the access point closer to the tasmota and everything works perfectly without slowing down. That's why I wondered if it was possible to specify to the tasmota which of the three access points to connect to.

ascillato, 2018-05-13:
I understand.

Sorry but there is no command for that.

May be you can ask on https://github.com/esp8266/Arduino if they are planning to do something like that. By now, I could not find anything on the wifi libraries. So, if it is not in the ESP8266 wifi libraries, Tasmota can't do anything about that.

ascillato, 2018-05-13:
Please, if they have any answer, share it here.

By now, as nothing can be done, please close this issue. Reopen it when the developers from esp8266 wifi libraries add or plan to add that option.

Thanks

arendst, 2018-05-13:
The espeasy guys are doing something with bssid and also love the wifi stack... https://github.com/letscontrolit/ESPEasy/issues/1302

ascillato, 2018-05-13:
Ups, my bad. I rechecked and found it on the wifi libraries ([BSSID](https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/scan-class.html#bssid))

ascillato2, 2018-06-27:
This issue does not have activity in lasts weeks. If something new comes up, let's reopen it. Thanks

closed 2018-06-27 by ascillato2