arendst/Tasmota #2712: Can't make initial connection to SonOff Basic browsing to 192.168.4.1

sparkgap, 2018-05-12:
I have Raspberry Pi loaded with OpenHAB and Mosquito MQTT broker, using Arduino IDE to initial configure my Sonoff Baasic with Sonoff-Tasmota  5.13.1

After successfully loading (no compile or flash errors) my SonOff Basic with Sonoff-Tasmota,  **I can not connect (browse) to 192.169.4.1 immediately after flashing.**   Nor can I set a static IP address via editing user_config and recompiling.   The serial monitor (w/ initial unmodified user_config) shows the following ......  
*********************************cut'n paste serial monitor ********************************************
00:00:00 WIF: Connecting to AP1  in mode 11N as sonoff-0437...
00:00:15 WIF: Connect failed with AP timeout
00:00:15 WIF: Connecting to AP1  in mode 11N as sonoff-0437...
00:00:30 WIF: Connect failed with AP timeout
00:00:30 WIF: WifiManager active for 3 minutes
00:00:31 HTP: Web server active on sonoff-0437 with IP address 192.168.4.1
00:02:00 RSL: tele/sonoff/UPTIME = {"Time":"1970-01-01T00:02:00","Uptime":"0T00:02:00"}
00:03:34 APP: Restarting
 ets Jan  8 2013,rst cause:1, boot mode:(1,7)
 ets Jan  8 2013,rst cause:4, boot `mode:(1,7)
wdt reset
*****************************end**************************
Following your OpenHAB wiki, I understand that this is the first thing I need to do, configure each SonOff with a unique MQTT topic name.   **I can not get to the config screen you show.**   It is not clear (to me) how I can get to that configuration screen.   I've tried to edit user_config , adding my wifi name and passowrd.  I know it is not taking user_config because I changed SWITCH_MODE from TOGGLE to FOLLOW and pushing button does not change toggle ON/OFF action as seen in the serial monitor.  Furthermore.... how to use the user_config_override.h is confusing to me.  You say change the value of CFG_HOLDER , but what do I chnage it to..... it has a value of 0x20161209

reloxx13, 2018-05-12:
This is the correct way:
1.) static ip is the wrong way for your issue, reset it to default in your user_config.h
2.) add your wifi config in your user_config.h or better use the user_config_override.h (usage described in the file)
3.) read the wiki about CFG_HOLDER. change it, upload, and change it back to default to avoid further loss of config by updating https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#config-problems-can-cause-boot-loops-items-set-in-user_config-to-not-appear-etc
4.) your device gets an ip from your router dhcp now

Dirty Fix for your problem:
If the WifiManager is up, connect to the sonoff Wifi in your Wifi List.
Enter the IP 192.169.4.1 in your Browser.

For both solutions:
If it does not show up, use expressif library @2.3.0 (for atom&platformio look at the platformio.ini file)

thenameiwantedwastaken, 2018-05-13:
In your initial attempt, the sonoff failed to connect to your wifi. Check the details for your SSID and password  are correct in the user_config_override.h.

Open the web interface to your wifi modem / access point. There should be some way to view the attached devices in there. If your sonoff has connected succesfully, you will see it as an attached device. Your modem will have assigned an IP address to it, so you can then copy that IP address into a browser and you should see the user interface.

So, forget about 192.168.4.1... re-set and re-start as described in the answer above, and then check your modem web interface to see if the sonoff has connected, and if it has, what IP was assigned to it.

decadenza, 2018-05-13:
I have the same problem.
First I tried to enter in AP mode (using the button with 4 short press), I saw a SSID "sonoff-2257" but couldn't connect to it.

 Then I set the correct wifi SSID and password in user_config.h and reflashed. It seems that the sonoff cannot connect to the router (the led keeps blinking). Still no luck.

What should I do?

sparkgap, 2018-05-14:
I am unable to point my browser to 192.168.4.1 or to my DHCP set address (now 192.168.1.113) and access the SonOff Basic configuration window.   I have read all the wikis and and followed advice herein, yet I am unable to make sense of it all, especially the USE_CONFIG_OVERRIDE and CFG_HOLDER value thing, which appears to be a date and the need to flash twice (for some reason) to preserve user_config.h  So far user_config has never changed other than what I edited.   One place it says hold the button 4 seconds and another says 40 seconds.   So confusing......

All I want to do is configure 12 SonOff Basic modules with unique Topic names.   I have refreshed user_config.h with a fresh copy from my download area, set IP to 0.0.0.0 (for DHCP), added my mask and gateway, router name, and password, then subtracted one minute from CFG_Holder date value (followed instructions, but don't understand why), then changed button action from TOGGLE to FOLLOW (as a test) and compiled/flashed.  The following Serial Monitor sequence constantly repeats....
********************** paste *************
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_2A01B5, GroupTopic sonoffs) Version 5.13.1-2_4_1
00:00:00 WIF: Connecting to AP1 craftyglass in mode 11N as sonoff-0437...
04:17:37 WIF: Connected
04:17:37 DNS: Initialized
04:17:37 HTP: Web server active on sonoff-0437.local with IP address 192.168.1.113
04:17:37 MQT: Attempting connection...
04:17:38 DNS: Query done. MQTT services found 0
04:17:38 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
04:17:58 WIF: Connect failed as AP cannot be reached
04:17:58 WIF: Connecting to AP1 craftyglass in mode 11N as sonoff-0437...
04:18:07 WIF: Connect failed as AP cannot be reached
04:18:07 WIF: WifiManager active for 3 minutes
04:18:07 HTP: Web server stopped
04:18:07 HTP: Web server active on sonoff-0437 with IP address 192.168.4.1
04:21:10 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
********************** end paste ********************
wiki says holding button 4 seconds (or 40 sec in another place) causes user_config.h to be loaded, but this does not seem to happen because button action as shown in Serial Monitor does not change from TOGGLE to FOLLOW, yet the date value seems to be changing to Jan  8 2013 as seen in the pasted data above.   So confused :(

reloxx13, 2018-05-14:
If you are not using MQTT, disable it by the user_config.h.

The default MQTT library blocks the sonoff if their is no server.

Disable by:
`#define MQTT_USE 0`

MQTT Lib:
```
/*-------------------------------------------------------------------------------------------*\
 * Select ONE of possible three MQTT library types below
\*-------------------------------------------------------------------------------------------*/
  // Default MQTT driver for both non-TLS and TLS connections. Blocks network if MQTT server is unavailable.
#define MQTT_LIBRARY_TYPE      MQTT_PUBSUBCLIENT   // Use PubSubClient library
  // Alternative MQTT driver does not block network when MQTT server is unavailable. No TLS support
//#define MQTT_LIBRARY_TYPE      MQTT_TASMOTAMQTT    // Use TasmotaMqtt library (+4k4 code, +4k mem) - non-TLS only
  // Alternative MQTT driver does not block network when MQTT server is unavailable. No TLS support
//#define MQTT_LIBRARY_TYPE      MQTT_ESPMQTTARDUINO // Use (patched) esp-mqtt-arduino library (+4k8 code, +4k mem) - non-TLS only
```

OR if you use MQTT:

change the lib to MQTT_TASMOTAMQTT and enter your correct MQTT Server Adress in the user_config.h 

`#define MQTT_HOST "" // [MqttHost]`

You need to change the config holder again for both cases in the user_config.h

I described the [CFG_HOLDER](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#cfg_holder-explained) here yesterday.

******
04:18:07 HTP: Web server active on sonoff-0437 with IP address 192.168.4.1
04:21:10 APP: Restarting

in this time their should be a sonoff-0437 wifi network which you have to connect to.

And again: If it does not show up, use expressif library @2.3.0 (for atom&platformio look at the platformio.ini file)

sparkgap, 2018-05-15:
reloxx13 thanks for the help.  Sorry to be so clueless.

I added my Mosquitto MQTT IP address, username, and password, to user_config.h then slightly changed the date value in CDG_HOLDER and compiled/flashed.   This  time I was able to browse to the SonOFf Basic web based configuration screen using (DHCP set) 192.168.1.113.  I was able to toggle the green led ON and OFF, but......  shortly thereafter the web page became erratic and, then unresponsive.  Pushing button quickly 4 times gives 192.168.4.1 , but I can not browse to that address.  When the 192.168.4.1 pops op in the serial monitor it gives quick flashing LED, but I have never been able to connect.   I try different things and get different results, just adding to the confusion.  If just could understood what is going on under-the-hood I could isolate my troubleshooting to a single cause/effect and figure things out myself.  But..... I still do not understand what changing a date value in CFG_HOLDER does.  I read your explanation and have exhausted the wikis.   The state of CFG_HOLDER is being stored somewhere and being acted upon somehow, possibly in sonoff eeprom or a flag within my local Arduino IDE files.   I only downloaded the sonoff-tasmota 5.13.1 , do I need other necessary libraries?  

So far, my understanding is that if I (ever so slightly) change the CFG_HOLDER + or - one, (actually to any new value) the information in SECTION1 gets **# faithfully** loaded (after another compile/flash) to the SonOFF Basic.   And that the value has nothing to do with today's date.  And If I want to make another change to the same SonOff or flash a new sonoff, I only need to change the CFG_HOLDER value again for change to take effect.   I don't understand the need to flash two times and change CFG_HOLDER, as noting ever gets messed up (modified)  in the user_config.h

My need is simple, I need to upload sonoff-tasmota onto 10 Sonoff Bascis and assign a unique Topic to each......... but can't get there.

reloxx13, 2018-05-15:
not sure if you did, so im askin, did you understand that if the wifi manager starts (192.168.1.4) you need to connect your wifi from your pc with the now for 3 mins upcoming "sonoff-xxxx" (xxxx rdm numbers) wifi network.

the WIFI Manager from your sonoff is NOT in your local home wlan.

also, if you set a static IP on your local machine network adapater, disable it when you are using the wifi manager, their could be an IP issue. this often gets forgotten if connecting to another wifi.

you need to change twice to stay on the defualt CFG_HOLDER. if you stay on a custom CFG_HOLDER and you try to update your sonoff in a few weeks/months and forget to use your own custom CFG_HOLDER, all your configs will be gone.

and once again, use expressif 2.3.0 and as suggested, try MQTT_TASMOTAMQTT

> And again: If it does not show up, use expressif library @2.3.0 (for atom&platformio look at the platformio.ini file)

sparkgap, 2018-05-15:
I have distilled thing down to this sequence of vents ----  causing my restart loop:
1) Upon power-up, the sonoff serial monitor says "WiFi Connected"
2) Then says "WiFi Server Active" on 192.168,1.115 (my DHCP assigned IP)
3) Then MQTT shows "Connected" folowed by many MQTT steps are reported.
4) Then WiFI "Connect failed as AP cannot be reached"
5) Then WiFi manager goes active for 3 minutes allowing for connection to 192.168.4.1
6) Then Web Server stopped and the sequence repeats (over and over).

During the "active" time on both 192.168.1.115 and 192.168.4.1 , I can not browse to the sonoff screen.  But....  I can switch the LED on/off by pushing button.  I have tried several different sonoffs and allowed DHCP to find several different IP addresses.  Changes to user_config.h seem to be taking, because with each file save and compile cycle,  I change sonoff name, like sonoff-1, sonoff-2, sonoff-3, etc and it changes in the serial monitor display.  Yes, I changed the value of CFG_HOLDER every time + or - a little. 

At times it works....... after recompile/upload I can occasionally connect to 192.168.1.115 where I can toggle LED and switch screens to view settings.    At times the browsing is sluggish and slow.  I have never been able to connect to 192.168.4.1  Another funny thing...... The first time I loaded sonoff-tasmota, the sonoff console worked perfectly and I played around with the interface config and timers for hours (was quite impressed) , but after trying to flash my second sonoff nothing has worked properly.   So the initial config seems to have been corrupted, but I have refreshed sonoff files from my download area several times, even did many fresh downloads.   ????

Now on to some of your concerns:   I'm not using static IPs.   I can't browse to 192.168.4.1 during the 3 minute period.  I realize the WiFi manager is coming from the sonoff and not a function of raspberry openHAB server or lan.   I have no idea what this means ---> You wrote "...and once again, use expressif 2.3.0 and as suggested, try MQTT_TASMOTAMQTT"   I have spent months learning Arduino IDE and know nothing of atom&platformio...... before I'd do that (learn another interface) I'd get a bunch of mechanical timer to control my IoT light switches ;(    And lastly...... when I did my Arduino IDE initial set-up, I could not determine if my sonoffs are pre 2.4.0. 2.4.0 or 2.4.1 and did nothing other than just moving the files to the required directories.  Could this be the problem.  Where are the part numbers, 2.4.x  ?

This is really frustrating me, the time spent, but not so much the complexity, but the lack of understanding the new terminology.   Computers and IP networks have been my occupation for 40 years, but this IoT is much more diversified to grasp quickly, too many moving parts, Rasberry, OpenHAM, PuTTY, Linux command line, Arduino IDE, C+,  V Studio, MQTT, etc, etc all at once just to turn on a light bulb.  After months of frustration I switched to NodeRed, but found it also as complicated.    The openHAB support community was much more active than NodeRed, so I'm back here again.   And I thank you for your help.

*********** the Loop ***********
00:00:00 Project sonoff-1 Sonoff (Topic sonoff-1, Fallback DVES_7A5DB2, GroupTopic sonoffs) Version 5.13.1-2_4_1
00:00:00 WIF: Connecting to AP1 craftyglass in mode 11N as sonoff-1-7602...
00:00:07 WIF: Connected
00:00:07 DNS: Initialized
00:00:07 HTP: Web server active on sonoff-1-7602.local with IP address 192.168.1.115
00:00:08 MQT: Attempting connection...
00:00:08 MQT: Connected
00:00:08 MQT: tele/sonoff-1/LWT = Online (retained)
00:00:08 MQT: cmnd/sonoff-1/POWER = 
00:00:08 MQT: tele/sonoff-1/INFO1 = {"Module":"Sonoff Basic","Version":"5.13.1","FallbackTopic":"DVES_7A5DB2","GroupTopic":"sonoffs"}
00:00:08 MQT: tele/sonoff-1/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-1-7602","IPAddress":"192.168.1.115"}
00:00:08 MQT: tele/sonoff-1/INFO3 = {"RestartReason":"Power on"}
00:00:09 MQT: stat/sonoff-1/RESULT = {"POWER":"ON"}
00:00:09 MQT: stat/sonoff-1/POWER = ON
00:00:16 MQT: tele/sonoff-1/STATE = {"Time":"1970-01-01T00:00:16","Uptime":"0T00:00:16","Vcc":3.302,"POWER":"ON","Wifi":{"AP":1,"SSId":"craftyglass","RSSI":52,"APMac":"20:AA:4B:A4:79:22"}}
00:00:27 WIF: Connect failed as AP cannot be reached
00:00:27 WIF: Connecting to AP1 craftyglass in mode 11N as sonoff-1-7602...
00:00:36 WIF: Connect failed as AP cannot be reached
00:00:36 WIF: WifiManager active for 3 minutes
00:00:36 HTP: Web server stopped
00:00:36 HTP: Web server active on sonoff-1-7602 with IP address 192.168.4.1
00:02:00 RSL: tele/sonoff-1/UPTIME = {"Time":"1970-01-01T00:02:00","Uptime":"0T00:01:59"}
00:03:39 APP: Restarting
ets Jan  8 2013,rst cause:1, boot mode:(3,6)
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
******************* repeat as above *****************

reloxx13, 2018-05-15:
im not sure for arduino ide,
but following this:
https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE#install-esp8266-board-software
use this URL for 2.3.0:
http://arduino.esp8266.com/versions/2.3.0/package_esp8266com_index.json

and than the copy file section for "before 2.4.0
https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE#esp8266-board-version-before-240

alternativly just try platformio with this guide: https://github.com/arendst/Sonoff-Tasmota/wiki/Beginner-Guide---Create-your-own-Firmware-Build

maybe you should first try to get it to work without mqtt and if everything works as expected add mqtt.

sparkgap, 2018-05-15:
Will look into the above suggestions but a few quick ones before you go to bed.  It's 5:00PM here in Texas.

I'm still very confused about CFG_HOLDER with regard to the double flash, change, upload and compile terminology.   In you previous response you used the word "change" twice.  And in wikis I read flash twice but nothing about about compile twice.    In the Arduino IDE compile and upload is used.    I'm assuming flash and upload are the same.   

So........ when I change settings in user_config.h it is my understanding that I should.... 
1) Make desired changes to user_config.h
2) change CFG_HOLDER to any value other than the default.  
2) then Compile 
3) then Flash (a.k.a Upload in IDE)
4) then restore CFG_HOLDER to default
5) then Compile again
6) then Flash again (a.k.a Upload in IDE)

I don't see the purpose of step 5 and 6.   I do understand how CFG_HOLDER if changed from 
default can be a flag to an automated update process to prevent over write. 
******************
In your above response....... Is the 2.4.0  a hardware (board version) of software version?   I've already read all the wikis during installation last week, but did not know which one to choose..... looked like a board version to me.   I have no idea how to find..... no similar number on board or chips. 
*****************
I assume Platform  IO and Atom are just another IDE and I will still edit user_config.h and have to contend with CFG_HOLDER and the need to double compile/flash.
**********
I downloaded and extracted the .bin but only found a .bin file which I assume is already compiled, but will have to figure out how to upload (flash it) with Atom or Arduino i assume.

reloxx13, 2018-05-15:
> And in wikis I read flash twice but nothing about about compile twice.

you cannot change it on the device without compiling it, so logically ofcourse you have to compile it before flashing

you dont need to flash twice now, your last step if the config is working is to flash with the default cfg holder BEFORE you take any other changes on the web ui of tasmota, cuz those changes would be lost after your last flash with the default cfg holder.

if you put an apple in slot 2 and in 6 month you got another apple which is default in slot 1, your apple in slot 2 is lost if you forget to put the new apple in slot 2.
the same happens with your config if you update to a new tasmota version.

>  I'm assuming flash and upload are the same.

if you press on upload, the flash chip on your device gets flashed

> Is the 2.4.0 a hardware (board version) of software version?

2.4.0 is the expressif library version which tasmota uses. and with 2.4.0 and 2.4.1 wifi is very unstable.

> I assume Platform IO and Atom are just another IDE and I will still edit user_config.h and have to contend with CFG_HOLDER and the need to double compile/flash.

yep, but you change one line and get the other library, while in arduino ide you got a 3-4 step guide and an ugly interface.
with atom&platformio you get also a local installation of esptool to flash the bin i posted.

without atom&platformio you have to install python and esptool yourself which can be a pain in the a**, too :D

sparkgap, 2018-05-16:
Head spinning.  Too many suggested paths to follow.  So I'm backing up, taking it from the top,  and starting all over again at the Arduion IDE section where the following 3 options are presented https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE#copy-files and which I did not follow  because I could not find files and locations.
ESP8266 Board version before 2.4.0
ESP8266 Board version 2.4.0
ESP8266 Board version 2.4.1
1) Which one should I uses (no explanation was given)
2) Where is  --> printf and scanf  (could not find to delete, are they embedded in a script somewhere?)
3) Could not find this folder ---> portable\packages\esp8266\hardware\esp8266\2.4.0 although I did find the replacement text.   Just need to know where to put replacement text?

Now it looks like 5.4.0 is out.  Should I go with that?  I had been using 5.13.1

Again my goal is simple..... To install sonoff-tasmota on about 10 sonoffs, so that I can get on with learning sonoff-tasmota commands and writing openHAB rules using MQTT binding.   Now I'm totally bogged down with simply installing sonaff-tasmota.  I'm beginning to wonder if sonoff-tasmota is only "half-baked" ..... so frustrated .... so much time lost.

sparkgap, 2018-05-17:
Rellox13,
OK,  I found the file locations for 2.4.1 version and it looks like I have that version (built in) so I will upgrade the 2.4.1 per the wiki..... and see what happens.    This step was never done cause I could never find file location and cause it was labled Arduino15 (not Arduino).  If this don't work I will upgrade to tasmots 5.14.0 which has just been released.

But.....  you have highly recommended downgrading to 2.3.0 for wifi stability, but not sure I can do that through the Arduino IDE.  Although I find folders with all 3 versions, but not sure how to force Arduiono IDE to select 2.3.3 

What is the Expresslf anyway?  Just blindly following instructions so not really sure why I do this.

For anyone following this thread and may be wondering what the issue is, here it is -->  I can't get tasmota to flash on my SonOff Basic to get to the point where I can browse the configuration screen.   It gets stuck in a restart loop.

sparkgap, 2018-05-18:
rellox13
Progress.   Using Arduino IDE and new 5.14.0 and copying ...\2.4.1\platform.txt   per instructions seems to be working with repeatability.  Web interface working.... can set timers.  MQTT seems to be connecting and happy.   Late now... so will shut down for the night.  If I can get a command working out of openHAB tomorrow, I'll call it fixed.

reloxx13, 2018-05-23:
note: 2.4.1 has no wifi light sleep = more power consumption = higher bill

sparkgap, 2018-05-30:
Continued from 6 days ago...
Everything working now.  I have several rules working and item properly configured.
I still have one question which is actually the title of this string.   This does not work.....
`4 short presses: Start Wifi manager providing an Access Point with IP address 192.168.4.1 and a web server allowing the configuration of Wifi. The LED will blink during the config period. A single button press during this period will abort and restart sonoff.`

ascillato, 2018-06-21:
Hi,

If your issue is solved, please close it. Thanks! :+1:

closed 2018-06-21 by sparkgap

amore75, 2018-07-15:
@sparkgap 

I'm having the same issue and been trying to figure out what I'm doing wrong for the past 5 days.  Could you please share what exactly you did to get your sonoff working?!