arendst/Tasmota #2715: Requwest fo adding SDM630-Modbus 100A (3phase of SDM120)

roblad, 2018-05-13:
Hi,

Hi,

Is it possible to extent that to 3 phase model

SDM630-Modbus 100A, it is almost the same as SDM120 - extension of modbus values are needed as described 
https://hackster.imgix.net/uploads/attachments/406214/memory_map_pImNpHDybz.png?auto=compress%2Cformat&w=680&h=510&fit=max

sources:
https://forum.arduino.cc/index.php?topic=417768.0

https://github.com/peninquen/Modbus-Energy-Monitor-Arduino

https://www.hackster.io/samuelphy/energy-meter-logger-6a3468

http://www.2flow.se/pdf/SDM120db.pdf

roblad, 2018-05-13:
Hi,

I will order it , I will receive it app. 1 month, after that of cource I can test it for you, but if you will integrate 360 on monday I will order from aliexpress after your confirmation :-)

roblad, 2018-05-13:
Hi,

But could be  as options for groups of data set in user_config for mqtt and disply messages on the www.

gtortone, 2018-05-14:
Hi,
regarding MQTT message size, now my SDM120 publish messages of ~150 bytes;
I start adding support of SDM630 with these metrics: 

https://hackster.imgix.net/uploads/attachments/406214/memory_map_pImNpHDybz.png?auto=compress%2Cformat&w=680&h=510&fit=max

it will increase MQTT message size at ~500-600 bytes

MatzeJoerling, 2018-05-14:
Hi, only the red/magenta values? what about the last 4, freq, import / export energy, total active energy?

roblad, 2018-05-14:
Hi,

I do not know what is en export and import differences, one of them is measured energy, it should be checked, the module gives possibility to attach to solar and probably one of them is for exported energy to power line. Frequency is not needed, but there are some nice other values, but I will test them separately, today I am ordering

roblad, 2018-05-14:
Hi,

Ordered :-)

roblad, 2018-05-14:
Hi,

I think that

0x0166 // L1 - SDM630_TOTAL_ACTIVE_ENERGY [Wh]
0x0168 // L2 - SDM630_TOTAL_ACTIVE_ENERGY [Wh]
0x016A // L3 - SDM630_TOTAL_ACTIVE_ENERGY [Wh]

that would be a summ from them and it is not a n added value for measure

it is enough
0x0156 // Total - SDM630_TOTAL_ACTIVE_ENERGY [Wh]
and

0x0046, // Total - SDM630_FREQUENCY [Hz]

Did you find somthing as one hour energy or summ total energy ? as it is in sonoff POW ?

do not give any added value and that too

0x0012, // L1 - SDM630_APPARENT_POWER [VA]
0x0014, // L2 - SDM630_APPARENT_POWER [VA]
0x0016, // L3 - SDM630_APPARENT_POWER [VA]

arendst, 2018-05-14:
For the JSON message I suggest naming as short as possible but also to have the parameters one level lower than the root to make usage of rules possible as it is now.

1) {"SDM630":{"VoltageL1":230,"VoltageL2":235 …   }}
2) {"SDM630":{"Voltage":[230,235,234],"Current":[10,12,4],"Power":[100 …  }}
3) {"SDM630":{"L1":{"Voltage":230,"Current":10....},{"L2":{"Voltage .. }}}

Option 1 will work with rules now but the string will be long.
I prefer option 2 as it is the shortest. But I will have to check if rules is still working with this. If not I have to update rule processing.
Option 3 definitively won't work and needs a rule rewrite.

Regarding webpage that looks fine to have all three values on one line.

gtortone, 2018-05-14:
@arendst I also prefer option (2) for JSON message that is also close to web presentation of values...

roblad, 2018-05-14:
Hi,

It could be 

I plan to add in Sonoff-Tasmota homepage something like this for each parameter
related to three phases:

SDM630 Voltage (1/2/3) 230/235/234 V for HTTP

For JSON I agree with @arendst

roblad, 2018-05-14:
Hi,

I think that

0x0166 // L1 - SDM630_TOTAL_ACTIVE_ENERGY [Wh]
0x0168 // L2 - SDM630_TOTAL_ACTIVE_ENERGY [Wh]
0x016A // L3 - SDM630_TOTAL_ACTIVE_ENERGY [Wh]

that would be a sum from them and it is not a n added value for measure

it is enough
0x0156 // Total - SDM630_TOTAL_ACTIVE_ENERGY [Wh]
and

0x0046, // Total - SDM630_FREQUENCY [Hz]

Did you find something as one hour energy or sum total energy ? as it is in sonoff POW ?

do not give any added value and that too

0x0012, // L1 - SDM630_APPARENT_POWER [VA]
0x0014, // L2 - SDM630_APPARENT_POWER [VA]
0x0016, // L3 - SDM630_APPARENT_POWER [VA]

there are:

Total kwh register 0x0056
Average line to neutral volts register 0x002A
Average line to line volts. register 0x00CE
Total system power register 0x0034

it could be useful with saving - totals energy today, total energy yesterday

and that important parameter, which shows jumps of voltages in % from standard one i.e. 230V line to neutral - it can be a signal that something wrong is on our electricity supplier and can destroy some electronic equipment

Average line to neutral volts THD register 0x00F8

and optionaly

Average line to line volts THD register 0x0054

https://pl.wikipedia.org/wiki/Zniekszta%C5%82cenia_nieliniowe#/media/File:Class_b_transfer_characteristic.png

see

https://bg-etech.de/download/manual/SDM630Register1-5.pdf

gtortone, 2018-05-14:
@roblad ok, I agree with you to only provide sum metrics for some attributes... could you summarize your _full_ requirements in a post ? for consumption statistics (today, yesterday) I'did not study POW code yet, but I will do it after I release a first version of SDM630 code...

roblad, 2018-05-14:
OK,  may be @arendst will help a little bit :-)

roblad, 2018-05-14:
0x0000, // L1 - SDM630_VOLTAGE [V]
0x0002, // L2 - SDM630_VOLTAGE [V]
0x0004, // L3 - SDM630_VOLTAGE [V]
0x0006, // L1 - SDM630_CURRENT [A]
0x0008, // L2 - SDM630_CURRENT [A]
0x000A, // L3 - SDM630_CURRENT [A]
0x000C, // L1 - SDM630_POWER [W]
0x000E, // L2 - SDM630_POWER [W]
0x0010, // L3 - SDM630_POWER [W]
0x0018, // L1 - SDM630_REACTIVE_POWER [VAR]
0x001A, // L2 - SDM630_REACTIVE_POWER [VAR]
0x001C, // L3 - SDM630_REACTIVE_POWER [VAR]
0x001E, // L1 - SDM630_POWER_FACTOR
0x0020, // L2 - SDM630_POWER_FACTOR
0x0022, // L3 - SDM630_POWER_FACTOR
0x0156 // Total - SDM630_TOTAL_ACTIVE_ENERGY [Wh]

SUGESTION:

0x0056, // Total kwh
0x002A, // Average line to neutral volts
0x00CE, // Average line to line volts 
0x0034, // Total system power
0x00F8, // Average line to neutral volts THD  ( Average line to neutral volts jump)

OPTIONALY:

0x0054, //Average line to line volts THD  ( Average line to line volts jump)

MatzeJoerling, 2018-05-14:
but pls send all possible values to mqtt! everyone could pick what he needs in his home automation magic super calc tool! but if some values are missing i have to hassle with plugin code.
i dont need them on tasmota wemos device website, perhaps only power is of any value. 

that bunch of measurements you dont want do display on such a device, you want to get the data to influx,sql, whatever database and display them with grafana or similar.

perhaps some of the values are of some interest for this new rule feature...

gtortone, 2018-05-14:
@MatzeJoerling there is a limitation of 1000 bytes for each MQTT message... I'm trying to select a set of 'interesting' (for users) values to publish...

gtortone, 2018-05-14:
@arendst or we can think about sending multiple MQTT message for this sensor (e.g. one message related to statistics and one for 'diagnostic' purposes) ?

gtortone, 2018-05-14:
@roblad 0x0056 is tagged as "Maximum total system power demand" . Is what do you want ?

roblad, 2018-05-14:
My mistakes

Total kwh - kwh 0x0156
Average line to line volts THD -％ 0x0154

gtortone, 2018-05-15:
Hi,
PR just proposed for SDM630:
https://github.com/arendst/Sonoff-Tasmota/pull/2735

now, we need to test it !

gtortone, 2018-05-15:
Hi,
for testing purposes I try SDM630 code on a Sonoff Basic connected with a SDM120 power meter:

[image]

MQTT message seems also to be fine (and valid):

{
	"Time": "2018-05-15T17:03:40",
	"ENERGY": {
		"Total": 33.879,
		"ActivePower": [182, 0, 0],
		"ReactivePower": [-181, 0, 0],
		"Factor": [0.71, 0.00, 0.00],
		"Voltage": [237, 0, 0],
		"Current": [1.364, 0.000, 0.000]
	}
}

gtortone, 2018-05-15:
Hi,
I can work on this but I prefer to verify if code works on a SDM630 because I tested it on a SDM120 (1-phase)... PR is available here https://github.com/arendst/Sonoff-Tasmota/pull/2735/commits/9376dca9925c481432c79f07563432cbae414a0d

roblad, 2018-05-15:
We have to wait for China post :-)

roblad, 2018-05-15:
Will you synchronise with @arendst dev branch ?

gtortone, 2018-05-15:
I submitted a PR now it is @arendst turn ! :smile:

roblad, 2018-05-15:
I sync
https://github.com/roblad/Sonoff-Tasmota

roblad, 2018-05-16:
BTW, the 120 has function inside to set value of total energy or starts from 0 and the end?

gtortone, 2018-05-16:
SDM120 starts from 0 to the end and it is not possible to reset kWh

roblad, 2018-05-16:
Could yo think about that parameter in user.config to set initial value for adding to measured values from device

eeprom23, 2018-05-16:
Is sdm530 also work or only 120,630?

roblad, 2018-05-17:
Hi,

You probably forgot about domoticz :-)

gtortone, 2018-05-17:
@roblad I have three values for each voltage, current and power while Domoticz accept a single value only for each metric... any suggestions ? :smile:

roblad, 2018-05-17:
Hi,

The same is for DS sensors, and you can take values  for sum of 3phase divided by 3 :-), and not for all but that one that are available from Sonoff POW :-)

gtortone, 2018-05-17:
for three voltages value an average is fine, but for current (A) and power (kWh) a sum should be fine... isn't it ?

roblad, 2018-05-17:
#ifdef USE_DOMOTICZ
    if (show_energy_period) {  // Only send if telemetry
      dtostrfd(energy_total * 1000, 1, energy_total_chr);
      DomoticzSensorPowerEnergy((uint16_t)energy_power, energy_total_chr);  // PowerUsage, EnergyToday
      DomoticzSensor(DZ_VOLTAGE, energy_voltage_chr);  // Voltage
      DomoticzSensor(DZ_CURRENT, energy_current_chr);  // Current
    }
#endif  // USE_DOMOTICZ

gtortone, 2018-05-17:
these are from a single-phase device with one voltage value, one current value and one power value...

roblad, 2018-05-17:
Hi,

BTW, is it possible to add initial value or save state value as in counter that also in the config you will be able to set initial sdm630_energy_total energy when the sensor starts fro 0 after reset or new one annatched ?

roblad, 2018-05-18:
Hi,

Ok, but starts from 0 I need to set initial value either internally or in tasmota, currently i.e I have 145kwh cosumption and I want to start from 146

MatzeJoerling, 2018-05-18:
this Domoticz  seems to be highly unflexible ;-) put some node-red between you sensors and domoticz.
cripple the sensor (Average Voltage, Amps) is not the right way, my 2ct.

MatzeJoerling, 2018-05-18:
dont get this initial value thing! 
when tasmota starts and reads modbus all KWH values are read from SDM! And that are persistant values!

roblad, 2018-05-18:
Hi,
When I plug SDM it will starts from 0, I do not know if it is possible to set any value as persistant value inside, I need to calibrate with my current consumption and continue with measure of tasmota.  I do not know node-red, I have domoticz and I need to transfer some values there as normal tasmota has.

MatzeJoerling, 2018-05-18:
what SDM do you have? will double check my 630 if it looses the values, but have to switch off my house for that test ;-) will report, but not next time...

roblad, 2018-05-18:
Hi, but there is a qiestion if yo can set some initial value to eeprom on it?

roblad, 2018-05-22:
@MatzeJoerling  

Hi,

Did you tested ?

MatzeJoerling, 2018-05-23:
@roblad 
sorry, no. i have to shutdown the whole house to test that. sdm630 is my main counter for everything.

closed 2018-07-10 by ascillato2

orimate, 2018-07-25:
Hi, how should I connect the sdm120 with wemos?

jascdk, 2018-10-09:
Hi all:) I just got the SDM630 device from eastron. It is working great - BUT do any of you know, if it is possible to delete / reset the TOTAL energy value? I cannot do it via the CLR option on the meter.:(

MatzeJoerling, 2018-10-09:
have the same, running fine since months, and for you question:
no, you can´t! As described in the manual.

jascdk, 2018-10-09:
> have the same, running fine since months, and for you question:
> no, you can´t! As described in the manual.

Not even with Modbus commands?

jascdk, 2018-10-17:
Hi all:) Would it be possible to add the Frequency to the sdm630 in tasmota? Also a function, where it shows how much power in kWh it uses per day would be nice (is this possible)?

dr-apple, 2018-10-21:
Hi i have tested: Total kwh - kwh 0x0156 but it is ervery 0.0 ? Why ?

`/*
  xsns_25_sdm630.ino - Eastron SDM630-Modbus energy meter support for Sonoff-Tasmota

  Copyright (C) 2018  Gennaro Tortone

  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#ifdef USE_SDM630

/*********************************************************************************************\
 * Eastron SDM630-Modbus energy meter
 *
 * Based on: https://github.com/reaper7/SDM_Energy_Meter
\*********************************************************************************************/

#include <TasmotaSerial.h>

TasmotaSerial *SDM630Serial;

uint8_t sdm630_type = 1;
//uint8_t sdm630_state = 0;

float sdm630_voltage[] = {0,0,0};
float sdm630_current[] = {0,0,0,0};
float sdm630_active_power[] = {0,0,0,0};
float sdm630_reactive_power[] = {0,0,0};
float sdm630_power_factor[] = {0,0,0};
float sdm630_energy_total = 0;
float sdm630_hz_total = 0;
bool SDM630_ModbusReceiveReady()
{
  return (SDM630Serial->available() > 1);
}

void SDM630_ModbusSend(uint8_t function_code, uint16_t start_address, uint16_t register_count)
{
  uint8_t frame[8];

  frame[0] = 0x01;          // default SDM630 Modbus address
  frame[1] = function_code;
  frame[2] = (uint8_t)(start_address >> 8);
  frame[3] = (uint8_t)(start_address);
  frame[4] = (uint8_t)(register_count >> 8);
  frame[5] = (uint8_t)(register_count);

  uint16_t crc = SDM630_calculateCRC(frame, 6);  // calculate out crc only from first 6 bytes
  frame[6] = lowByte(crc);
  frame[7] = highByte(crc);

  while (SDM630Serial->available() > 0)  {  // read serial if any old data is available
    SDM630Serial->read();
  }

  SDM630Serial->flush();
  SDM630Serial->write(frame, sizeof(frame));
}
[... 285 lines cut ...]
{
  boolean result = false;

  if (sdm630_type) {
    switch (function) {
      case FUNC_INIT:
        SDM630Init();
        break;
      case FUNC_EVERY_250_MSECOND:
        SDM630250ms();
        break;
      case FUNC_JSON_APPEND:
        SDM630Show(1);
        break;
#ifdef USE_WEBSERVER
      case FUNC_WEB_APPEND:
        SDM630Show(0);
        break;
#endif  // USE_WEBSERVER
    }
  }
  return result;
}

#endif`

gtortone, 2018-10-21:
Hi @dr-apple,
you modified source code,
try to use default xsns_25_sdm630.ino source code and check if value is correct...

dr-apple, 2018-10-21:
i have try with the original xsns_25_sdm630.ino but it is the same, 0 kwh :-(

gtortone, 2018-10-21:
@roblad, @jascdk do you have same problem of @dr-apple ?

jascdk, 2018-10-21:
Hi - nope mine is working absolutely fine 😊

dr-apple, 2018-10-21:
[image]

dr-apple, 2018-10-21:
What is wrong ?

gtortone, 2018-10-21:
@dr-apple can you send your model number?  if I remember there are two models of SDM630...

gtortone, 2018-10-21:
@dr-apple this is a SDM630M-DC that seems to be different from a common SDM630. SDM modules are produced by Eastron but this is produced (or branded) from B+G ETECH...  I suggest you to verify datasheet of module to check if the variable space is the same...

jascdk, 2018-10-21:
[image]

This is my module - and yes it is the eastern module - working great:=)

[3 more replies]